The-OpenROAD-Project/OpenLane #804: docker not running

quantrpeter, 2021-12-25:
Hi All
     I can make mount and get inside the vm. But i can't see anything running by "docker ps", why?

```
~/workspace/OpenLane>make mount
cd /Users/peter/workspace/OpenLane && \
		docker run --rm -v /Users/peter/workspace/OpenLane:/openlane -v /Users/peter/workspace/OpenLane/designs:/openlane/install -v /Users/peter/workspace/OpenLane/pdks:/Users/peter/workspace/OpenLane/pdks -e PDK_ROOT=/Users/peter/workspace/OpenLane/pdks --user 501:20 -ti efabless/openlane:2021.12.21_01.44.20
bash-4.2$ exit
exit
~/workspace/OpenLane>docker ps
CONTAINER ID   IMAGE     COMMAND   CREATED   STATUS    PORTS     NAMES
```

thanks

maliberty, 2021-12-25:
When you exit the container it is no longer running so there is nothing to see in ps.

quantrpeter, 2021-12-26:
thank you hero @maliberty

closed 2021-12-26 by quantrpeter