The-OpenROAD-Project/OpenLane #806: Mac problem

quantrpeter, 2021-12-25:
Hi All
   When i "make pdk", i got:

```
/Users/peter/workspace/OpenLane>make pdk                                                               
cd /Users/peter/workspace/OpenLane/pdks/skywater-pdk && \
		git checkout main && git submodule init && git pull --no-recurse-submodules && \
		git checkout -qf c094b6e83a4f9298e47f696ec5a7fd53535ec5eb
Previous HEAD position was c094b6e Merge pull request #333 from donn/patch-1
Switched to branch 'main'
Your branch is up to date with 'origin/main'.
Already up to date.
cd /Users/peter/workspace/OpenLane/pdks/skywater-pdk && \
		git submodule update --init libraries/sky130_fd_sc_hd/latest && \
		git submodule update --init libraries/sky130_fd_io/latest && \
		git submodule update --init libraries/sky130_fd_sc_hvl/latest && \
		git submodule update --init libraries/sky130_fd_pr/latest && \
		/Library/Developer/CommandLineTools/usr/bin/make -j4 timing
cd /Users/peter/workspace/OpenLane/pdks/skywater-pdk; git submodule update --init third_party/make-env
source /Users/peter/workspace/OpenLane/pdks/skywater-pdk/env/conda/bin/activate && conda env update --name skywater-pdk-scripts --file environment.yml

ConfigurationLoadError: Unable to load configuration file.
  path: /Users/peter/workspace/OpenLane/pdks/skywater-pdk/env/conda/.condarc
  reason: invalid yaml at line 3, column 0

make[1]: *** [/Users/peter/workspace/OpenLane/pdks/skywater-pdk/env/conda/envs/skywater-pdk-scripts/bin/python] Error 1
make: *** [skywater-library] Error 2
```

cat /Users/peter/workspace/OpenLane/pdks/skywater-pdk/env/conda/.condarc
```
envs_dirs:
  - /Users/peter/workspace/OpenLane/pdks/skywater-pdk/env/conda/envs
onda-pkgs
```

donn, 2021-12-27:
Issue template would have, once again, been appreciated, but I know this issue. This is a spurious issue that's also been plaguing our CI for a hot minute.

It comes from upstream skywater-pdk's insistence on using conda, which a monolithic and unwieldy piece of software, for configuring the Python environment. This is not the first time we've had a trouble spot with this software.

A simple retry typically fixes it.

donn, 2021-12-27:
Oh, also, this is not Mac-specific.

dave-roo, 2021-12-28:
I am having the same problem on a fresh ubuntu 20 install.

quantrpeter, 2021-12-29:
I rerun it, it works, i believe is parallelism tasks problem

quantrpeter, 2021-12-30:
@dave-roo , yes, your solution works

closed 2022-04-11 by donn