The-OpenROAD-Project/OpenLane #810: Error at Floorplan phase

ufukyill, 2021-12-27:
### Description
Hi 
I'm trying to run my rtl code through OpenLane flow interactively. "run_synthesis" and "run_floorplan" phases were successful. But at
"run_placement" phase getting this error : 
[image]
(My rtl code is in Verilog HDL which is about 2k rows. Im not sure about whether this effect this or running interactively.)
### Environment
```
Python: v3.6.9
Kernel: Linux v5.4.0-91-generic
Distribution: ubuntu 18.04
Container Engine: docker v20.10.11
OpenLane Git Version: 2021.12.16_04.46.54
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

commit ec570990c76a65b9bf23d89a161dbe075ed8fa15
Author: Donn <me@donn.website>
Date:   Thu Dec 16 19:44:52 2021 +0200

    Remove All Unused Scripts + More Consistent Script Naming (#765)
    
    * Remove Unusued Scripts
    
    ```rb
    scripts = `find ./scripts -type f -maxdepth 1`.chomp.split("\n")
    
    unused_files = []
    
    for script in scripts
        basename = File.basename(script)
        STDERR.puts "Searching for #{basename}..."
        ag_files = `ag --count #{basename}`.chomp.split("\n")
        if ag_files.count > 0
            STDERR.puts "Found #{basename} in #{ag_files.count} files:"
            for file in ag_files
                STDERR.puts "\t-#{file}"
            end
        else
            STDERR.puts "Could not find #{basename} in any files!"
            unused_files << script
        end
    end
    
    STDERR.puts "Unused files: #{unused_files}"
    puts "#{unused_files.map { |f| "'#{f}'" }.join(" ")}"
    ```
    
    * Rename all `camelCase` scripts to `snake_case` for consistency (except mergeLef.py)
    
    ```rb
    scripts = `find ./scripts -type f -maxdepth 1`.chomp.split("\n")
    
    unused_files = []
    
    for script in scripts
        if script.downcase != script
            puts script
        end
    end
    ```
    
    * More naming consistency, folded useless scripts

commit c1083b591020938d54726a32f0dd5adc0136ab2d
Author: Donn <me@donn.website>
Date:   Thu Dec 16 18:35:04 2021 +0200

    Centralize Calling OpenROAD + Automatic Reproducibles (#764)
    
    All OpenROAD scripts are now called with `run_openroad_script`, a variant of `try_catch`, that upon detecting a failure will automatically invoke ./scripts/or_issue.py and create a reproducible within the run directory.
    
    Also tweaks to how yosys is invoked.

commit c481f38c441183857a0361e9d376331eb6860e74
Author: Donn <me@donn.website>
Date:   Wed Dec 15 19:37:51 2021 +0200

    Update OpenROAD to e9d88df, update OpenROAD boost to 1.76 (#762)

```

### Reproduction Material

`run_placement`

### Expected behavior
Expecting to succesful floorplan phase.

### Logs
```
###############################################################################
# Design Rules
###############################################################################
set_max_fanout 5.0000 [current_design]
[INFO]: Setting RC values...
[INFO RSZ-0027] Inserted 3 input buffers.
[INFO RSZ-0028] Inserted 2 output buffers.
[INFO RSZ-0058] Using max wire length 2319um.
[INFO RSZ-0034] Found 15537 slew violations.
[INFO RSZ-0036] Found 807 capacitance violations.
[INFO RSZ-0037] Found 1 long wires.
[INFO RSZ-0038] Inserted 2428 buffers in 15535 nets.
[INFO RSZ-0039] Resized 55104 instances.
[ERROR DPL-0019] detailed placement failed on decryptAES/genVarLoopDecrypt\[14\].SubByte/subByte\[8\].subValue/_1034_
Error: resizer.tcl, 76 DPL-0019
[ERROR]: during executing openroad script /openlane/scripts/openroad/resizer.tcl
[ERROR]: Exit code: 1
[ERROR]: Last 10 lines:
child process exited abnormally

[INFO]: Creating reproducible...
[INFO]: Saving runtime environment...

or_issue.py OpenROAD Issue Packager

EFABLESS CORPORATION AND ALL AUTHORS OF THE OPENLANE PROJECT SHALL NOT BE HELD
LIABLE FOR ANY LEAKS THAT MAY OCCUR TO ANY PROPRIETARY DATA AS A RESULT OF USING
THIS SCRIPT. THIS SCRIPT IS PROVIDED ON AN "AS IS" BASIS, WITHOUT WARRANTIES OR
CONDITIONS OF ANY KIND.

BY USING THIS SCRIPT, YOU ACKNOWLEDGE THAT YOU FULLY UNDERSTAND THIS DISCLAIMER
AND ALL IT ENTAILS.

Parsing config file(s)…
Setting up /openlane/designs/top_level/runs/run27_12/openroad_issue_reproducible…
⚠ /openlane/designs/top_level/runs/run27_12/tmp/placement/8-resizer.sdc was not found, might be a product. Skipping
⭕ Done.
[INFO]: Reproducible packaged: Please tarball and upload designs/top_level/runs/run27_12/openroad_issue_reproducible if you're going to submit an issue.
[INFO]: Calculating Runtime From the Start...
[INFO]: Generating Final Summary Report...
[INFO]: Design Name: top_level
Run Directory: /openlane/designs/top_level/runs/run27_12
Source not found.
----------------------------------------

LVS Summary:
Source: /openlane/designs/top_level/runs/run27_12/logs/finishing/top_level.lvs.lef.log
Source not found.
----------------------------------------

Antenna Summary:
No antenna report found.
[INFO]: check full report here: /openlane/designs/top_level/runs/run27_12/reports/final_summary_report.csv
[INFO]: Saving runtime environment...
[ERROR]: Flow failed.

%  

```

maliberty, 2021-12-27:
You need to upload the issue reproducer.  @donn what is the path to the generated file?

ufukyill, 2021-12-27:
What do you mean by saying issue reproducer? Do you mean netlist file by saying generated file ? I'm new to all of these so don't mind me please.

maliberty, 2021-12-27:
I mean https://github.com/The-OpenROAD-Project/OpenLane/blob/master/docs/source/using_or_issue.md but I think it is autogenerated now but I don't recall where.

maliberty, 2021-12-27:
Your run should have issued this message:
 "Reproducible packaged: Please tarball and upload $reproducible_dir_relative if you're going to submit an issue."

with the path in it.   If you look for openroad_issue_reproducible I think you'll find it.

ufukyill, 2021-12-27:
Thanks. Editted by adding tarball.

maliberty, 2021-12-27:
@donn I see scripts/openroad/opendp.tcl:
`set_placement_padding -global -right $::env(CELL_PAD)
`
but not in scripts/openroad/replace.tcl.  I makes no sense to apply padding during detailed placement that wasn't applied during global placement.

donn, 2021-12-28:
#811. Do you believe this is the only issue here? @maliberty

maliberty, 2021-12-28:
@donn I suspect so but I can't tell until the change goes through.

donn, 2021-12-29:
Well, the relevant change has gone through. I'm going to close this for now and if it resurfaces, @ufukyill please feel free to reopen.

ufukyill, 2021-12-29:
Should i update the OpenLane or smth?

donn, 2021-12-29:
Yup, you should.

maliberty, 2021-12-29:
Please provide an updated testcase

ufukyill, 2021-12-29:
https://github.com/ufukyill/Issue-submit/blob/main/V2openroad_issue_reproducible.tar.xz

ufukyill, 2021-12-30:
Any suggestions ?

maliberty, 2021-12-30:
I just opened it and the design still looks very dense.  Are you sure this is with padding in global placement?

maliberty, 2021-12-30:
are you using the default density?

ufukyill, 2021-12-30:
I exported  CELL_PAD=0 in openlane after "make mount" in last case but density was default. I have already tried as everything is default then i did some changes on config.tcl. Here have a look : https://github.com/ufukyill/Issue-submit/blob/main/config.tcl
Source RTL :  https://github.com/ufukyill/Issue-submit/blob/main/top_level.v
On the other hand, the reason that i run flow interactively is that virtual machine stucks at some point on flow (5.20.smth) so i decided to run flow interactively. I may not be aware of things that i should set or change if there are.

maliberty, 2021-12-30:
Is there a reason you need PAD=0?
In the test case provided you have: export CELL_PAD='4';

maliberty, 2021-12-30:
If I use CELL_PAD='1';  it goes through with no problem.  I suspect you ran global placement with 0 padding and detailed placement with 4 padding which won't work.

ufukyill, 2022-01-02:
I exported CELL_PAD='1' and changed it in the config.tcl too. Using config.tcl given below and as i increased my VM RAM to about 11 gb, i passed placement phase without problem.
Thanks a lot for assistance to @maliberty @donn . 
config :
`set ::env(DESIGN_NAME) top_level

set ::env(VERILOG_FILES) "$::env(DESIGN_DIR)/src/top_level.v"

set ::env(CLOCK_PERIOD) "10.0"
set ::env(CLOCK_PORT) "clock50MHz"
set ::env(FP_CORE_UTIL) {30}
set ::env(CELL_PAD) {1}
set ::env(PL_TARGET_DENSITY) {0.35}
set ::env(GLB_RESIZER_HOLD_MAX_BUFFER_PERCENT) {60}
set ::env(CLOCK_NET) $::env(CLOCK_PORT)

set filename $::env(DESIGN_DIR)/$::env(PDK)_$::env(STD_CELL_LIBRARY)_config.tcl
if { [file exists $filename] == 1} {
	source $filename
}`

closed 2022-01-02 by ufukyill