The-OpenROAD-Project/OpenLane #828: [question] How to set up OpenLane without Docker?

yurivict, 2021-12-29:
Docket isn't available for my platform.

Are there instructions how to set up OpenLane without it?

vijayank88, 2021-12-30:
@yurivict 
refer this https://github.com/The-OpenROAD-Project/OpenLane/blob/master/docs/source/local_installs.md

donn, 2021-12-30:
@yurivict Also, feel free to ask us anything if your intent here is to get OpenLane running on FreeBSD. We already try to avoid GNU-specific APIs but there may be one or two corner cases.

yurivict, 2021-12-30:
Yes, I am trying to run OpenLane on FreeBSD. Thank you.

Trying to port the dependencies.
So far I found that ```cugr``` doesn't build on non-Linux OSes because it contains pre-built static libraries libdef.a and liblef.a that are only for Linux.

It seems that the the ```rsyn``` project should be able to install these libraries but it doesn't build: https://github.com/RsynTeam/rsyn-x/issues/28

That's what stopped me so far.

donn, 2021-12-30:
cugr/drcu are not 100% required, they are simply alternatives to using the global and detailed routers in OpenROAD. The local installer should make no attempt to install them in fact.

donn, 2023-04-13:
Staled out.