The-OpenROAD-Project/OpenLane #829: [ERROR GPL-0302] Use a higher -density or re-floorplan with a larger core area.

quantrpeter, 2021-12-30:
Hi All
   1. I am following this https://www.youtube.com/watch?v=d0hPdkYg5QI&t=3171s to run a small example
   2. I created my_design.v and init the project by command : **flow.tcl -design my_design -src my_sources/my_design.v -init_design_config**

```
module my_design(
    input clk,
    output reg[7:0] count
);
initial count = 0;
always @ (posedge clk) begin
    count <= count + 1'b1;
end
endmodule
```
    3. run the openlane by : **flow.tcl -design my_design -tag first_run**
    4. Then I got this error

```
[INFO]: Running Global Placement...
OpenROAD 719c6dc617d4b8d6896500bb41cdf057864f9e09
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[INFO ODB-0222] Reading LEF file: /openlane/designs/my_design/runs/first_run/tmp/merged_unpadded.lef
[INFO ODB-0223]     Created 13 technology layers
[INFO ODB-0224]     Created 25 technology vias
[INFO ODB-0225]     Created 441 library cells
[INFO ODB-0226] Finished LEF file:  /openlane/designs/my_design/runs/first_run/tmp/merged_unpadded.lef
[INFO ODB-0127] Reading DEF file: /openlane/designs/my_design/runs/first_run/tmp/floorplan/6-pdn.def
[INFO ODB-0128] Design: my_design
[INFO ODB-0130]     Created 11 pins.
[INFO ODB-0131]     Created 45 components and 252 component-terminals.
[INFO ODB-0132]     Created 2 special nets and 170 connections.
[INFO ODB-0133]     Created 25 nets and 82 connections.
[INFO ODB-0134] Finished DEF file: /openlane/designs/my_design/runs/first_run/tmp/floorplan/6-pdn.def
[INFO GPL-0002] DBU: 1000
[INFO GPL-0003] SiteSize: 460 2720
[INFO GPL-0004] CoreAreaLxLy: 5520 10880
[INFO GPL-0005] CoreAreaUxUy: 29440 32640
[INFO GPL-0006] NumInstances: 45
[INFO GPL-0007] NumPlaceInstances: 24
[INFO GPL-0008] NumFixedInstances: 21
[INFO GPL-0009] NumDummyInstances: 0
[INFO GPL-0010] NumNets: 25
[INFO GPL-0011] NumPins: 91
[INFO GPL-0012] DieAreaLxLy: 0 0
[INFO GPL-0013] DieAreaUxUy: 35135 45855
[INFO GPL-0014] CoreAreaLxLy: 5520 10880
[INFO GPL-0015] CoreAreaUxUy: 29440 32640
[INFO GPL-0016] CoreArea: 520499200
[INFO GPL-0017] NonPlaceInstsArea: 66313600
[INFO GPL-0018] PlaceInstsArea: 290278400
[INFO GPL-0019] Util(%): 63.91
[INFO GPL-0020] StdInstsArea: 290278400
[INFO GPL-0021] MacroInstsArea: 0
[InitialPlace]  Iter: 1 CG Error: 0.00000003 HPWL: 529515
[InitialPlace]  Iter: 2 CG Error: 0.00000012 HPWL: 471278
[InitialPlace]  Iter: 3 CG Error: 0.00000002 HPWL: 464836
[InitialPlace]  Iter: 4 CG Error: 0.00000010 HPWL: 465562
[InitialPlace]  Iter: 5 CG Error: 0.00000004 HPWL: 464233
[ERROR GPL-0302] Use a higher -density or re-floorplan with a larger core area.
Given target density: 0.55
Suggested target density: 0.64
Error: replace.tcl, 86 GPL-0302
[ERROR]: during executing openroad script /openlane/scripts/openroad/replace.tcl
[ERROR]: Exit code: 1
[ERROR]: Last 10 lines:
child process exited abnormally

[INFO]: Creating reproducible...
[INFO]: Saving runtime environment...

or_issue.py OpenROAD Issue Packager

EFABLESS CORPORATION AND ALL AUTHORS OF THE OPENLANE PROJECT SHALL NOT BE HELD
LIABLE FOR ANY LEAKS THAT MAY OCCUR TO ANY PROPRIETARY DATA AS A RESULT OF USING
THIS SCRIPT. THIS SCRIPT IS PROVIDED ON AN "AS IS" BASIS, WITHOUT WARRANTIES OR
CONDITIONS OF ANY KIND.

BY USING THIS SCRIPT, YOU ACKNOWLEDGE THAT YOU FULLY UNDERSTAND THIS DISCLAIMER
AND ALL IT ENTAILS.

Parsing config file(s)…
Setting up /openlane/designs/my_design/runs/first_run/openroad_issue_reproducible…
⭕️ Done.
[INFO]: Reproducible packaged: Please tarball and upload designs/my_design/runs/first_run/openroad_issue_reproducible if you're going to submit an issue.

    while executing
"run_openroad_script $::env(SCRIPTS_DIR)/openroad/replace.tcl -indexed_log [index_file $::env(placement_logs)/global.log]"
    (procedure "global_placement_or" line 13)
    invoked from within
"global_placement_or"
    (procedure "run_placement" line 17)
    invoked from within
"run_placement"
    (procedure "run_placement_step" line 8)
    invoked from within
"[lindex $step_exe 0] [lindex $step_exe 1] "
    (procedure "run_non_interactive_mode" line 54)
    invoked from within
"run_non_interactive_mode {*}$argv"
    invoked from within
"if { [info exists flags_map(-interactive)] || [info exists flags_map(-it)] } {
        puts_info "Running interactively"
        if { [info exists arg_values(-file)..."
    (file "/openlane/flow.tcl" line 385)
OpenLane Container (2f73604):/openlane$
```

Please help
thanks

vijayank88, 2021-12-30:
@quantrpeter 
share your config.tcl

vijayank88, 2021-12-30:
also try built-in designs from the directory openlane/designs/
Different example configuration explained here

vijayank88, 2021-12-30:
for tiny design try below format config.tcl
```
# User config
set ::env(DESIGN_NAME)  my_design

# Change if needed
set ::env(VERILOG_FILES) [glob $::env(DESIGN_DIR)/src/*.v]

# Fill this
set ::env(CLOCK_PERIOD) "10.0"
set ::env(CLOCK_PORT) "clk"

set filename $::env(DESIGN_DIR)/$::env(PDK)_$::env(STD_CELL_LIBRARY)_config.tcl
if { [file exists $filename] == 1} {
	source $filename
}
set ::env(FP_CORE_UTIL) "10"
set ::env(PL_TARGET_DENSITY) "0.7"
set ::env(PL_RANDOM_GLB_PLACMENT) "1"
```

closed 2021-12-30 by vijayank88

quantrpeter, 2021-12-30:
After i add, it works. If this line is a must, flow.tcl should add it automatically?
set ::env(PL_TARGET_DENSITY) "0.7"

below is the original file generated by the flow.tcl

```
# User config
set ::env(DESIGN_NAME) my_design

# Change if needed
set ::env(VERILOG_FILES) [glob $::env(DESIGN_DIR)/src/*.v]

# Fill this
set ::env(CLOCK_PERIOD) "10.0"
set ::env(CLOCK_PORT) "clk"
set ::env(CELL_PAD) 4

set filename $::env(DESIGN_DIR)/$::env(PDK)_$::env(STD_CELL_LIBRARY)_config.tcl
if { [file exists $filename] == 1} {
	source $filename
}
```

vijayank88, 2021-12-30:
@quantrpeter  learn flow variables which you can tweak to resolve your errors. Go through documentation for details

quantrpeter, 2021-12-31:
thanks a lot @vijayank88