The-OpenROAD-Project/OpenLane #832: Openlane should have way to specify size of placement halo around macros

antonblanchard, 2021-12-30:
I'm having global routing issues around some hard macros in Microwatt. On the OpenROAD flow, you can specify a placement halo around macros, giving more resources to route out of them. I can't find a way to do this in Openlane.

donn, 2021-12-31:
Will look into it- do you have a link to the requisite variable(s) in ORFS?

maliberty, 2022-01-01:
MACRO_PLACE_HALO & MACRO_PLACE_CHANNEL

closed 2022-01-03 by donn

antonblanchard, 2022-01-03:
Thanks @donn. However what happens if we are manually placing macros (`MACRO_PLACEMENT_CFG`)? `manual_macro_place.py` inserts macros directly into the database, and doesn't use openroad.

antonblanchard, 2022-01-03:
The syntax for the Openroad and Openlane manual macro placement files is different unfortunately.

PR #841 (2022-01-03): Add Macro Placement Configuration Variables
Adds PL_MACRO_HALO and PL_MACRO_CHANNEL.

---

Resolves #832.