The-OpenROAD-Project/OpenLane #833: fork project and pull-openlane

quantrpeter, 2021-12-31:
Hi
   I forked the project and do "make pull-openlane", error occurs:
```
~/workspace/OpenLane>make pull-openlane
Pulling most recent OpenLane image relative to your commit...
docker pull latest
Using default tag: latest
Error response from daemon: pull access denied for latest, repository does not exist or may require 'docker login': denied: requested access to the resource is denied
make: *** [pull-openlane] Error 1
```
thanks
Peter

vijayank88, 2021-12-31:
@quantrpeter For any issue file, survey must
```
python3 ./env.py issue-survey
```

quantrpeter, 2021-12-31:
```
~/workspace/OpenLane>python3 ./env.py issue-survey
Python: v3.9.7
Kernel: Darwin v21.1.0
Distribution: macOS 12.0.1
Container Engine: docker v20.10.11
OpenLane Git Version: 2021.12.31_01.51.42
---
PDK Version Verification Status: FAILED
/Users/peter/workspace/OpenLane/pdks/sky130A not found.
Traceback (most recent call last):
  File "/Users/peter/workspace/OpenLane/dependencies/verify_versions.py", line 124, in verify_versions
    raise Exception(f"{sky130_dir} not found.")
Exception: /Users/peter/workspace/OpenLane/pdks/sky130A not found.

Failed to compare PDKs.
---
Git Log (Last 3 Commits)

commit 368477576d2d266f5097b11089a5be0b294f1c1a
Author: Vitor Bandeira <vvbandeira@users.noreply.github.com>
Date:   Thu Dec 30 18:56:52 2021 -0300

    Disable des design from Jenkins (#831)
    
    [skip ci]
    
    Signed-off-by: Vitor Bandeira <vitor.vbandeira@gmail.com>

commit 2f73604ffcd230665da6c3ee25def2054f7bfed5
Author: Vitor Bandeira <vvbandeira@users.noreply.github.com>
Date:   Wed Dec 29 15:57:40 2021 -0300

    Remove base_image build from Jenkinsfile (#827)
    
    [skip ci]
    
    Signed-off-by: Vitor Bandeira <vitor.vbandeira@gmail.com>

commit 5645e8649d21b7d5f035bf10c786544a318a5169
Merge: e1807e0 e7244cf
Author: Vitor Bandeira <vvbandeira@users.noreply.github.com>
Date:   Wed Dec 29 13:04:24 2021 -0300

    Merge pull request #826 from vvbandeira/jenkins
    
    Install click since OR CI does not have by default
```

vijayank88, 2021-12-31:
@quantrpeter Have you installed
Prerequisites

At a minimum:

    Docker 19.03.12+
    GNU Make
    Python 3.6+ with PIP
    Click, Pyyaml: python3 -m pip install pyyaml click

quantrpeter, 2022-01-01:
@vijayank88 yes, i did

donn, 2022-01-03:
@vijayank88 For the future, you can check for this info using the issue survey, that's what it's for:

```
Issue Survey Runs -> Python 3.6+ is installed
Container Engine: docker v20.10.11 -> Docker minimum is installed
No errors reported -> pyyaml, click installed
```

@quantrpeter I'm investigating.

donn, 2022-01-05:
`make pull-openlane` won't work on GitHub forks.

You'd need to:

```
export OPENLANE_IMAGE_NAME=quantrpeter/openlane:current
make openlane
```

then export `export OPENLANE_IMAGE_NAME=quantrpeter/openlane:current` before all invocations.

closed 2022-01-05 by donn