The-OpenROAD-Project/OpenLane #834: no standard cell found after synthesis

vijayank88, 2021-12-31:
### Description
no standard cell found after synthesis in design statistics.  Reported in slack channel https://skywater-pdk.slack.com/archives/C016UL7AQ73/p1640850070331700
[openroad_issue_reproducible.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/7796024/openroad_issue_reproducible.zip)

### Environment
```
Please run the following command in the OpenLane folder:

python3 ./env.py issue-survey
Python: v3.6.8
Kernel: Linux v3.10.0-1160.45.1.el7.x86_64
Distribution: centos 7
Container Engine: docker v20.10.12
OpenLane Git Version: 2021.12.30_01.40.40
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

commit 2f73604ffcd230665da6c3ee25def2054f7bfed5
Author: Vitor Bandeira <vvbandeira@users.noreply.github.com>
Date:   Wed Dec 29 15:57:40 2021 -0300

    Remove base_image build from Jenkinsfile (#827)
    
    [skip ci]
    
    Signed-off-by: Vitor Bandeira <vitor.vbandeira@gmail.com>

commit 5645e8649d21b7d5f035bf10c786544a318a5169
Merge: e1807e0 e7244cf
Author: Vitor Bandeira <vvbandeira@users.noreply.github.com>
Date:   Wed Dec 29 13:04:24 2021 -0300

    Merge pull request #826 from vvbandeira/jenkins
    
    Install click since OR CI does not have by default

commit e7244cf736754645bb2218215ceb1b0d3a9db564
Author: Vitor Bandeira <vitor.vbandeira@gmail.com>
Date:   Wed Dec 29 10:18:18 2021 -0300

    Install click since OR CI does not have by default
    
    [skip ci]
    
    Signed-off-by: Vitor Bandeira <vitor.vbandeira@gmail.com>

```

### Reproduction Material
* tar file attached. verilog source attached as well in src directory.

### Expected behavior
successful GDSII generation 

### Logs
```

15.5. Executing OPT_CLEAN pass (remove unused cells and wires).
Finding unused cells or wires in module \main..
Removed 1010 unused cells and 1736 unused wires.
<suppressed ~1104 debug messages>

20. Printing statistics.

=== main ===

   Number of wires:                  2
   Number of wire bits:              2
   Number of public wires:           2
   Number of public wire bits:       2
   Number of memories:               0
   Number of memory bits:            0
   Number of processes:              0
   Number of cells:                  0

```

donn, 2022-01-03:
Reproducible is insufficient- the OpenROAD issue reproducible means nothing as the failure appears to be in Yosys. Please upload a full OpenLane design.

vijayank88, 2022-01-03:
@donn 
design file available in src_verilog/main.v

donn, 2022-01-03:
... `tar -czvf ./design.tar.gz <path/to/design>`

vijayank88, 2022-01-03:
[main.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/7801555/main.tar.gz)
 My run directory attached

donn, 2022-01-03:
The module has ***no*** outputs, only two inputs, clk and rst.

Being that the module has no outputs, it has no functionality, so any synthesizer will optimize the whole thing out for efficiency's sake. This is not OpenLane specific: a synthesis script as simple as `yosys -p "read_verilog src/risc32.v; hierarchy -top main; synth"` will do the exact same thing.

I would recommend the designer either invest in GPIO for this module or add a dummy output like `output [31:0] ReadData`.

closed 2022-01-03 by donn

EmJunaid, 2022-01-03:
@donn yes you are right, design does not has any output because it's very basic level single cycle processor that's just executing instructions and as memories are also part of core so even data is also not an output. I'm working on implementation of some peripherals like, UART, SPI, RAM etc. I just try this design to understand the flow of Openlane. so if I just add a dummy output variable, will my design be able to complete the openlane flow?

donn, 2022-01-03:
As discussed on slack, no. The output variable has to be part of your code path.

Yosys finds codepaths with no output and nukes them. This is to save chip space.