The-OpenROAD-Project/OpenLane #835: local install not working

quantrpeter, 2022-01-01:
Hi, i am using ubuntu 20.04 lts, i have manually built opensta and it still saying opentimer not found. At the end it also throw exception that i can't understand, thanks

```
Using yosys verilog synthesis tool at: /usr/bin/yosys
Using graywolf placement tool at: /usr/bin/graywolf
RePlAce not found.  Use --with-replace=<DIR> (optional)
Using qrouter detail route tool at: /usr/bin/qrouter
Using Magic layout tool at: /usr/bin/magic
Using Netgen LVS tool at: /usr/bin/netgen
Using Vesta STA tool (internal)
OpenTimer not found.  Use --with-opentimer=<DIR> (optional)
Using OpenSTA static timing analysys tool at: /usr/local/bin/sta
----------------------------------------------------
All tools needed for complete flow have been found.
Run "make" followed by "sudo make install" to install qflow.

----------------------------------------------------
rm -f -f vlog2Spice.o vlog2Verilog.o vlog2Def.o vlog2Cel.o vlogFanout.o DEF2Verilog.o addspacers.o vesta.o spice2delay.o rc2dly.o blif2BSpice.o blif2Verilog.o blifFanout.o hash.o readliberty.o readlef.o readdef.o readverilog.o
rm -f -f vlog2Spice vlog2Verilog vlog2Def vlog2Cel vlogFanout DEF2Verilog addspacers vesta spice2delay rc2dly blif2BSpice blif2Verilog blifFanout
make: *** No rule to make target 'vlogtoverilog'.  Stop.
Traceback (most recent call last):
  File "./env.py", line 536, in <module>
    main()
  File "./env.py", line 533, in main
    commands[args[0]]()
  File "./env.py", line 407, in local_install
    installer.run()
  File "./env.py", line 393, in run
    install()
  File "./env.py", line 47, in __exit__
    raise exc_value
  File "./env.py", line 393, in run
    install()
  File "./env.py", line 377, in install
    f.write(tool.version_string)
  File "./env.py", line 47, in __exit__
    raise exc_value
  File "./env.py", line 368, in install
    subprocess.run([
  File "./env.py", line 47, in __exit__
    raise exc_value
  File "./env.py", line 368, in install
    subprocess.run([
  File "./env.py", line 47, in __exit__
    raise exc_value
  File "./env.py", line 368, in install
    subprocess.run([
  File "/usr/lib/python3.8/subprocess.py", line 516, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command '['bash', '-c', '                                    set -e\n                                    source /home/peter/workspace/OpenLane/install/venv/bin/activate\n                                    # Note that vlogtoverilog is part of the qflow repository.\n./configure\ncd src\nmake clean\nmake -j$(nproc) vlogtoverilog\ncp vlogtoverilog $PREFIX/bin\n\n
       ']' returned non-zero exit status 2.
```

FaragElsayed2, 2022-01-02:
Hi @quantrpeter 

This error due to old "./dependencies/tool_metadata.yml" file, as  vlogtoverilog target has been changed to vlog2Verilog

donn, 2022-01-02:
The tool manifest is not out of date, but the build instructions in it are.

donn, 2022-01-02:
The PR is out. Unfortunately, as it is the weekend, I can't get any approvals, so it'll have to be merged tomorrow.

Apologies for the inconvenience.

PR #838 (2022-01-03): Various Local Install Fixes
~ Fixed build script for vlog2Verilog
~ Fixed dependencies for Ubuntu-20.04

---
Resolves #835
closed 2022-01-03 by donn