The-OpenROAD-Project/OpenLane #852: Macro not found in Macro placement

microSharjeel, 2022-01-07:
Hi,
I am trying to place a already hardened Macro in my design. I have followed the manual_macro_placement_test design example and am encountering the following error
Traceback (most recent call last):
  File "/openLANE_flow/scripts/manual_macro_place.py", line 119, in <module>
    assert not macros, ("Macros not found:", macros)
AssertionError: ('Macros not found:', {'ram': ['5590', '168230', 'N']})
If anyone can guide me what is the problem why OpenLane can not find the Macro although the .lef and .gds files are there in Macros folder
Kindly guide
regards

vijayank88, 2022-01-07:
environment survey must and re-producible tar file must to debug the issue

vijayank88, 2022-01-07:
@microSharjeel  are you trying built-in example or your own design?
If its your own design mean need re-producible tar file needed with environment survey

microSharjeel, 2022-01-07:
Thanks for reply 
I am attaching the tar of the design since the design is my own.  I am just trying to place macros of an SRAM in my design the .gds and .lef files are there in the Macros
regard
[iob_sp_ram_be.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/7829292/iob_sp_ram_be.zip)
s

microSharjeel, 2022-01-07:
Are you able to reproduce the error 
best regards

vijayank88, 2022-01-07:
@microSharjeel 
Refer built-in example manual_macro_placement design from openlane, it will work.
From your config.tcl dont use both DIE_AREA and FP_CORE_UTIL... 
Next src/*.v add (*blackbox*) before module definition. Format available in built-in example

maliberty, 2022-01-08:
Test cases should be packaged with https://github.com/The-OpenROAD-Project/OpenLane/blob/master/docs/source/using_or_issue.md

microSharjeel, 2022-01-08:
This issue is resolved by following the built in example of OpenLane

closed 2022-01-08 by microSharjeel

0616ygh, 2022-01-10:
hello If possible, I would like to know how you finally solved this problem? I encountered the same error as you, also using openRAM using sky water130. Thanks a million!!

microSharjeel, 2022-01-10:
@0616ygh  can we connect on google hangout etc I hope it will be mutually beneficial

0616ygh, 2022-01-10:
@microSharjeel I'm sorry but I don't have a google hangout...I'm a student from China mainland, we don't have hangout QAQ
My question is that after I read Openlane->design->manual_macro_placement_test, I find I don't have :

set ::env(MACRO_PLACEMENT_CFG) $::env(DESIGN_DIR)/macro_placement.cfg
set ::env(BASE_SDC_FILE) $::env(DESIGN_DIR)/base.sdc

these 2 files can't be found in OpenRAM package...
[image]

0616ygh, 2022-01-10:
@microSharjeel Thx for your reply~

28aashish, 2022-01-15:
@microSharjeel,
 I am getting errors after Resizer Timing Optimizations 

"[ERROR GRT-0029] Pin ctrlStorage/dout0[0] does not have geometries below the max routing layer (met5).
"
 Description of my design:-
"Designing a controller with custom size memory.SRAM Memory was created with the help of OpenRAM "