The-OpenROAD-Project/OpenLane #857: [ERROR MPL-0004] No macros found.

0616ygh, 2022-01-10:
When I run Openlane for my RTL design, the tool generated an error log:
[ERROR MPL-0004] No macros found.
Error: basic_mp.tcl, 28 MPL-0004
and I found the "basic_mp.tcl",line 28 is:macro_placement -channel {0 0} -halo {0 0}
I don't know why this bug occurs....

vijayank88, 2022-01-10:
@0616ygh 
Refer built-in example manual_macro_placement design from openlane, it will work.
If still error repeats, provide test cases should be packaged with https://github.com/The-OpenROAD-Project/OpenLane/blob/master/docs/source/using_or_issue.md

0616ygh, 2022-01-10:
Thank you a lot ! I will try your method and debug for it. Thank you a lot

closed 2022-01-10 by 0616ygh

0616ygh, 2022-01-10:
Thx you very much. But I find I don't have macro_placement.cfg and base.sdc I don't understand how these parameters are set
[image]

maliberty, 2022-01-10:
Just because you can work around the issue with manual macro placement doesn't mean there isn't a problem with automatic macro placement.  If you want that looked at your should provide a test case.  If you need further info you should reopen the issue.