The-OpenROAD-Project/OpenLane #858: [ERROR GRT-0229] Vertical edge usage exceeds the maximum allowed. Error: groute.tcl, 71 GRT-0229

ufukyill, 2022-01-10:
### Description
A clear and concise description of what the bug is.

### Environment
```
Please run the following command in the OpenLane folder:

python3 ./env.py issue-survey

And copy and paste the ENTIRE output between the triple-backticks. Please do not gzip and upload the output.

If there's no env.py, you are using an out of date version of OpenLane and should probably update. 
```

### Reproduction Material
* Upload a tarball containing the relevant design.
* List the commands used to run the design.

If you see a message like `Reproducible packaged: Please tarball and upload <PATH> if you're going to submit an issue` in your logs, please also tarball and include that path. This will greatly speed up the fixing process.

### Expected behavior
A clear and concise description of what you expected to happen.

### Logs
```
Add any relevant logs here. Please do ensure they're enclosed by the triple-backticks.
```

vijayank88, 2022-01-10:
@ufukyill 
update your config.tcl following variables and run the flow, working fine at my end
```
set ::env(DIODE_PADDING) {1};
set ::env(DIODE_INSERTION_STRATEGY) {5};
```

donn, 2022-01-10:
I'll be honest, this issue's been affecting us and the OpenLane CI as well. I'm not sure when it showed up, but I don't like it.

ufukyill, 2022-01-10:
> @ufukyill update your config.tcl following variables and run the flow, working fine at my end
> 
> ```
> set ::env(DIODE_PADDING) {1};
> set ::env(DIODE_INSERTION_STRATEGY) {5};
> ```

i updated config and reprep -design then run_rouing however still giving same error.

vijayank88, 2022-01-10:
@ufukyill just run your test case alone with above update. See my tail of logs
```
===========================================================================
 report_power
============================================================================
Group                  Internal  Switching    Leakage      Total
                          Power      Power      Power      Power
----------------------------------------------------------------
Sequential             6.75e-02   5.98e-02   5.76e-08   1.27e-01   6.5%
Combinational          5.34e-01   1.29e+00   3.15e-07   1.83e+00  93.5%
Macro                  0.00e+00   0.00e+00   0.00e+00   0.00e+00   0.0%
Pad                    0.00e+00   0.00e+00   0.00e+00   0.00e+00   0.0%
----------------------------------------------------------------
Total                  6.02e-01   1.35e+00   3.73e-07   1.95e+00 100.0%
                          30.8%      69.2%       0.0%
power_report_end
area_report

===========================================================================
 report_design_area
============================================================================
Design area 1180836 u^2 42% utilization.
area_report_end
```

ufukyill, 2022-01-10:
@vijayank88 what do you mean by saying "just run test case alone" ? what should i do differently from i  said here?

> > @ufukyill update your config.tcl following variables and run the flow, working fine at my end
> > ```
> > set ::env(DIODE_PADDING) {1};
> > set ::env(DIODE_INSERTION_STRATEGY) {5};
> > ```
> 
> i updated config and reprep -design then run_rouing however still giving same error.

vijayank88, 2022-01-10:
From OpenLane docker you can launch openroad. goto your openroad-reproducible folder and run
openroad>source run.tcl

ufukyill, 2022-01-10:
I have created a new run afted updating config. now its still running. How can i use the next command ? 
(`write_powered_verilog` followed by `set_netlist $::env(routing_logs)/$::env(DESIGN_NAME).powered.v` )

maliberty, 2022-01-10:
I ran your testcase with the original strategy without any error in the latest OR (ddbc2abd9cb7f29bc07fd9a2192ed5b2523075af).

ufukyill, 2022-01-11:
I encountered an error without any specific error code at the 1st optimization iteration. Here is end of the log :
```
[INFO DRT-0194] Start detail routing.
[INFO DRT-0195] Start 0th optimization iteration.
    Completing 10% with 0 violations.
    elapsed time = 00:28:01, memory = 5494.29 (MB).
    Completing 20% with 0 violations.
    elapsed time = 01:01:00, memory = 8769.23 (MB).
    Completing 30% with 123851 violations.
    elapsed time = 01:24:38, memory = 9301.39 (MB).
    Completing 40% with 123851 violations.
    elapsed time = 01:56:15, memory = 9365.93 (MB).
    Completing 50% with 123851 violations.
    elapsed time = 02:26:32, memory = 10089.87 (MB).
    Completing 60% with 248144 violations.
    elapsed time = 03:04:48, memory = 9721.90 (MB).
    Completing 70% with 248144 violations.
    elapsed time = 03:40:22, memory = 9803.73 (MB).
    Completing 80% with 365376 violations.
    elapsed time = 05:17:58, memory = 9746.37 (MB).
    Completing 90% with 365376 violations.
    elapsed time = 05:51:17, memory = 9815.42 (MB).
    Completing 100% with 483807 violations.
    elapsed time = 06:33:49, memory = 10293.14 (MB).
[INFO DRT-0199]   Number of violations = 641780.
[INFO DRT-0267] cpu time = 10:53:38, elapsed time = 06:34:30, memory = 10011.06 (MB), peak = 10370.48 (MB)
Total wire length = 7862175 um.
Total wire length on LAYER li1 = 0 um.
Total wire length on LAYER met1 = 2932197 um.
Total wire length on LAYER met2 = 3053747 um.
Total wire length on LAYER met3 = 1056610 um.
Total wire length on LAYER met4 = 818130 um.
Total wire length on LAYER met5 = 1488 um.
Total number of vias = 1639427.
Up-via summary (total 1639427):.

--------------------------
 FR_MASTERSLICE          0
            li1     708032
           met1     785994
           met2     122588
           met3      22809
           met4          4
--------------------------
                   1639427

[INFO DRT-0195] Start 1st optimization iteration.
    Completing 10% with 641780 violations.
    elapsed time = 00:27:01, memory = 10117.57 (MB).
    Completing 20% with 641780 violations.
    elapsed time = 00:57:13, memory = 10172.01 (MB).
    Completing 30% with 566383 violations.
    elapsed time = 01:27:32, memory = 9894.66 (MB).
    Completing 40% with 566383 violations.
    elapsed time = 02:01:58, memory = 10156.57 (MB).
[ERROR]: during executing openroad script /openlane/scripts/openroad/droute.tcl
[ERROR]: Exit code: 1
[ERROR]: Last 10 lines:
child killed: kill signal
```
Here is the tarball : https://github.com/ufukyill/Issue-submit/blob/main/1001_1_2nd_openroad_issue_reproducible.tar.xz
Due to ram is not enough i guess, isn't it?

vijayank88, 2022-01-11:
@ufukyill 
we both following this past two weeks. I mentioned clearly you need sufficient RAM to run this. 12GB RAM you mentioned it I think.
@donn is this enough to run this design in OpenLane?

ufukyill, 2022-01-11:
> @ufukyill 
> we both following this past two weeks. I mentioned clearly you need sufficient RAM to run this. 12GB RAM you mentioned it I think.
> @donn is this enough to run this design in OpenLane?

10.8 gb RAM sharing with virtual machine right now.

vijayank88, 2022-01-11:
@ufukyill Last try increase your VM RAM to 14GB and try it.

ufukyill, 2022-01-11:
Then computer might not be able to run the virtualBox and system itself etc. (Total 16gb) Once i have increased about 13.5 gb however then computer did not respond me again.

vijayank88, 2022-01-11:
@ufukyill  why can't use your local machine to utilize full RAM?

ufukyill, 2022-01-11:
I don't have enough time to build dual boot and openlane again i guess.

maliberty, 2022-01-11:
are you using virtual box or docker?

Btw, this issue has no connection to the original report and now the title is irrelevant.  It would have been better to close this and open a new one for the unrelated dr problem.

ufukyill, 2022-01-11:
Im using Oracle VM VirtualBox

maliberty, 2022-01-11:
I see the virtual size hitting 13.4 in my run

maliberty, 2022-01-11:
Docker tends to be lighter weight fwiw.  Your design is quite dense and I ran it overnight and it is still going.   Did you increase the placement density?

closed 2022-01-19 by ufukyill