The-OpenROAD-Project/OpenLane #859: Illegal Overlaps in the Design

ufukyill, 2022-01-11:
### Description

When running a flow interactively on the OpenLane, at the run_magic_spice_export stage, I have faced an error like;
[ERROR]: There are illegal overlaps (e.g., routes over obstructions) in your design.

### Environment
```
Kernel: Linux v5.4.0-92-generic
Distribution: ubuntu 18.04
Python: v3.6.9 (OK)
Container Engine: docker v20.10.12 (OK)
OpenLane Git Version: 2022.01.10_01.49.32
pip:click: INSTALLED
pip:pyyaml: INSTALLED
pip:venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

af03973 2022-01-10T05:11:05+02:00 Add info about OpenROAD tarballs (#855) - Donn -  (HEAD -> master, origin/master, origin/HEAD)
99cf10c 2022-01-09T21:15:47+02:00 Add an ECO Flow for hold timing violations fixes (#791) - Mehdi Saligane -  (tag: 2022.01.10_01.49.32)
569e10c 2022-01-06T16:34:12+02:00 Fixes Requested By Caravel Team (#847) - Donn -  (tag: 2022.01.07_01.36.15)

```

### Reproduction Material

* List the commands used to run the design.
...
run_routing
write_powered_verilog
set_netlist $::env(routing_logs)/$::env(DESIGN_NAME).powered.v
run_magic
run_magic_spice_export

There is no message like reproducible package.

### Expected behavior
After the routing process, I was not expecting to see any overlaps over routes and obstructions. 

### Logs
```
Extracting sky130_fd_sc_hd__dfxtp_1 into sky130_fd_sc_hd__dfxtp_1.ext:
Extracting sky130_fd_sc_hd__clkbuf_2 into sky130_fd_sc_hd__clkbuf_2.ext:
Extracting sky130_fd_sc_hd__clkbuf_4 into sky130_fd_sc_hd__clkbuf_4.ext:
Extracting sky130_fd_sc_hd__clkbuf_16 into sky130_fd_sc_hd__clkbuf_16.ext:
Extracting sky130_fd_sc_hd__a22o_1 into sky130_fd_sc_hd__a22o_1.ext:
Extracting sky130_fd_sc_hd__clkbuf_1 into sky130_fd_sc_hd__clkbuf_1.ext:
Extracting sky130_fd_sc_hd__dfxtp_4 into sky130_fd_sc_hd__dfxtp_4.ext:
Extracting sky130_fd_sc_hd__decap_4 into sky130_fd_sc_hd__decap_4.ext:
Extracting sky130_fd_sc_hd__decap_8 into sky130_fd_sc_hd__decap_8.ext:
Extracting sky130_fd_sc_hd__buf_6 into sky130_fd_sc_hd__buf_6.ext:
Extracting sky130_fd_sc_hd__decap_6 into sky130_fd_sc_hd__decap_6.ext:
Extracting sky130_fd_sc_hd__fill_2 into sky130_fd_sc_hd__fill_2.ext:
Extracting sky130_fd_sc_hd__decap_12 into sky130_fd_sc_hd__decap_12.ext:
Extracting sky130_fd_sc_hd__diode_2 into sky130_fd_sc_hd__diode_2.ext:
Extracting sky130_fd_sc_hd__tapvpwrvgnd_1 into sky130_fd_sc_hd__tapvpwrvgnd_1.ext:
Extracting sky130_fd_sc_hd__buf_12 into sky130_fd_sc_hd__buf_12.ext:
Extracting sky130_fd_sc_hd__buf_8 into sky130_fd_sc_hd__buf_8.ext:
Extracting sky130_ef_sc_hd__fakediode_2 into sky130_ef_sc_hd__fakediode_2.ext:
Extracting sky130_fd_sc_hd__fill_1 into sky130_fd_sc_hd__fill_1.ext:
Extracting sky130_fd_sc_hd__decap_3 into sky130_fd_sc_hd__decap_3.ext:
Extracting sky130_fd_sc_hd__buf_4 into sky130_fd_sc_hd__buf_4.ext:
Extracting sky130_fd_sc_hd__buf_2 into sky130_fd_sc_hd__buf_2.ext:
Extracting key_creation into key_creation.ext:
key_creation: 90477 fatal errors
Total of 90477 errors (check feedback entries).
exttospice finished.
Using technology "sky130A", version 1.0.255-0-g476f742
[ERROR]: There are illegal overlaps (e.g., routes over obstructions) in your design.
[ERROR]: See /openlane/designs/key_creation/runs/int/logs/finishing/36-ext2spice.feedback.txt for more.

```

maliberty, 2022-01-11:
You'll need to package the routing step with https://github.com/The-OpenROAD-Project/OpenLane/blob/master/docs/source/using_or_issue.md.

It would also be helpful to have a screenshot of the design open in the GUI to see what these routes look like.

antonblanchard, 2022-01-24:
Not sure if it is related, but we are chasing a similar error at https://github.com/RTimothyEdwards/open_pdks/issues/209

RTimothyEdwards, 2022-01-24:
As mentioned in the open_pdks issue cited above by @antonblanchard , I discovered that seven of the cells in the HD library have labeling of power and ground done slightly differently than all the others;  this difference, along with an error in magic when handling labels when analyzing the layout for LEF output, is causing a difference in the makeup of the VPWR and VGND pins in the LEF view of those seven cells, with the local interconnect part of the pin being assigned to a port instead of an obstruction.  Any of these seven cells abutting any of the other cells will cause an "illegal overlap" error in magic where the power and ground pins abut.  I fixed the problem in magic in version `8.3.262`, so if magic is updated and open_pdks is re-run on the HD library, then the resulting abstract views will be consistent among all of the HD standard cells, and the problem should not occur.

This, of course, depends on whether or not the issue as cited is the same as the one @antonblanchard identified, but I think it is very likely.

donnie-j, 2022-02-19:
> ... I fixed the problem in magic in version `8.3.262`, so if magic is updated and open_pdks is re-run on the HD library, then the resulting abstract views will be consistent among all of the HD [...]

Thanks @RTimothyEdwards

I can't speak to the specific cause of the issue @antonblanchard ran into, but I can confirm that I ran into this issue, and updating magic and (perhaps) rebuilding Open PDK fixes it.

RTimothyEdwards, 2022-02-20:
Good to know, thanks.

kareefardi, 2023-02-06:
This is probably fixed.

closed 2023-02-06 by kareefardi