The-OpenROAD-Project/OpenLane #866: Detailed placement fails when inserting buffers

donghl17, 2022-01-14:
### Description
I was trying to insert buffers into the design, but it dies within `detailed_placement` command. 
After the routing step, I placed several buffers into the APU using opendb commands. Then I ran the detailed placement, but it reported an error.
Is there an environment variable that can avoid this error?

### Environment
```
Python: v3.6.8                                                                                                          
Kernel: Linux v3.10.0-1160.41.1.el7.x86_64                                                                              
Distribution: centos 7                                                                                                 
 Container Engine: docker v20.10.12                                                                                      
OpenLane Git Version: 2021.12.15_04.14.23                                                                              
---                                                                                                                     
PDK Version Verification Status: OK                                                                                     
--- 
```

### Logs
```
[ERROR DPL-0019] detailed placement failed on clkbuf_leaf_7_clk                                                         
Error: insert_buffer.tcl, 176 DPL-0019
```

maliberty, 2022-01-14:
No there is no variable; you need to provide a test case.  Its certainly possible you placed too many instances in an area without enough room to legalize them.

donghl17, 2022-01-14:
I tested it on APU, and I am even not able to insert just one buffer.  Is there a way to make the original placement sparser?

maliberty, 2022-01-14:
PL_TARGET_DENSITY
see https://github.com/The-OpenROAD-Project/OpenLane/blob/master/configuration/README.md

msaligane, 2022-01-15:
@donghl17 why is there no testcase attached...

msaligane, 2022-01-15:
@maliberty I am suspecting that as well. But I think there is now a new timing optimization step in detail route which might be also contributing to this.

maliberty, 2022-01-15:
@msaligane I don't know what "new timing optimization step in detail route" is.  Would you expand or provide a pointer?

donghl17, 2022-01-16:
@maliberty We first thought the optimization step inside `resizer_timing` made the detailed placement fail. But it was actually the `ins_fill_cells` step inside routing that would fill all the free space with filler cells. https://github.com/The-OpenROAD-Project/OpenLane/blob/c01e53ac54644f2551740e28529aedd23dc12114/scripts/tcl_commands/routing.tcl#L426
That's why we were not able to insert buffers into the design. We now skip the `ins_fill_cells` step and run it after all the buffers are inserted.
[image]

donghl17, 2022-01-16:
We just pushed a new PR #868 where we fix the problem:
- change the execution order of ECO and Ins_fill_cells to avoid detailed placement errors
- maintain how the openlane flow is constructed, so ECO is transparent to openlane

This issue can be closed now.

msaligane, 2022-01-16:
Great! thanks @donghl17

closed 2022-01-16 by msaligane