The-OpenROAD-Project/OpenLane #869: Specifying FP_DEF_TEMPLATE crashes the flow.

armleo, 2022-01-17:
### Description
Specifying FP_DEF_TEMPLATE crashes the flow. Issue traced to: `apply_def_template.py`. The bug is that click converts arguments cases to lowercase, but the CLI def expects templateDEF and userDEF. Tried manually fixing this. But instead got another bug of the same nature in defutils. Didnt debug further, decided to report.

### Environment
```
Kernel: Linux v5.11.0-46-generic

Distribution: ubuntu 21.04

Python: v3.9.5 (OK)

Container Engine: docker v20.10.10 (OK)

OpenLane Git Version: 2022.01.13_01.51.43

pip:click: INSTALLED

pip:pyyaml: INSTALLED

pip:venv: INSTALLED

---

PDK Version Verification Status: OK

---

Git Log (Last 3 Commits)

c01e53a 2022-01-12T18:40:17-03:00 Fix for make pdk (#861) - Vitor Bandeira -  (HEAD -> master, tag: 2022.01.13_01.51.43, origin/master, origin/HEAD)

e33dd27 2022-01-12T20:23:47+02:00 [BOT] Update openroad_app (#818) - Openlane Bot -  ()

45cb357 2022-01-12T19:50:06+02:00 Global Routing Fixes & Enhancements (#856) - Donn -  ()
```

### Reproduction Material
Any design with specified FP_DEF_TEMPLATE crashes.
### Expected behavior
It does not crash.
### Logs
```
[INFO]: Changing layout from /openlane/designs/user_analog_project_wrapper/runs/RUN_2022.01.17_10.12.50/tmp/floorplan/3-initial_fp.def to /openlane/designs/user_analog_project_wrapper/runs/RUN_2022.01.17_10.12.50/tmp/floorplan/4-io.def

[INFO]: Applying DEF template...

[ERROR]: during executing: "openroad -python /openlane/scripts/apply_def_template.py --def-template /openlane/designs/user_analog_project_wrapper/macros/def/user_analog_project_wrapper.def /openlane/designs/user_analog_project_wrapper/runs/RUN_2022.01.17_10.12.50/tmp/floorplan/4-io.def"

[ERROR]: Exit code: 1

[ERROR]: Last 10 lines:

    cli()

  File "/usr/local/lib/python3.6/site-packages/click/core.py", line 1128, in __call__

    return self.main(*args, **kwargs)

  File "/usr/local/lib/python3.6/site-packages/click/core.py", line 1053, in main

    rv = self.invoke(ctx)

  File "/usr/local/lib/python3.6/site-packages/click/core.py", line 1395, in invoke

    return ctx.invoke(self.callback, **ctx.params)

  File "/usr/local/lib/python3.6/site-packages/click/core.py", line 754, in invoke

    return __callback(*args, **kwargs)

TypeError: cli() got an unexpected keyword argument 'userdef'

[INFO]: Calculating Runtime From the Start...

[INFO]: Generating Final Summary Report...

[INFO]: Design Name: user_analog_project_wrapper

Run Directory: /openlane/designs/user_analog_project_wrapper/runs/RUN_2022.01.17_10.12.50

Source not found.

----------------------------------------

LVS Summary:

Source: /openlane/designs/user_analog_project_wrapper/runs/RUN_2022.01.17_10.12.50/logs/finishing/user_analog_project_wrapper.lvs.lef.log

Source not found.

----------------------------------------

Antenna Summary:

No antenna report found.

[INFO]: check full report here: /openlane/designs/user_analog_project_wrapper/runs/RUN_2022.01.17_10.12.50/reports/final_summary_report.csv

[INFO]: Saving runtime environment...

[ERROR]: Flow failed.

    while executing

"flow_fail"

    (procedure "try_catch" line 17)

    invoked from within

"try_catch $::env(OPENROAD_BIN) -python $::env(SCRIPTS_DIR)/apply_def_template.py --def-template $::env(FP_DEF_TEMPLATE) $::env(CURRENT_DEF)"

    (procedure "apply_def_template" line 4)

    invoked from within

"apply_def_template"

    (procedure "run_floorplan" line 39)

    invoked from within

"[lindex $step_exe 0] [lindex $step_exe 1] "

    (procedure "run_non_interactive_mode" line 54)

    invoked from within

"run_non_interactive_mode {*}$argv"

    invoked from within

"if { [info exists flags_map(-interactive)] || [info exists flags_map(-it)] } {

	puts_info "Running interactively"

	if { [info exists arg_values(-file)..."

    (file "./flow.tcl" line 387)

```

armleo, 2022-01-18:
UPD: Last issue was caused by out of memory. So only two above mentioned bugs exist right now. Hopefully will make a .def based design that can be added to list of designs to properly test this kind of bugs.

armleo, 2022-01-22:
Looks like #864 by @donn fixes this issue, can the pull request be merged?

donn, 2022-01-22:
Not yet, no. It's not ready. I'll push a more granular fix for this on Monday.

armleo, 2022-01-22:
New update: Looks like the first issue, with userDEF and templateDEF is not fixed by the pull request. IDK why it's not reprudicble in the CI, I assume, because of Click version mismatch?

closed 2022-02-01 by armleo