The-OpenROAD-Project/OpenLane #870: Fan out issues with muxes uncovers various bugs

antonblanchard, 2022-01-18:
This all began when I started to wonder why I had large fan outs on a simple register file:

```
module Microwatt_FP_DFFRFile (
`ifdef USE_POWER_PINS
    inout VPWR,
    inout VGND,
`endif
    input [6:0]   R1, R2, R3, RW,
    input [63:0]  DW,
    output [63:0] D1, D2, D3,
    input CLK,
    input WE
);

    reg [63:0] registers[0:79];

    assign D1 = registers[R1];
    assign D2 = registers[R2];
    assign D3 = registers[R3];

    always @(posedge CLK) begin
        if (WE)
            registers[RW] <= DW;
    end

endmodule
```

```
Fanout     Cap    Slew   Delay    Time   Description
-----------------------------------------------------------------------------
                          0.00    0.00   clock CLK (rise edge)
                          0.00    0.00   clock network delay (propagated)
                          4.00    4.00 ^ input external delay
                  0.05    0.04    4.04 ^ R1[0] (in)
     1    0.01                           R1[0] (net)
                  0.05    0.00    4.04 ^ input65/A (sky130_fd_sc_hd__buf_12)
                  0.28    0.25    4.28 ^ input65/X (sky130_fd_sc_hd__buf_12)
    30    0.25                           net65 (net)
                  0.28    0.02    4.30 ^ repeater550/A (sky130_fd_sc_hd__buf_12)
                  0.32    0.34    4.64 ^ repeater550/X (sky130_fd_sc_hd__buf_12)
    31    0.28                           net550 (net)
                  0.32    0.02    4.66 ^ repeater545/A (sky130_fd_sc_hd__buf_12)
                  0.41    0.34    5.01 ^ repeater545/X (sky130_fd_sc_hd__buf_12)
    42    0.37                           net545 (net)
                  0.43    0.08    5.08 ^ repeater544/A (sky130_fd_sc_hd__buf_12)
                  0.38    0.37    5.46 ^ repeater544/X (sky130_fd_sc_hd__buf_12)
    40    0.34                           net544 (net)
                  0.40    0.06    5.52 ^ repeater543/A (sky130_fd_sc_hd__buf_12)
                  0.28    0.30    5.82 ^ repeater543/X (sky130_fd_sc_hd__buf_12)
    27    0.25                           net543 (net)
                  0.31    0.07    5.89 ^ repeater542/A (sky130_fd_sc_hd__buf_12)
                  0.45    0.40    6.29 ^ repeater542/X (sky130_fd_sc_hd__buf_12)
    46    0.40                           net542 (net)
                  0.46    0.06    6.35 ^ repeater541/A (sky130_fd_sc_hd__buf_12)
                  0.46    0.39    6.74 ^ repeater541/X (sky130_fd_sc_hd__buf_12)
    50    0.41                           net541 (net)
                  0.55    0.17    6.91 ^ repeater540/A (sky130_fd_sc_hd__buf_12)
                  0.30    0.34    7.24 ^ repeater540/X (sky130_fd_sc_hd__buf_12)
    31    0.27                           net540 (net)
                  0.33    0.07    7.32 ^ repeater538/A (sky130_fd_sc_hd__buf_12)
                  0.35    0.34    7.66 ^ repeater538/X (sky130_fd_sc_hd__buf_12)
    40    0.32                           net538 (net)
                  0.38    0.07    7.73 ^ repeater537/A (sky130_fd_sc_hd__buf_12)
                  0.33    0.32    8.05 ^ repeater537/X (sky130_fd_sc_hd__buf_12)
    37    0.29                           net537 (net)
                  0.33    0.04    8.09 ^ repeater536/A (sky130_fd_sc_hd__buf_12)
                  0.31    0.31    8.41 ^ repeater536/X (sky130_fd_sc_hd__buf_12)
    34    0.28                           net536 (net)
                  0.35    0.08    8.49 ^ _25097_/S0 (sky130_fd_sc_hd__mux4_1)
                  0.09    0.64    9.13 v _25097_/X (sky130_fd_sc_hd__mux4_1)
     1    0.00                           _01082_ (net)
```

The potential amount of fan out is enormous (1x10^17?). We should have been able to handle this with a few stages of fan out, not 11.

The first thing I noticed is that yosys buffering doesn't work with manually tech mapped instances (eg our muxes in Openlane). We ask yosys for a max fanout of `SYNTH_MAX_FANOUT` but on the inputs of muxes we end up with quite large fanouts (eg 1280) that subsequent steps have to sort out. This looks like a yosys or abc bug, and I'll follow up with them.

It does however highlight our first issue in Openlane - we are manually tech mapping 2 and 4 input muxes, when abc can do it for us. Removing the manual tech map, and adding something like `&get,-n;&st;&dch;&nf;&put;` to the abc args should fix it. This is in fact what ORFS does. Fixing this not only fixes the above fan out issue, but also means we can map the inverting output 2 input mux (`mux2i`).

I tested that fix, and I see much better results. `SYNTH_MAX_FANOUT` is set to 10, and see 4 levels of fan out, and timing is much better:

```
Fanout     Cap    Slew   Delay    Time   Description
-----------------------------------------------------------------------------
                          0.00    0.00   clock CLK (rise edge)
                          0.00    0.00   clock network delay (propagated)
                          4.00    4.00 ^ input external delay
                  0.02    0.01    4.01 ^ R2[0] (in)
     1    0.00                           R2[0] (net)
                  0.02    0.00    4.01 ^ input72/A (sky130_fd_sc_hd__buf_4)
                  0.21    0.22    4.24 ^ input72/X (sky130_fd_sc_hd__buf_4)
     1    0.07                           net72 (net)
                  0.21    0.02    4.26 ^ _22074_/A (sky130_fd_sc_hd__buf_8)
                  0.18    0.25    4.50 ^ _22074_/X (sky130_fd_sc_hd__buf_8)
     9    0.11                           _05122_ (net)
                  0.18    0.01    4.51 ^ _22075_/A (sky130_fd_sc_hd__buf_8)
                  0.18    0.24    4.76 ^ _22075_/X (sky130_fd_sc_hd__buf_8)
    10    0.11                           _05123_ (net)
                  0.18    0.00    4.76 ^ _22103_/A (sky130_fd_sc_hd__buf_12)
                  0.40    0.37    5.13 ^ _22103_/X (sky130_fd_sc_hd__buf_12)
    10    0.37                           _05151_ (net)
                  0.47    0.13    5.26 ^ _22495_/A (sky130_fd_sc_hd__buf_12)
                  0.21    0.31    5.56 ^ _22495_/X (sky130_fd_sc_hd__buf_12)
    10    0.18                           _05538_ (net)
                  0.21    0.01    5.57 ^ _22935_/S0 (sky130_fd_sc_hd__mux4_1)
                  0.09    0.62    6.19 v _22935_/X (sky130_fd_sc_hd__mux4_1)
     1    0.00                           _05970_ (net)
                  0.09    0.00    6.19 v _22937_/A0 (sky130_fd_sc_hd__mux2_1)
                  0.11    0.38    6.57 v _22937_/X (sky130_fd_sc_hd__mux2_1)
     1    0.01                           _05972_ (net)
                  0.11    0.00    6.57 v _22942_/A2 (sky130_fd_sc_hd__mux4_2)
                  0.21    0.69    7.26 v _22942_/X (sky130_fd_sc_hd__mux4_2)
     1    0.06                           _05977_ (net)
                  0.21    0.01    7.27 v _22963_/A0 (sky130_fd_sc_hd__mux2_8)
                  0.14    0.50    7.77 v _22963_/X (sky130_fd_sc_hd__mux2_8)
     1    0.13                           _05998_ (net)
                  0.14    0.01    7.78 v _22980_/A2 (sky130_fd_sc_hd__o21ba_4)
                  0.14    0.35    8.13 v _22980_/X (sky130_fd_sc_hd__o21ba_4)
     1    0.09                           net162 (net)
                  0.14    0.01    8.14 v output162/A (sky130_fd_sc_hd__buf_2)
                  0.09    0.23    8.37 v output162/X (sky130_fd_sc_hd__buf_2)
     1    0.03                           D2[13] (net)
                  0.09    0.00    8.37 v D2[13] (out)
```

@maliberty this brings up a question for ORFS - there is no configurability for abc fan out, so you get the default of 10. I'm not sure if that is an optimal number for all technologies. I'm not even sure if the intent is for yosys to do fan out modifications, or to leave that for OpenROAD.

We also should ask why the subsequent stages of OpenROAD didn't fix it as well as yosys did. Can the resizer insert trees of buffers? I guess the resizer is being driven by max capacitance and max slew, so the sky130hd liberty file should be looked at too.

antonblanchard, 2022-01-18:
> We also should ask why the subsequent stages of OpenROAD didn't fix it as well as yosys did. Can the resizer insert trees of buffers? I guess the resizer is being driven by max capacitance and max slew, so the sky130hd liberty file should be looked at too.

Another thought - my timing constraints might need to be tightened to force the resizer to work harder.

maliberty, 2022-01-18:
The rsz will buffer large nets but not necessarily optimally.  It doesn't insert buffer trees but just tries to fix violations which will result in a bunch of buffering.

donn, 2022-01-23:
I'm... not really sure what's actionable here

antonblanchard, 2022-01-24:
I opened https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/issues/323 to track the ORFS suggestion.

antonblanchard, 2022-01-24:
@donn I've now opened bugs against both yosys and ORFS, and we've fixed the mux tech map issue, so I think we can close this.

closed 2022-01-24 by antonblanchard

donn, 2022-01-27:
Hey anton, mind telling us where you got the commands from? We're curious.

antonblanchard, 2022-01-27:
I pulled it out of ORFS:

https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/blob/master/flow/scripts/abc_speed.script

We should probably talk to the abc maintainer. There's a boatload of different parameters in the Openroad abc script (I wonder where they came from), as well as the fan out issue with tech mapped things we uncovered recently.