The-OpenROAD-Project/OpenLane #879: Small die areas crashes the flow

armleo, 2022-01-22:
### Description
Setting a small die area crashes the flow, as the minimized PDN has pitch smaller than 2 x (width + spacing). Maybe a relevant warning can be added? An error when the run is just starting stating that the die area too small, should do fine. Perfectly a fix of this.

### Environment
```
Kernel: Linux v5.11.0-49-generic

Distribution: ubuntu 21.04

Python: v3.9.5 (OK)

Container Engine: docker v20.10.10 (OK)

OpenLane Git Version: 2022.01.13_01.51.43

pip:click: INSTALLED

pip:pyyaml: INSTALLED

pip:venv: INSTALLED

---

PDK Version Verification Status: OK

---

Git Log (Last 3 Commits)

8ad16ce 2022-01-20T13:08:19+02:00 Move reading SDC file earlier (#871) - Anton Blanchard -  (HEAD -> master, origin/master, origin/HEAD)

6871aa8 2022-01-20T13:07:26+02:00 Remove manual tech mapping of muxes and get abc to map them (#873) - Anton Blanchard -  ()

c01e53a 2022-01-12T18:40:17-03:00 Fix for make pdk (#861) - Vitor Bandeira -  (tag: 2022.01.13_01.51.43)
```

### Reproduction Material
[small_die_test.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/7918999/small_die_test.zip)
### Expected behavior

Flow not crashing, or crashing with a warning about DIE AREA being too small.

### Logs
```
[ERROR PDN-0085] Pitch 6.079999999999998 specified for layer met5 is less than 2 x (width + spacing) (width=1.6, spacing=3.04).

Error: pdn_cfg.tcl, 39 PDN-0085

```

donn, 2022-01-23:
We'll see. Convenience issues are not really high priority given that there's one of me.

kareefardi, 2024-07-14:
Adding OL2 tag to this as such an enhancement is not needed with OL2 since with the ease of rerunning steps.