The-OpenROAD-Project/OpenLane #881: [Idea, Q&A] Add issue regression tests

armleo, 2022-01-22:
### Prompt
Currently only perfomance benchmarks are done and no compatability/feature tests are done.

For example

1. if you try to run a design wtih FP_DEF_TEMPLATE specified, it crashes, see: https://github.com/The-OpenROAD-Project/OpenLane/issues/869.

2. There is also other issues caused by a lot of reworks in the python scripts like defutil.py.

3. There is cases where small die areas crash the flow (see: https://github.com/The-OpenROAD-Project/OpenLane/issues/879)

4. Hi-Z inouts are connected to tie-low cell and it's impossible to disconnect it.

5. Specifying USE ANALOG in LEF should force the router to ignore the pin, however the flow crashes stating that USE ANALOG is an invalid.

6. Placing an LEF cell which contains met3 causes the flow to place met3 blockages over the macro (I assume at least), which in turn causes some met3 pins to be inaccessible.

This is just a couple I noticed, probably there is more.
### Proposal
My proposal is to add a set of tests that cover issues that happened in the past. This will help to reduce the gap between 100% coverage of all options and current state of OpenLane.

I am open to discussion, and if OpenLane devs arent against it, I can even implement it with integration with current CI flow.

armleo, 2022-01-22:
For example: Specifying the def currently causes the flow to not crash (with applied relevant fixes to above mentioned issues), however the floorplan's pin and macro locations dont match the specified def. This was not the case with v0.23.

If I added my def_test as a part of bug regression testing, this would allow me to create a test script which compares the flows reports and defs/gds to what should have been in the output, but only checking for the relevant parts, like pin/macro locations.

Obviously this creates a little more work, but this will reduce the amount of issues at the end of the day and make OpenLane more production-ready.

donn, 2022-01-23:
100% agreed. If you can get something implemented, please, by all means.

armleo, 2022-01-23:
Let me know what you think about my idea of implementing this.

Modify flow.tcl to source checks/post_run_hooks.tcl which will do the checking. This will allow to add a just any forms of checks in the flow, without introducing too much new code.

Then for issues, we will have minimal designs which reproduce this issues and checks can be added to checks/post_run_hooks.tcl that make sure that issue that we had with the design is didn't comeback as result of code changes.

For example: To test issue #869 I would have run with FP_DEF_TEMPLATE specified. The run will be called def_test and will contain post_run_hooks.tcl which will compare pins from a known-to-be-good def to pin locations from final def file. For other issues related to FP_DEF_TEMPLATE other tests might be introduced to post_run_hooks.tcl, while not adding new designs.

For issues that require a design like Hi-Z inout issue, a designs/issue_878_disconnected_outputs_shouldnt_be_tied_low and will contain the hooks + short readme explaining the issue and giving links and other info.

donn, 2022-01-23:
Sober, but perhaps hide this feature behind a flag. For security purposes. The `config.tcl` is already enough a security headache as is.

armleo, 2022-01-23:
Following does not generate an stdout, coudl you help me debug it?

exec $::env(OPENROAD_BIN) -python $::env(DESIGN_DIR)/hooks/post_run.py

armleo, 2022-01-24:
Currently I don't know how to fill the regression_tests csvs, so my def_test design will be there and will not crash the flow.

donn, 2022-02-28:
Thank you for the issue and thank you for contributing. Most users don't get this far. :)

closed 2022-02-28 by donn