The-OpenROAD-Project/OpenLane #884: Manual macro placement for a module other than top level

microSharjeel, 2022-01-26:
### Prompt
Manual macro placement only works when the module that is black box is in the top level of the rtl design. I have design in which I am supposed to place macro for SRAM and that SRAM is deep in the design and not in the top level. OpenLane gives an error (Macro not found when the black box to be place is not in the top level). Is there any solution for this or if this can be enhanced in OpenLane

### Proposal
A clear and concise description of what you want to happen.

vijayank88, 2022-01-26:
@microSharjeel  
Please provide issue_survey and reproducible tar file

armleo, 2022-01-26:
Had the same issue. Was caused by misconfiguration and Verilog RTL issue.

microSharjeel, 2022-01-26:
@armleo  if you solved it ? if so then can you help ?

armleo, 2022-01-26:
My issue was that the sram was not recognized as blackbox. Check synthesis verilog, it should have the balckbox cell.

Check the name too. I made an mistake where the name was incorrect. Check the floorplan def + synthesis netlist for that one.

microSharjeel, 2022-01-26:
@armleo  Also idk, how to properly escape the name. This might be most likely cause.> ? can not understand 
I want to place Macro from openRAM. I am using the .lef and .gds files from OpenRAM and the name that I use for blackbox is the same name that is there in the .LEF file

microSharjeel, 2022-01-26:
@armleo  Have  you placed macro in your design at a level other than top level ? If this is the exact scenario then I would like to know about your config.tcl file if you can share that

armleo, 2022-01-26:
Unfortunately I cant share it.

The most likely cause is that the name of macro does not exactly match the name in sybthesised netlist. If the synthesis netlist has hierarchy, then the name of the cell has to be escaped according to hierarchy.

microSharjeel, 2022-01-26:
[image]

microSharjeel, 2022-01-26:
The name of the macro is in highlight 
sram_32_128_sky130A

microSharjeel, 2022-01-26:
I have used the hierarchical name from the synthesized netlist. I want to write this for anyone who wants to place macro in RTL design at a level other than top level. In this case the hierarchical name of the macro should be used. This issue is now resolved and i will close it 
@armleo  Thanks a lot for help

closed 2022-01-26 by microSharjeel