The-OpenROAD-Project/OpenLane #892: DEF pin locations are not applied [Regression, Bug]

armleo, 2022-01-27:
### Description
DEF pins locations are not used FP_DEF_TEMPLATE. Also the location of the component is not used from DEF. This might cause confusion, as FP_DEF_TEMPLATE's names assume that components locations are also used, but there is no proper documentation regarding the priority over manual macro placement.

Keep in mind #891 DOES NOT fix this. It just resurfaces it. If you use my branch you will see the issue if you run `./flow.tcl -design def_test -run_hooks`. It will run the def_test and the generated locations of the pins will not match the DEF.
### Environment
```
Kernel: Linux v5.11.0-49-generic

Distribution: ubuntu 21.04

Python: v3.9.5 (OK)

Container Engine: docker v20.10.10 (OK)

OpenLane Git Version: issue_regression_flow-dev

pip:click: INSTALLED

pip:pyyaml: INSTALLED

pip:venv: INSTALLED

---

PDK Version Verification Status: OK

---

Git Log (Last 3 Commits)

11bb94c 2022-01-28T01:00:25+04:00 Fixed flake8 complaining about tech variable in def_test/hooks/post_run.py - armleo -  (HEAD -> issue_regression_flow, origin/master, origin/HEAD)

dc5a5c2 2022-01-28T00:57:23+04:00 Fixed formatting - armleo -  ()

76273b6 2022-01-28T00:50:21+04:00 Fixed formatting - armleo -  ()
```

### Reproduction Material
See pull request above.
### Expected behavior
DEF's pin locations should be used properly. In v0.23 this was done properly def_util.py merge broke this. Or it might have happened in the past but resurfaced right now.
### Logs
Not relevant

armleo, 2022-01-28:
I am trying to fix defutil.py

Does anybody know how to define new Pins using opendb? Or maybe any documentation regarding the python API of OpenDB?

armleo, 2022-02-01:
Figured out, the integration tests show how its done, and the API is the same in python.

Currently trying to copy all the pins from one def to another, while filtering the ones that are in the first def.

armleo, 2022-02-28:
Currently with the Pull request accepted this has been partially resolved. However multiple pins are still not handled properly. Please don't close this, so I can fix this.

donn, 2022-02-28:
I have no intention to, don't worry.

armleo, 2022-04-15:
I am no longer working on the fix for this as I have no time.

donn, 2022-04-16:
That's totally fine! Thank you for telling me and not leaving us hanging.

maliberty, 2022-06-23:
@donn why don't you just use the pin placer in openroad?

donn, 2022-06-23:
@maliberty This is for a specific scenario where the DEF pin placement has to match another DEF file 1:1. Can the OR pin placer do that?

armleo, 2022-06-23:
Issue: Need to use def file that has pins and locations/layers of the pins
solution in v0.23: copied content of the pins section
problem: The pin might or might not have a net attached to it and the net was copied too, causing strange issues.
In reqork def util.py before my fix: The specifying the def template did not work at all.
After my pull request: The pins work, as long as they are not VIAs, rectangular and singular. Multiple pins with the same name are improperly handled.

donn, 2022-06-23:
Yeah, that was a mishap on my part. I meant #927.

maliberty, 2022-06-23:
If you know exactly where you want the pins to go you can use place_pin  for manual placement.

donn, 2022-06-23:
We'd have to write a script that extracts the pin location then concatenate a `place_pin` Tcl script instead of copying over the pin section from another DEF file. Why?

donn, 2022-06-23:
Eh, I'm good. I'll just add a "contaminated database" flag for that scenario.

donn, 2022-08-24:
Think this was fixed by #1160.

closed 2022-08-24 by donn

armleo, 2022-08-24:
The issue was open because of "multiple terminals are still not handled properly".

Do you want to reopen it or create new one regarding this? @donn

donn, 2022-08-24:
Create a new one if you'd please.