The-OpenROAD-Project/OpenLane #893: spm design fails during Synthesis ABC mapping with set ::env(STD_CELL_LIBRARY) "sky130_fd_sc_hvl"

vijayank88, 2022-01-28:
### Description
spm design fails if I use `set ::env(STD_CELL_LIBRARY) "sky130_fd_sc_hvl"` in config.tcl

### Environment
```
Kernel: Linux v3.10.0-1160.49.1.el7.x86_64
Distribution: centos 7
Python: v3.6.8 (OK)
Critical Alert: No Docker or Docker-compatible container engine was found: {e}.
OpenLane Git Version: 2022.01.28_02.18.21
pip:click: INSTALLED
pip:pyyaml: INSTALLED
pip:venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

4db7f91 2022-01-27T22:13:26+02:00 Update OpenROAD to HEAD (#890) - Donn -  (HEAD -> master, tag: 2022.01.28_02.18.21, origin/master, origin/HEAD)
bff0186 2022-01-27T18:31:54+02:00 Some QoL changes to PDK Installation Make Targets (#888) - Donn -  ()
d974919 2022-01-27T16:33:44+02:00 Remove endpoint dependency from base image (#887) - Vitor Bandeira -  ()
 
```

### Reproduction Material
* spm built-in design: https://github.com/The-OpenROAD-Project/OpenLane/tree/master/designs/spm
Update config.tcl as follows
```
# Design
set ::env(DESIGN_NAME) "spm"

set ::env(VERILOG_FILES) [glob $::env(DESIGN_DIR)/src/*.v]

set ::env(CLOCK_PERIOD) "10.000"
set ::env(CLOCK_PORT) "clk"

set ::env(FP_PIN_ORDER_CFG) $::env(DESIGN_DIR)/pin_order.cfg

set ::env(STD_CELL_LIBRARY) "sky130_fd_sc_hvl"

set filename ./designs/$::env(DESIGN_NAME)/$::env(PDK)_$::env(STD_CELL_LIBRARY)_config.tcl
if { [file exists $filename] == 1} {
	source $filename
}

```
Inside OpenLane Directory:
$make mount
$./flow.tcl -design spm -tag hvlTest

### Expected behavior
Successful flow complete

### Logs
```
17. Executing ABC pass (technology mapping using ABC).

17.1. Extracting gate netlist of module `\spm' to `/tmp/yosys-abc-gNDG8t/input.blif'..
Extracted 253 gates and 350 wires to a netlist network with 97 inputs and 128 outputs.

17.1.1. Executing ABC.
ERROR: Can't open ABC output file `/tmp/yosys-abc-gNDG8t/output.blif'.
[ERROR]: during executing: "yosys -c /openlane/scripts/yosys/synth.tcl -l /openlane/designs/spm/runs/RUN_2022.01.28_06.02.00/logs/synthesis/1-synthesis.log |& tee >&@stdout"
[ERROR]: Exit code: 1
[ERROR]: Last 10 lines:
child process exited abnormally

[INFO]: Calculating Runtime From the Start...
[INFO]: Generating Final Summary Report...
[INFO]: Design Name: spm
Run Directory: /openlane/designs/spm/runs/RUN_2022.01.28_06.02.00
Source not found.
----------------------------------------

LVS Summary:
Source: /openlane/designs/spm/runs/RUN_2022.01.28_06.02.00/logs/finishing/spm.lvs.lef.log
Source not found.
----------------------------------------

Antenna Summary:
No antenna report found.
[INFO]: check full report here: /openlane/designs/spm/runs/RUN_2022.01.28_06.02.00/reports/final_summary_report.csv
[INFO]: Saving runtime environment...
[ERROR]: Flow failed.

    while executing
"flow_fail"
    (procedure "try_catch" line 17)
    invoked from within
"try_catch $::env(SYNTH_BIN)  -c $::env(SYNTH_SCRIPT)  -l [index_file $::env(synthesis_logs)/synthesis.log]  |& tee $::env(TERMINAL_OUTPUT)"
    (procedure "run_yosys" line 32)
    invoked from within
"run_yosys"
    (procedure "run_synthesis" line 11)
    invoked from within
"[lindex $step_exe 0] [lindex $step_exe 1] "
    (procedure "run_non_interactive_mode" line 51)
    invoked from within
"run_non_interactive_mode {*}$argv"
    invoked from within
"if { [info exists flags_map(-interactive)] || [info exists flags_map(-it)] } {
	puts_info "Running interactively"
	if { [info exists arg_values(-file)..."
    (file "./flow.tcl" line 384)

```

donn, 2022-09-06:
Can you try this again? Now that Yosys has been updated I mean.

vijayank88, 2022-09-06:
I have updated `config.json` as follows,
```
{
    "DESIGN_NAME": "spm",
    "VERILOG_FILES": "dir::src/*.v",
    "CLOCK_PERIOD": 10,
    "CLOCK_PORT": "clk",
    "CLOCK_NET": "ref::$CLOCK_PORT",
    "FP_PDN_VOFFSET": 7,
    "FP_PDN_HOFFSET": 7,
    "FP_PIN_ORDER_CFG": "dir::pin_order.cfg",
    "STD_CELL_LIBRARY": "sky130_fd_sc_hvl",
    "pdk::sky130*": {
        "FP_CORE_UTIL": 45,
        "scl::sky130_fd_sc_hd": {
            "CLOCK_PERIOD": 10
        },
        "scl::sky130_fd_sc_hdll": {
            "CLOCK_PERIOD": 10
        },
        "scl::sky130_fd_sc_hs": {
            "CLOCK_PERIOD": 8
        },
        "scl::sky130_fd_sc_ls": {
            "CLOCK_PERIOD": 10,
            "SYNTH_MAX_FANOUT": 5
        },
        "scl::sky130_fd_sc_ms": {
            "CLOCK_PERIOD": 10
        }
    },
    "pdk::gf180mcuC": {
        "CLOCK_PERIOD": 24.0,
        "FP_CORE_UTIL": 40,
        "SYNTH_MAX_FANOUT": 4,
        "PL_TARGET_DENSITY": 0.45
    }
}
```
Got following error:
```
OpenLane Container (d609968):/openlane$ ./flow.tcl -design spm -tag hvl_test -overwrite
OpenLane d609968cca9ebc6aae6ba8ada9c16b1ed3ce4d0a
All rights reserved. (c) 2020-2022 Efabless Corporation and contributors.
Available under the Apache License, version 2.0. See the LICENSE file for more details.

[INFO]: Using configuration in 'designs/spm/config.json'...
[INFO]: PDK Root: /home/vijayan/OPENROAD_FLOW/OLFS/sky130_OL/OpenLane/pdks
[INFO]: Process Design Kit: sky130A
[INFO]: Standard Cell Library: sky130_fd_sc_hvl
[INFO]: Optimization Standard Cell Library: sky130_fd_sc_hd
[INFO]: Run Directory: /openlane/designs/spm/runs/hvl_test
[INFO]: Removing existing /openlane/designs/spm/runs/hvl_test...
[INFO]: Preparing LEF files for the nom corner...
[INFO]: Preparing LEF files for the min corner...
[INFO]: Preparing LEF files for the max corner...
no such variable
    (read trace on "::env(LIB_SYNTH_OPT)")
    invoked from within
"foreach lib $::env(LIB_SYNTH_OPT) {
                    set fbasename [file rootname [file tail $lib]]
                    set lib_resizer $::env(synt..."
    (procedure "prep" line 329)
    invoked from within
"prep {*}$args"
    (procedure "run_non_interactive_mode" line 12)
    invoked from within
"run_non_interactive_mode {*}$argv"
    invoked from within
"if { [info exists flags_map(-interactive)] || [info exists flags_map(-it)] } {
    if { [info exists arg_values(-file)] } {
        run_file [file nor..."
    (file "./flow.tcl" line 401)
```
Is anything to update `.json` file?

donn, 2022-09-06:
No, this is another bug.

kareefardi, 2023-09-10:
This is no longer the case using OpenLane `7e5a2e9fb274c0a100b4859a927adce7089455ff`. Please open a new issue if you run into other problems with `sky130_fd_sc_hvl`

closed 2023-09-10 by kareefardi