The-OpenROAD-Project/OpenLane #901: Netlists Do Not Match

mousaq92, 2022-02-03:
Hello,

I am trying to run the openlane flow on a small design using the Skywater PDK. I get a "Netlists do not match" error. How can I resolve this issue?

Thanks,

Mousa

armleo, 2022-02-03:
More information is required. Please follow the issue template. We can't help because: We dont have any information about you design, tool versions, openlane version, etc

mousaq92, 2022-02-03:
Thank you for your reply. There are LVS errors in the design according to Netgen LVS. I am not sure where to find the issue template. I'm assuming that this issue is because of device merging. It is unable to recognize some of the devices in Sky130. I pulled openlane this week.

armleo, 2022-02-03:
When you click on "new issue" you can see two options. Select the one matching the bug. In the created window it will have title / description. Fill out all the information in description including your tarball containing your design.

d-m-bailey, 2022-02-04:
@mousaq92 can you post your LVS results file or maybe your repo so we can duplicate the results?

dralabeing, 2022-02-04:
@mousaq92 Please resubmit your testcase based on the required input template format

donn, 2022-02-09:
You're all very diplomatic. Here's how I deal with issues that ignore the template:

closed 2022-02-09 by donn