The-OpenROAD-Project/OpenLane #906: sky130hs fails on diode error

maliberty, 2022-02-07:
### Description
If I run gcd with sky130hs (using #905 ) I get:

```
[WARNING STA-0328] port 'DIODE' not found.
[ERROR GRT-0069] Diode not found.
Error: groute.tcl, 74 GRT-0069

```
### Environment
Head of master

### Reproduction Material
In description / GH

### Expected behavior
No error

### Logs

tdene, 2022-02-07:
I believe this is caused by [https://github.com/google/skywater-pdk/issues/358](https://github.com/google/skywater-pdk/issues/358)

kareefardi, 2023-09-10:
For some reason this no longer happens. Double checking if I missed something.

kareefardi, 2023-09-12:
This is probably handled at OpenROAD's end since the mentioned issue in this thread was not fixed. Closing this if similar issues appear please open a new issue.

closed 2023-09-12 by kareefardi