The-OpenROAD-Project/OpenLane #907: Add "quick_run" recipe to Makefile to allow for easy outside scripting

tdene, 2022-02-07:
### Prompt
The OpenLane flow works well from inside its Docker container.

A valuable use-case, however, is running OpenLane via script, for example on procedurally-generated designs.

The current way of running OpenLane on a design is:

`make mount`
`./flow.tcl -design X`

This can be simplified down to a single Makefile recipe.

At the same time, as far as the submitter of this issue can see, there is currently no way to change the default STD_CELL_LIBRARY settings upon mounting the Docker container. This functionality too can be provided with minimal changes to OpenLane.

### Proposal

Modify the Makefile as follows:

 * Add the following kind of recipe:
<code>
+.PHONY: adder
quick_run:
       cd $(OPENLANE_DIR) && \
               $(ENV_COMMAND) sh -c "./flow.tcl -design $(QUICK_RUN_DESIGN)"
</code>

 * Change line 49 to the following:
<code>
TEST_DESIGN ?= spm
QUICK_RUN_DESIGN ?= spm
</code>

 * Change lines 63-65 to
<code>
export PDK_ROOT ?= ./pdks
export PDK_ROOT := $(shell python3 -c "import os; print(os.path.realpath('$(PDK_ROOT)'), end='')")
export STD_CELL_LIBRARY ?= sky130_fd_sc_hd
PDK_OPTS = -v $(PDK_ROOT):$(PDK_ROOT) -e PDK_ROOT=$(PDK_ROOT) -e STD_CELL_LIBRARY=$(STD_CELL_LIBRARY)
</code>

mithro, 2022-02-07:
FYI - @proppy @ethanmoon

QuantamHD, 2022-02-07:
@mithro I don't own @EthanMoon

donn, 2022-02-09:
Good changes, can you please put them in a PR?

tdene, 2022-02-09:
Thanks @donn !
See #910

commit 0c06531c (2022-02-13): Add quick_design recipe to Makefile (#910)

This single-purpose recipe mounts the docker container and runs a desired
design through the flow. Provides better support for running OpenLane via
script.

Resolves #907.