The-OpenROAD-Project/OpenLane #909: [mpw5a] OpenLane global router segfaults

armleo, 2022-02-09:
### Description
I don't really know why but Global Router is crashing if there is a pin in top level verilog that has the same name as the pin in .def. 

Logically if the pin does not exist in .def then it has to be placed, and if it exists in .def but not on top level, it should issue a warning. However in this edge case, we specify in .def and in top level a pin that matches power/ground. Generated sdc contains get_ports VPWR, which I assume crashes the global router?

### Environment
```
Python: v3.9.5

Kernel: Linux v5.11.0-49-generic

Distribution: ubuntu 21.04

Container Engine: docker v20.10.10

OpenLane Git Version: 2021.11.23_01.42.34

---

PDK Version Verification Status: OK

---

Git Log (Last 3 Commits)

commit e1883edfbcc8c7e96491158521522b7a66541745

Author: Manar <manarabdelatty@aucegypt.edu>

Date:   Mon Nov 22 07:59:18 2021 -0600

    Update resizer (#726)

commit 6245b0e675d0e6751bdfc5362c545efe29d1077c

Author: Manar <manarabdelatty@aucegypt.edu>

Date:   Fri Nov 19 13:38:50 2021 -0600

    Fix sram install (#724)

    

    Also disable by default

commit e0e6fbc1bd90d1c73aae19769d997da8c3c47512

Author: Manar <manarabdelatty@aucegypt.edu>

Date:   Fri Nov 19 12:30:08 2021 -0600

    Add option for enabling/disabling repair_tie_fanout (#722)

    

    * Add option for enabling/disabling repair_tie_fanout

    

    * add escape characters

    

    * Add option for enabling sta report power

```

### Reproduction Material
[inverter_def_power_pin_segfault.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/8029468/inverter_def_power_pin_segfault.zip)

### Expected behavior
Not segfaulting.

### Logs
```
[ERROR]: during executing: "openroad -exit /openlane/scripts/openroad/groute.tcl |& tee >&@stdout /openlane/designs/inverter_def_power_pin_segfault/runs/RUN_2022.02.09_04.44.48/logs/routing/15-fastroute.log"

[ERROR]: Exit code: 1

[ERROR]: Last 10 lines:

child killed: segmentation violation

[ERROR]: Please check openroad  log file

[ERROR]: Dumping to /openlane/designs/inverter_def_power_pin_segfault/runs/RUN_2022.02.09_04.44.48/error.log

[INFO]: Calculating Runtime From the Start...

[INFO]: flow failed for inverter/2022.02.09_04.44.48 in 0h0m49s

[INFO]: Generating Final Summary Report...

[INFO]: Design Name: inverter

Run Directory: /openlane/designs/inverter_def_power_pin_segfault/runs/RUN_2022.02.09_04.44.48

Source not found.

----------------------------------------

LVS Summary:

Source: /openlane/designs/inverter_def_power_pin_segfault/runs/RUN_2022.02.09_04.44.48/results/lvs/inverter.lvs_parsed.gds.log

Source not found.

----------------------------------------

Antenna Summary:

No antenna report found.

[INFO]: check full report here: /openlane/designs/inverter_def_power_pin_segfault/runs/RUN_2022.02.09_04.44.48/reports/final_summary_report.csv

[INFO]: Saving Runtime Environment

[ERROR]: Flow Failed.

    while executing

"try_catch $::env(OPENROAD_BIN) -exit $::env(SCRIPTS_DIR)/openroad/groute.tcl |& tee $::env(TERMINAL_OUTPUT) $saveLOG"

    (procedure "global_routing_fastroute" line 5)

    invoked from within

"global_routing_fastroute"

    (procedure "global_routing" line 10)

    invoked from within

"global_routing"

    (procedure "run_routing" line 35)

    invoked from within

"run_routing"

    (procedure "run_routing_step" line 10)

    invoked from within

"[lindex $step_exe 0] [lindex $step_exe 1] "

    (procedure "run_non_interactive_mode" line 43)

    invoked from within

"run_non_interactive_mode {*}$argv"

    invoked from within

"if { [info exists flags_map(-interactive)] || [info exists flags_map(-it)] } {

	puts_info "Running interactively"

	if { [info exists arg_values(-file)..."

    (file "./flow.tcl" line 356)
```

I recommend including this design in designs/. I will add it to issue_regression testing once #881 is implemented.

maliberty, 2022-02-09:
Please package a reproducable test case per https://github.com/The-OpenROAD-Project/OpenLane/blob/master/docs/source/using_or_issue.md

@donn I thought this was autogenerated now with a message?

donn, 2022-02-09:
@maliberty This is on the mpw-5a, a version from November. We'd have to reproduce this on a reasonably recent version first.

armleo, 2022-02-10:
[RUN_2022.02.09_04.44.48_groute_packaged.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/8038121/RUN_2022.02.09_04.44.48_groute_packaged.tar.gz)

IDK what this gives, but here it is. my zip contained all of the files in question anyway.

maliberty, 2022-02-14:
- VPWR + NET VPWR + DIRECTION INOUT + USE SIGNAL  <- INTENTIONALLY DECLARED

This should be USE POWER not USE SIGNAL.  I think want pdngen to build a power grid not a signal router to make point to point connections.

armleo, 2022-02-14:
This issue is about segfault. "OpenLane should not crash no matter how bad the input is". There is multiple solutions to this otherwise.

maliberty, 2022-02-14:
When run with the latest OR there is no segv so there is nothing to fix.  I was just pointing out this probably won't do what you want in any case.

donn, 2022-02-14:
We'll keep the issue open for reference until mpw-5a is over.

PR #862 (2022-02-18): [BOT] Update openroad_app
This is an automated PR.

See the individual commits for details.
---
Resolves #909, Resolves #932.
closed 2022-02-18 by donn