The-OpenROAD-Project/OpenLane #912: [mpw5a] OpenLane errors out: [ERROR DRT-0074] No ap for PIN/fastio_in[16].

armleo, 2022-02-10:
### Description
mpw5a caravel based OpenLane installation.
OpenLane errors out: [ERROR DRT-0074] No ap for PIN/fastio_in[16].

I examined DEF/GDS and everything. There seems no reason for this to happen.

Things I checked:
1. Pin is located inside die area from config.tcl or def die area, which are exactly the same.
2. I suspected that the pin was too small, but increasing or decreasing pin width does not change anything.
3. I suspected that it didn't reach the main logic area, but increasing the length didn't change anything
4. I changed type of the pin from FIXED to PLACED, but didn't get any changes.
5. Used a def generated by OpenLane itself. It works.

Why I am trying to use a DEF and not locations generated by OpenLane pin config? Because I can't specify specific locations of the pins, including metal, spacing and width in any other way.

### Environment
```
Python: v3.9.5

Kernel: Linux v5.11.0-49-generic

Distribution: ubuntu 21.04

Container Engine: docker v20.10.10

OpenLane Git Version: 2021.11.23_01.42.34

---

PDK Version Verification Status: OK

---

Git Log (Last 3 Commits)

commit e1883edfbcc8c7e96491158521522b7a66541745

Author: Manar <manarabdelatty@aucegypt.edu>

Date:   Mon Nov 22 07:59:18 2021 -0600

    Update resizer (#726)

commit 6245b0e675d0e6751bdfc5362c545efe29d1077c

Author: Manar <manarabdelatty@aucegypt.edu>

Date:   Fri Nov 19 13:38:50 2021 -0600

    Fix sram install (#724)

    

    Also disable by default

commit e0e6fbc1bd90d1c73aae19769d997da8c3c47512

Author: Manar <manarabdelatty@aucegypt.edu>

Date:   Fri Nov 19 12:30:08 2021 -0600

    Add option for enabling/disabling repair_tie_fanout (#722)

    

    * Add option for enabling/disabling repair_tie_fanout

    

    * add escape characters

    

    * Add option for enabling sta report power

```

### Reproduction Material

tar gz containing two designs `carrack_wrapper` and `carrack_wrapper_user`:
[Uploading carrack.tar.gz…]()

carrack wrapper is the workspace used to generate the def with command: `./flow.tcl -interactive -file designs/carrack_wrapper/interactive.tcl`

carrack_wrapper_user is design that errors out: `./flow.tcl -design carrack_wrapper_user -tag carrack_wrapper_user -overwrite`

Manually generated or_issue archive:
[or_issue.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/8042958/or_issue.tar.gz)

or_issue command: python3 ./scripts/or_issue.py -s scripts/openroad/droute.tcl /openlane/designs/carrack_wrapper_user/runs/carrack_wrapper_user/tmp/routing/17-addspacers.def

### Expected behavior
No error and properly placed PnR GDS with pins from my def as an output :)

### Logs
```
[INFO DRT-0165] Start pin access.

[ERROR DRT-0074] No ap for PIN/fastio_in[16].

terminate called after throwing an instance of 'std::runtime_error'

  what():  DRT-0074

[ERROR]: during executing: "openroad -exit /openlane/scripts/openroad/droute.tcl |& tee >&@stdout /openlane/designs/carrack_wrapper_user/runs/carrack_wrapper_user/logs/routing/19-tritonRoute.log"

[ERROR]: Exit code: 1

[ERROR]: Last 10 lines:

child killed: SIGABRT

[ERROR]: Please check openroad  log file

[ERROR]: Dumping to /openlane/designs/carrack_wrapper_user/runs/carrack_wrapper_user/error.log

[INFO]: Calculating Runtime From the Start...

[INFO]: flow failed for carrack_wrapper/2022.02.10_17.00.33 in 0h6m12s

[INFO]: Generating Final Summary Report...

[INFO]: Design Name: carrack_wrapper

Run Directory: /openlane/designs/carrack_wrapper_user/runs/carrack_wrapper_user

Source not found.

----------------------------------------

LVS Summary:

Source: /openlane/designs/carrack_wrapper_user/runs/carrack_wrapper_user/results/lvs/carrack_wrapper.lvs_parsed.gds.log

Source not found.

----------------------------------------

Antenna Summary:

No antenna report found.

[INFO]: check full report here: /openlane/designs/carrack_wrapper_user/runs/carrack_wrapper_user/reports/final_summary_report.csv

[INFO]: Saving Runtime Environment

[ERROR]: Flow Failed.

    while executing

"try_catch $::env(OPENROAD_BIN) -exit $::env(SCRIPTS_DIR)/openroad/droute.tcl |& tee $::env(TERMINAL_OUTPUT) [index_file $::env(tritonRoute_log_file_ta..."

    (procedure "detailed_routing_tritonroute" line 6)

    invoked from within

"detailed_routing_tritonroute"

    (procedure "detailed_routing" line 13)

    invoked from within

"detailed_routing"

    (procedure "run_routing" line 54)

    invoked from within

"run_routing"

    (procedure "run_routing_step" line 10)

    invoked from within

"[lindex $step_exe 0] [lindex $step_exe 1] "

    (procedure "run_non_interactive_mode" line 43)

    invoked from within

"run_non_interactive_mode {*}$argv"

    invoked from within

"if { [info exists flags_map(-interactive)] || [info exists flags_map(-it)] } {

	puts_info "Running interactively"

	if { [info exists arg_values(-file)..."

    (file "./flow.tcl" line 356)

```

maliberty, 2022-02-10:
Your pin is off the manufacturing grid:

    - fastio_in[16] + NET fastio_in[16] + DIRECTION INPUT + USE SIGNAL
      + PORT
        + LAYER met2 ( -140 -2000 ) ( 140 2000 )
        + FIXED ( 122434 **3193985** ) N ;

See "MANUFACTURINGGRID 0.005 ;" in your LEF.  Since the pin shape itself is illegal no valid access can be made to it.

maliberty, 2022-02-10:
How was your pin placement generated?

maliberty, 2022-02-10:
Your script is quite complicated and hard to follow.  However 122.434 from DEF  is nowhere near 
puts [expr 329.99000 - 189.52 + 69.785 + 0.28]
210.535

So you must have a bug.  That isn't rounding error.

maliberty, 2022-02-10:
I think you are looking at the wrong pin:

 puts [expr 241.89 - 189.52 + 69.785 + 0.28]
122.43499999999997

shows the issue.  Try
set pos [expr 241.89 - 189.52 + 69.785 + 0.28]
puts [expr round($pos * 1000) / 1000.0]  
122.435

armleo, 2022-02-10:
I think it was fixed. Found the code line: https://github.com/The-OpenROAD-Project/OpenROAD/blob/9afb4fd0ca5fe8568c32aac33847211b9689800a/src/ppl/src/IOPlacer.tcl#L288

How to know if this commit made to the OpenRoad used by me?

maliberty, 2022-02-10:
EDA layout tools generally use integer DBU to avoid such floating point issues (see https://github.com/The-OpenROAD-Project/OpenROAD/blob/master/docs/contrib/DatabaseMath.md).

The router has an internal check but it is more to make sure that it isn't creating such problems rather than checking user inputs.  We could apply it to the user pins as a pre-check.  I'll open an issue for that.

I'm not sure about your link to IOPlacer.tcl.  The issue was in your or_ioplace.tcl script.

maliberty, 2022-02-10:
https://github.com/The-OpenROAD-Project/OpenROAD/issues/1634

armleo, 2022-02-12:
@maliberty 

Hi,

Issue still persists. Called:

```
place_pin -pin_name "wbs_sel_i[1]" -layer met2 -location [list 516.67 2.0] -pin_size [list $met2_pin_width $pin_length]
```

Which results in following lines in DEF:

```
    - wbs_sel_i[1] + NET wbs_sel_i[1] + DIRECTION INPUT + USE SIGNAL

      + PORT

        + LAYER met2 ( -140 -2000 ) ( 140 2000 )

        + FIXED ( 516669 2000 ) N ;
```

maliberty, 2022-02-13:
I just tried

```
place_pin -pin_name {clk} -layer met2 -location {2.0 512.915} -pin_size {4 0.28}
write_def foo.def
```
and see
```
PINS 54 ;
    - clk + NET clk + DIRECTION INPUT + USE SIGNAL
      + PORT
        + LAYER met2 ( -2000 -140 ) ( 2000 140 )
        + FIXED ( 2000 512915 ) N ;
```
so I can't reproduce your issue.  Please try with the latest openroad.

armleo, 2022-02-13:
Could you try the mpw5a version? I am not not upstream version.

armleo, 2022-02-13:
Okay. Just so you know, this is not critical right now, as a workaround has been made.
For upstream I will implement a warning for this in the current apply def template. As for OpenRoad, you already created an issue.

donn, 2022-02-25:
wrong issue. sorry.

commit 161699a6 (2022-02-25): [BOT] Update openroad_app (#938)

Merging to fix #912.
closed 2022-04-11 by donn