The-OpenROAD-Project/OpenLane #913: run_cts error on OpenLane design backend implementation using  techno Tower 180nm PDK

tomislavilic, 2022-02-11:
### Description
Used OpenLane interactive flow. prep -design, run_synthesis, run_floorplan and run_placement finished OK but run_cts failed. 

### Environment
```
Please run the following command in the OpenLane folder:

python3 ./env.py issue-survey

And copy and paste the ENTIRE output between the triple-backticks. Please do not gzip and upload the output.

If there's no env.py, you are using an out of date version of OpenLane and should probably update. 
```Kernel: Linux v4.4.0-210-generic
Distribution: ubuntu 16.04
Python: v3.8.9 (OK)
Container Engine: docker v20.10.7 (OK)
OpenLane Git Version: 2022.02.10_02.15.04
pip:click: INSTALLED
pip:pyyaml: INSTALLED
pip:venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

2f75eb2 2022-02-09T07:41:34+02:00 [BOT] Update netgen (#817) - Openlane Bot -  (HEAD -> master, tag: 2022.02.10_02.15.04, origin/master, origin/HEAD)
1860ec4 2022-02-07T22:27:50+02:00 add sky130hs/hd setup for gcd (#905) - Matt Liberty -  (tag: 2022.02.08_01.48.52)
5f7d5ca 2022-02-05T04:20:30+02:00 More ASAP7 Tweaks (#904) - Donn -  (tag: 2022.02.06_01.54.17)```
[openroad_issue_reproducible_run_cvs_error.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/8046044/openroad_issue_reproducible_run_cvs_error.zip)

### Reproduction Material
* Upload a tarball containing the relevant design.
[thames_top_src.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/8046101/thames_top_src.zip)
* List the commands used to run the design.
```The backend flow run is executed in OpenLane interactive mode using following commands:
> cd < path to OpenLane instalation directory>
> ./flow.tcl -interactive
prep -design thames_top
run_synthesis
run_floorplan
run_placement
run_cts

If you see a message like `Reproducible packaged: Please tarball and upload <PATH> if you're going to submit an issue` in your logs, please also tarball and include that path. This will greatly speed up the fixing process.

### Expected behavior
A clear and concise description of what you expected to happen.
Expecting that run_cts runs OK without the error.

### Logs
```
Add any relevant logs here. Please do ensure they're enclosed by the triple-backticks.
``` This is content of cts.log:OpenROAD 3eafa6da2942ec22d27e2208c99c7a449af34fbf 
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[INFO ODB-0222] Reading LEF file: /openlane/designs/thames_top/runs/RUN_2022.02.11_04.15.19/tmp/merged_unpadded.lef
[INFO ODB-0223]     Created 9 technology layers
[INFO ODB-0224]     Created 21 technology vias
[INFO ODB-0225]     Created 603 library cells
[INFO ODB-0226] Finished LEF file:  /openlane/designs/thames_top/runs/RUN_2022.02.11_04.15.19/tmp/merged_unpadded.lef
[INFO ODB-0127] Reading DEF file: /openlane/designs/thames_top/runs/RUN_2022.02.11_04.15.19/results/placement/thames_top.def
[INFO ODB-0128] Design: thames_top
[INFO ODB-0130]     Created 109 pins.
[INFO ODB-0131]     Created 914 components and 4261 component-terminals.
[INFO ODB-0132]     Created 2 special nets and 1828 connections.
[INFO ODB-0133]     Created 767 nets and 2433 connections.
[INFO ODB-0134] Finished DEF file: /openlane/designs/thames_top/runs/RUN_2022.02.11_04.15.19/results/placement/thames_top.def
###############################################################################
# Created by write_sdc
# Fri Feb 11 04:16:38 2022
[... 257 lines cut ...]
16# 0x00007FC1E651436C in /lib64/libtcl8.5.so
17# 0x00007FC1E651C647 in /lib64/libtcl8.5.so
18# TclEvalObjEx in /lib64/libtcl8.5.so
19# 0x00007FC1E64D71D0 in /lib64/libtcl8.5.so
20# 0x00007FC1E64CFEB2 in /lib64/libtcl8.5.so
21# 0x00007FC1E651436C in /lib64/libtcl8.5.so
22# 0x00007FC1E651C647 in /lib64/libtcl8.5.so
23# TclEvalObjEx in /lib64/libtcl8.5.so
24# 0x00007FC1E64DAF00 in /lib64/libtcl8.5.so
25# 0x00007FC1E64CFEB2 in /lib64/libtcl8.5.so
26# 0x00007FC1E651436C in /lib64/libtcl8.5.so
27# TclObjInterpProcCore in /lib64/libtcl8.5.so
28# 0x00007FC1E64CFEB2 in /lib64/libtcl8.5.so
29# 0x00007FC1E651436C in /lib64/libtcl8.5.so
30# TclObjInterpProcCore in /lib64/libtcl8.5.so
31# 0x00007FC1E64CFEB2 in /lib64/libtcl8.5.so
32# 0x00007FC1E64D0F1E in /lib64/libtcl8.5.so
33# Tcl_EvalEx in /lib64/libtcl8.5.so
34# Tcl_Eval in /lib64/libtcl8.5.so
35# sta::sourceTclFile(char const*, bool, bool, Tcl_Interp*) in openroad
36# ord::tclAppInit(Tcl_Interp*) in openroad
37# Tcl_Main in /lib64/libtcl8.5.so
38# main in openroad
39# __libc_start_main in /lib64/libc.so.6
40# 0x0000000000FCE8B7 in openroad```

maliberty, 2022-02-11:
Please package a standalone test case with https://github.com/The-OpenROAD-Project/OpenLane/blob/master/docs/source/using_or_issue.md

citdemond, 2022-02-11:
Hi maliberty,
Thank you for your quick response in this matter.

As you requested here is attached the "a standalone test case with https://github.com/The-OpenROAD-Project/OpenLane/blob/master/docs/source/using_or_issue.md"

[build.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/8046340/build.zip)

Best Regards

citdemond, 2022-02-11:
Sorry, I closed this issue by mistake. It is still open .....

maliberty, 2022-02-11:
You have 
`export CTS_CLK_BUFFER_LIST='CLKBUF_X2_18_SVT_WB CLKBUF_X4_18_SVT_WB CLKBUF_X8_18_SVT_WB CLKBUF_X16_18_SVT_WB';
`
but in LIB_CTS (='./tmp/cts/cts.lib') you have no such cells as they are marked:
`  /* removed CLKBUF_X2_18_SVT_WB */
...
`

We should have a proper error message but your cts.lib is messed up.

maliberty, 2022-02-11:
@eder-matheus please improve the error handling

maliberty, 2022-02-11:
Just out of curiosity, what process is this?

closed 2022-02-11 by citdemond