The-OpenROAD-Project/OpenLane #914: Yosys crashes when reading abc.script with return code: 134

vijayank88, 2022-02-11:
### Description
run_synthesis failed while executing abc.script

### Environment
```
Kernel: Linux v3.10.0-1160.49.1.el7.x86_64
Distribution: centos 7
Python: v3.6.8 (OK)
Critical Alert: No Docker or Docker-compatible container engine was found: {e}.
OpenLane Git Version: 2f75eb2
pip:click: INSTALLED
pip:pyyaml: INSTALLED
pip:venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

2f75eb2 2022-02-09T07:41:34+02:00 [BOT] Update netgen (#817) - Openlane Bot -  (HEAD -> master, tag: 2022.02.10_02.15.04, origin/master, origin/HEAD)
1860ec4 2022-02-07T22:27:50+02:00 add sky130hs/hd setup for gcd (#905) - Matt Liberty -  (tag: 2022.02.08_01.48.52)
5f7d5ca 2022-02-05T04:20:30+02:00 More ASAP7 Tweaks (#904) - Donn -  (tag: 2022.02.06_01.54.17)
```

### Reproduction Material
* Upload a tarball containing the relevant design. 
[multi_encoder_prj.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/8047575/multi_encoder_prj.zip)

* List the commands used to run the design.
Extract zip file and place `multi_encoder_prj` inside OpenLane/designs/
Inside OpenLane container:
% ./flow.tcl  -design multi_encoder_prj

### Expected behavior
run_synthesis should be passed

### Logs
```
18. Executing ABC pass (technology mapping using ABC).

18.1. Extracting gate netlist of module `\multi_encoder_prj' to `/tmp/yosys-abc-00ihl1/input.blif'..
Extracted 11076 gates and 11289 wires to a netlist network with 212 inputs and 15 outputs.

18.1.1. Executing ABC.
sh: line 1:   879 Aborted                 (core dumped) /build/bin/yosys-abc -s -f /tmp/yosys-abc-00ihl1/abc.script 2>&1
ERROR: ABC: execution of command "/build/bin/yosys-abc -s -f /tmp/yosys-abc-00ihl1/abc.script 2>&1" failed: return code 134.
[ERROR]: during executing: "yosys -c /openlane/scripts/yosys/synth.tcl -l /openlane/designs/multi_encoder_prj/runs/run1/logs/synthesis/1-synthesis.log |& tee >&@stdout"
[ERROR]: Exit code: 1
[ERROR]: Last 10 lines:
child process exited abnormally

[INFO]: Calculating Runtime From the Start...
[INFO]: Generating Final Summary Report...
[INFO]: Design Name: multi_encoder_prj
Run Directory: /openlane/designs/multi_encoder_prj/runs/run1
Source not found.
----------------------------------------

LVS Summary:
Source: /openlane/designs/multi_encoder_prj/runs/run1/logs/finishing/multi_encoder_prj.lvs.lef.log
Source not found.
----------------------------------------

Antenna Summary:
No antenna report found.
[INFO]: check full report here: /openlane/designs/multi_encoder_prj/runs/run1/reports/final_summary_report.csv
[INFO]: Saving runtime environment...
[ERROR]: Flow failed.

    while executing
"flow_fail"
    (procedure "try_catch" line 17)
    invoked from within
"try_catch $::env(SYNTH_BIN)  -c $::env(SYNTH_SCRIPT)  -l [index_file $::env(synthesis_logs)/synthesis.log]  |& tee $::env(TERMINAL_OUTPUT)"
    (procedure "run_yosys" line 32)
    invoked from within
"run_yosys"
    (procedure "run_synthesis" line 11)
    invoked from within
"[lindex $step_exe 0] [lindex $step_exe 1] "
    (procedure "run_non_interactive_mode" line 51)
    invoked from within
"run_non_interactive_mode {*}$argv"
    invoked from within
"if { [info exists flags_map(-interactive)] || [info exists flags_map(-it)] } {
	puts_info "Running interactively"
	puts_info "Note, that post_run_hook..."
    (file "./flow.tcl" line 405)

```

vijayank88, 2023-07-17:
test case attached: 
[issue_reproducible.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/12068669/issue_reproducible.zip)

kareefardi, 2024-07-17:
No longer crashes with OpenLane 73cbe2bf78ef87fbf3ac55c34bc7c5515a1fdc00

closed 2024-07-17 by kareefardi