The-OpenROAD-Project/OpenLane #915: PDK installation error

jimbo1990, 2022-02-11:
### Description
I am having problems installing the PDKs it seems as thought it is trying to install the pdks to /home/username/OpenLane/pdks/skywater-pdk/ instead of  /home/username/OpenLane/pdks/sky130A, hence it gives me an error. Im not expert in linux at all so any help would be appreciated .

Thanks

### Environment
`Kernel: Linux v5.10.16.3-microsoft-standard-WSL2
Distribution: opensuse-leap 15.2
Python: v3.6.10 (OK)
Container Engine: docker v20.10.12 (OK)
OpenLane Git Version: 2022.02.10_02.15.04
pip:click: INSTALLED
pip:pyyaml: INSTALLED
pip:venv: INSTALLED

PDK Version Verification Status: FAILED
/home/username/OpenLane/pdks/sky130A not found.
Traceback (most recent call last):
  File "/home/username/OpenLane/dependencies/verify_versions.py", line 144, in verify_versions
    raise Exception(f"{sky130_dir} not found.")
Exception: /home/username/OpenLane/pdks/sky130A not found.

Failed to compare PDKs.

Git Log (Last 3 Commits)

2f75eb2 2022-02-09T07:41:34+02:00 [BOT] Update netgen (#817) - Openlane Bot -  (HEAD -> master, tag: 2022.02.10_02.15.04, origin/master, origin/HEAD)
1860ec4 2022-02-07T22:27:50+02:00 add sky130hs/hd setup for gcd (#905) - Matt Liberty -  (tag: 2022.02.08_01.48.52)
5f7d5ca 2022-02-05T04:20:30+02:00 More ASAP7 Tweaks (#904) - Donn -  (tag: 2022.02.06_01.54.17)`

### Expected behavior
Should install pdks to path /home/username/OpenLane/pdks/sky130A

vijayank88, 2022-02-12:
@jimbo1990 Try this documentation https://github.com/The-OpenROAD-Project/OpenLane/blob/master/docs/source/manual_pdk_installation.md

donn, 2022-02-13:
We'd need the logs for `make pdk` as well, plus what steps you're executing exactly.

The PDK builds in two steps: skywater-pdk and then sky130 pdk. It will thus create multiple folders.

donn, 2022-02-28:
Alright, well, feel free to open another issue if you need one.

closed 2022-02-28 by donn

Ali-Sabir2, 2022-08-12:
I am declaring the command for power ring in script file but after the floorplanning and placement. It must the power ring in my design. 
But I found the design without power ring, Below message shown in the terminal.
Can anyone help me to understand this ?  
Initialization of layers failed: Unable to open file: /home/shahid/.klayout/tech/sky130A/sky130A.lyp (errno=2)