The-OpenROAD-Project/OpenLane #920: Updating OpenLane not working

mdzakirhussain, 2022-02-14:
### Description
I am just follwing the steps to update openlane. But it fails. 

### Environment
Kernel: Linux v5.13.0-28-generic
Distribution: ubuntu 20.04
Python: v3.8.10 (OK)
Container Engine: UNKNOWN vUNKNOWN (UNSUPPORTED)
OpenLane Git Version: mpw-4
pip:click: INSTALLED
pip:pyyaml: INSTALLED
pip:venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

0624c5c 2022-02-04T05:29:13+02:00 Reinstate changes from 673a63e lost in 0a5afc3 (#903) - Matt Liberty -  (HEAD -> master, origin/master, origin/HEAD)
f01acb5 2022-02-02T16:44:10-08:00 Add scripts/pdk-linker.py for ORFS/OL integration and ioplacer update (#897) - Matt Liberty -  ()
aac68a0 2022-01-31T18:03:00+02:00 Added PDN and tap halo distinction (#896) - Marwan Abbas -  ()

### Logs
```
The following command 

sudo make pdk    

gives me the following error

git clone https://github.com/google/skywater-pdk /home/student/openlane2/OpenLane/pdks/skywater-pdk
fatal: destination path '/home/student/openlane2/OpenLane/pdks/skywater-pdk' already exists and is not an empty directory.
make: *** [dependencies/pdk.mk:40: /home/student/openlane2/OpenLane/pdks/skywater-pdk/LICENSE] Error 128

```

tdene, 2022-02-14:
This seems related to #921, and thus also linked to #922

tdene, 2022-02-14:
@mdzakirhussain This issue seems to have been resolved with #922 on my end

Can you confirm as well?

mdzakirhussain, 2022-02-15:
I am still getting the following 

student@student-VirtualBox:~/openlane$ sudo make test
[sudo] password for student: 
cd /home/student/openlane && \
	docker run --rm -v /home/student/openlane:/openlane -v /home/student/openlane/designs:/openlane/install -v /home/student/openlane/pdks:/home/student/openlane/pdks -e PDK_ROOT=/home/student/openlane/pdks -e STD_CELL_LIBRARY=sky130_fd_sc_hd --user 0:0 -e DISPLAY=:0 -v /tmp/.X11-unix:/tmp/.X11-unix -v /root/.Xauthority:/.Xauthority --network host efabless/openlane:2022.02.14_01.57.38 sh -c "./flow.tcl -design spm -tag openlane_test -disable_output -overwrite"
[INFO]: 
	___   ____   ___  ____   _       ____  ____     ___
	/   \ |    \ /  _]|    \ | |     /    ||    \   /  _]
	|   | |  o  )  [_ |  _  || |    |  o  ||  _  | /  [_
	| O | |   _/    _]|  |  || |___ |     ||  |  ||    _]
	|   | |  | |   [_ |  |  ||     ||  _  ||  |  ||   [_
	\___/ |__| |_____||__|__||_____||__|__||__|__||_____|

[INFO]: Version: 2022.02.14_01.57.38-4-g74b9e10
[INFO]: Running non-interactively
[INFO]: Using design configuration at /openlane/designs/spm/config.tcl
[INFO]: Sourcing Configurations from /openlane/designs/spm/config.tcl
[INFO]: PDKs root directory: /home/student/openlane/pdks
[INFO]: PDK: sky130A
[INFO]: Setting PDKPATH to /home/student/openlane/pdks/sky130A
[INFO]: Standard Cell Library: sky130_fd_sc_hd
[INFO]: Optimization Standard Cell Library is set to: sky130_fd_sc_hd
[INFO]: Sourcing Configurations from /openlane/designs/spm/config.tcl
[INFO]: Current run directory is /openlane/designs/spm/runs/openlane_test
[INFO]: Storing configs into config.tcl ...
[INFO]: Preparing LEF Files
[INFO]: Extracting the number of available metal layers from /home/student/openlane/pdks/sky130A/libs.ref/sky130_fd_sc_hd/techlef/sky130_fd_sc_hd.tlef
[INFO]: The available metal layers (6) are li1  met1  met2  met3  met4  met5
[INFO]: Merging LEF Files...
[INFO]: Trimming Liberty...
[INFO]: Generating Exclude List...
[INFO]: Generating Exclude List...
[INFO]: Creating ::env(DONT_USE_CELLS)...
[ERROR]: Tool klayout is required by the flow scripts being used, but appears to not be installed in the environment.
[ERROR]: Please update your environment. OpenLane will now quit.
[INFO]: Calculating Runtime From the Start...
[INFO]: Generating Final Summary Report...
[INFO]: Design Name: spm
Run Directory: /openlane/designs/spm/runs/openlane_test
Source not found.
----------------------------------------

LVS Summary:
Source: /openlane/designs/spm/runs/openlane_test/logs/finishing/spm.lvs.lef.log
Source not found.
----------------------------------------

Antenna Summary:
No antenna report found.
[INFO]: check full report here: /openlane/designs/spm/runs/openlane_test/reports/final_summary_report.csv
[INFO]: Saving runtime environment...
[ERROR]: Flow failed.

    while executing
"flow_fail"
    invoked from within
"if [catch {exec python3 $::env(OPENLANE_ROOT)/dependencies/verify_versions.py} ::env(VCHECK_OUTPUT)] {
        if { [info exists ::env(MISMATCHES_OK)]..."
    (procedure "prep" line 379)
    invoked from within
"prep {*}$args"
    (procedure "run_non_interactive_mode" line 11)
    invoked from within
"run_non_interactive_mode {*}$argv"
    invoked from within
"if { [info exists flags_map(-interactive)] || [info exists flags_map(-it)] } {
	puts_info "Running interactively"
	puts_info "Note, that post_run_hook..."
    (file "./flow.tcl" line 405)
make: *** [Makefile:122: test] Error 1

vijayank88, 2022-02-15:
@mdzakirhussain 
[ERROR]: Tool klayout is required by the flow scripts being used, but appears to not be installed in the environment.

Install klayout and try again

mdzakirhussain, 2022-02-15:
@vijayank88 

The whole tools comes inside the flow called openlane, this openlane runs in a docker container. Though I have in my system klayout installed, that klayout should come inside the docker container.  It means when we clone the openlane repository and follow installation, klayout needs to be included in that repository so that it comes inside the docker container.

vijayank88, 2022-02-15:
@mdzakirhussain 
please try `make mount` followed by `./flow.tcl -design spm` works or not?

mdzakirhussain, 2022-02-15:
@vijayank88,

I already had tried that, but ended up with the same problem.

vijayank88, 2022-02-15:
@mdzakirhussain 
We are not able to reproduce this issue at our end.
Please try to install in new directory, if same issue repeats provide detailed installation steps and logs

donn, 2022-02-18:
Fixed in master- for your specific situation, just `rm -rf /home/student/openlane2/OpenLane/pdks/skywater-pdk`.

closed 2022-02-18 by donn