The-OpenROAD-Project/OpenLane #923: OpenLane fails when used with pre-built PDK

RTimothyEdwards, 2022-02-14:
### Description

Setup: Caravel user project is configured with PDK_ROOT pointed to a pre-installed version of the sky130 PDK from open_pdks

Problem: When running "make user_proj_example", I get the error shown in "Logs", below.

The problem is that OpenLane is looking for a file called SOURCES in directory sky130A. There is no such file created by or installed by open_pdks. OpenLane may not just revise the PDK contents to its own liking. The contents of SOURCES that OpenLane appears to be looking for seems to be the same information that is provided in the pdk installation's .config/nodeinfo.json file. Why is that file not being used?

### Environment

I'm running on Fedora 35 Linux

### Reproduction Material

None needed, hopefully.

### Expected behavior

Openlane should preferably be using the relevant contents from the existing file `.config/nodeinfo.json`.  If that information is insufficent, then the Openlane developers should be contacting me about ensuring the open_pdks generates all the needed information.  I'm not wedded to the JSON format or the .config directory, but regardless it is open_pdks that should be generating this content, not Openlane, and especially not in such a way as to make it impossible to use an external distribution of the PDK.

### Logs
```
[ERROR]: Failed to compare PDKs.
Could not find SOURCES file for the installed sky130A PDK.
Traceback (most recent call last):
  File "/openlane/dependencies/verify_versions.py", line 74, in verify_versions
    sources_str = open(sources_file).read()
FileNotFoundError: [Errno 2] No such file or directory: '/usr/share/pdk/sky130A/SOURCES'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/openlane/dependencies/verify_versions.py", line 77, in verify_versions
    "Could not find SOURCES file for the installed sky130A PDK."
Exception: Could not find SOURCES file for the installed sky130A PDK.

[ERROR]: Please update your environment. OpenLane will now quit.
```

proppy, 2022-02-25:
@donn is that something I could take on? that's would also help me for the [conda packaging stuff](https://github.com/hdl/conda-eda/pulls?q=+is%3Apr+author%3Aproppy+) I've been working on.

donn, 2022-02-28:
No, I know exactly what this is. I'll attend to it.

donn, 2022-03-03:
blocked on https://github.com/RTimothyEdwards/open_pdks/issues/233

PR #1022 (2022-03-29): CI Fixes
+ Pin all pip versions
~ Verify versions made more PDK independent, reads .config/nodeinfo.json
~ Make `requirements_lint.txt` separate
~ Update to new Black: changes to ** operator formatting
~ Various spelling/grammar fixes
~ git:// -> https://

---
Resolves #923.

This is a mostly CI change with little effect on the execution of the flow itself.
closed 2022-03-29 by donn