The-OpenROAD-Project/OpenLane #928: openroad detailed router crash

kareefardi, 2022-02-17:
### Description
Crash at detailed routing.

### Environment
```
python3 ./env.py issue-survey

Kernel: Linux v4.19.0-18-cloud-amd64                                                                                                                         
Distribution: debian 10 
Python: v3.7.3 (OK)
Container Engine: docker v20.10.8 (OK)
OpenLane Git Version: 2022.02.16_01.55.24
pip:click: INSTALLED
pip:pyyaml: INSTALLED
pip:venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

5a443b5 2022-02-16T02:50:41-08:00 Revert "Remove manual tech mapping of muxes and get abc to map them (#873)" - kareem -  (HEAD -> master)                  
169346f 2022-02-15T17:15:02+02:00 Remove catch around PDN environment variables (#924) - Anton Blanchard -  (tag: 2022.02.16_01.55.24, origin/master, origin/HEAD)
74b9e10 2022-02-14T22:46:40+02:00 Changes PDK recipe dependency to be order-only (#922) - Teodor-Dumitru Ene -  (tag: 2022.02.15_02.01.39) 
```

### Reproduction Material
`openroad_issue_reproducible.tar.gz` automatically created by openlane.

### Expected behavior
An actual error message.

### Logs
A section of the detailed router log:
```
[INFO DRT-0036] via3 guide region query size = 0.
[INFO DRT-0036] met4 guide region query size = 0.
[INFO DRT-0036] via4 guide region query size = 0.
[INFO DRT-0036] met5 guide region query size = 0.
[INFO DRT-0179] Init gr pin query.
[INFO DRT-0185] Post process initialize RPin region query.
Stack trace:
 0# 0x0000000000FD1FE9 in openroad
 1# 0x00007F780053C400 in /lib64/libc.so.6
 2# fr::io::Parser::initRPin_rpin() in openroad
 3# fr::io::Parser::initRPin() in openroad
 4# triton_route::TritonRoute::init(bool) in openroad
 5# triton_route::TritonRoute::main() in openroad
 6# detailed_route_cmd(char const*, char const*, char const*, char const*, char const*, char const*, bool, int, char const*, char const*, int, double, char const*, char const*, int) in openroad               
 7# 0x000000000193DC4E in openroad
 8# 0x00007F78034CFEB2 in /lib64/libtcl8.5.so
 9# 0x00007F780351436C in /lib64/libtcl8.5.so
10# TclObjInterpProcCore in /lib64/libtcl8.5.so
11# 0x00007F78034CFEB2 in /lib64/libtcl8.5.so
12# 0x00007F780351436C in /lib64/libtcl8.5.so
13# 0x00007F780351C647 in /lib64/libtcl8.5.so
14# TclEvalObjEx in /lib64/libtcl8.5.so
15# 0x00007F780355727F in /lib64/libtcl8.5.so
16# 0x00007F78034CFEB2 in /lib64/libtcl8.5.so
17# 0x00007F780351436C in /lib64/libtcl8.5.so
18# 0x00007F780351C647 in /lib64/libtcl8.5.so
19# TclEvalObjEx in /lib64/libtcl8.5.so
20# 0x00007F78034D71D0 in /lib64/libtcl8.5.so
21# 0x00007F78034CFEB2 in /lib64/libtcl8.5.so
22# 0x00007F780351436C in /lib64/libtcl8.5.so
23# 0x00007F780351C647 in /lib64/libtcl8.5.so
24# TclEvalObjEx in /lib64/libtcl8.5.so
25# 0x00007F78034DAF00 in /lib64/libtcl8.5.so
26# 0x00007F78034CFEB2 in /lib64/libtcl8.5.so
27# 0x00007F780351436C in /lib64/libtcl8.5.so
28# TclObjInterpProcCore in /lib64/libtcl8.5.so
29# 0x00007F78034CFEB2 in /lib64/libtcl8.5.so
30# 0x00007F780351436C in /lib64/libtcl8.5.so
31# TclObjInterpProcCore in /lib64/libtcl8.5.so
32# 0x00007F78034CFEB2 in /lib64/libtcl8.5.so
33# 0x00007F78034D0F1E in /lib64/libtcl8.5.so
34# Tcl_EvalEx in /lib64/libtcl8.5.so
35# Tcl_Eval in /lib64/libtcl8.5.so
36# sta::sourceTclFile(char const*, bool, bool, Tcl_Interp*) in openroad
37# ord::tclAppInit(Tcl_Interp*) in openroad
38# Tcl_Main in /lib64/libtcl8.5.so
39# main in openroad
40# __libc_start_main in /lib64/libc.so.6
41# 0x0000000000FCE8B7 in openroad
[ERROR]: during executing openroad script /openlane/scripts/openroad/droute.tcl

```
[openroad_issue_reproducible.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/8084430/openroad_issue_reproducible.tar.gz)

osamahammad21, 2022-02-20:
looking into it

osamahammad21, 2022-02-20:
After some debugging here is what I've got:
The problem is in the following nets:
- vccd1 ( user1_vssd_lvclamp_pad VCCD1 ) ( user1_vccd_lvclamp_pad VCCD1 ) + USE SIGNAL ;
- vccd2 ( user2_vssd_lvclamp_pad VCCD1 ) ( user2_vccd_lvclamp_pad VCCD1 ) + USE SIGNAL ;
- vssd1 ( user1_vssd_lvclamp_pad VSSD1 ) ( user1_vccd_lvclamp_pad VSSD1 ) + USE SIGNAL ;
- vssd2 ( user2_vssd_lvclamp_pad VSSD1 ) ( user2_vccd_lvclamp_pad VSSD1 ) + USE SIGNAL ;

These nets seem to be power and ground supply nets that are passed to tritonroute through the def as signal nets. TritonRoute does not route power and ground nets. And they should be special nets.These nets should be handled by pdn.
However, I think this problem can be detected in io with an error if that is needed.

kareefardi, 2022-02-20:
Thanks for tracking it down. I will check what went wrong in generating the def. If this is easily detectable, exiting with an error message would be much more helpful.

kareefardi, 2022-02-20:
On second thoughts, I wonder why it crashed in the first place. I agree that it's odd for these nets to be found in the nets section but if, for instance, these nets aren't routable by tritonRoute, shouldn't openroad report that ?

osamahammad21, 2022-02-20:
The problem here is that the nets are ill defined. They should be supply nets with the proper use defined. But here they are signal nets having power/ground pins connected to them.
TritonRoute should be able to route signal nets but here it's not really a signal net.
However, if you change the use of the nets to power and ground nets without making them special, it will also crash which should be fixed.

maliberty, 2022-02-21:
@osamahammad21 please fix the crash you mentioned and also add a warning for any net to be routed that is connected to a mterm that has USE pwr/gnd (presumably that happens in this case).

closed 2022-02-25 by osamahammad21