The-OpenROAD-Project/OpenLane #929: Segmentation fault during groute

nguyendao-uom, 2022-02-17:
### Description
Segmentation fault during groute without any further warnings/errors for debugging

### Environment
```
Python: v3.6.8
Kernel: Linux v4.18.0-338.el8.x86_64
Distribution: centos 8
Container Engine: docker v20.10.8
OpenLane Git Version: 2021.11.23_01.42.34
---
PDK Version Verification Status: MISMATCH
The version of open_pdks installed does not match the one required by the OpenLane flow scripts (installed: 27ecf1c16911f7dd4428ffab96f62c1fb876ea70, expected: c5730b574461889c82858b08d12ba42423d9c2cb)
You may want to re-install the PDK by invoking `make pdk`.
---
Git Log (Last 3 Commits)

commit e1883edfbcc8c7e96491158521522b7a66541745
Author: Manar <manarabdelatty@aucegypt.edu>
Date:   Mon Nov 22 07:59:18 2021 -0600

    Update resizer (#726) 
```

### Reproduction Material
Run command:
1. make mount
2. bash-4.2$ ./flow.tcl -design fabric_3x4 -tag fabric_3x4 -overwrite

[fabric_3x4_groute_packaged.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/8087973/fabric_3x4_groute_packaged.tar.gz)

### Expected behavior
Finish the flow without errors

### Logs
```
[INFO]: Running Global Routing...
[INFO]: current step index: 13
OpenROAD b79f266fe41540eabc064bcaddfe19ed715ac5c2 
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[INFO ODB-0222] Reading LEF file: /openlane/designs/fabric_3x4/runs/fabric_3x4/tmp/merged_unpadded.lef
[INFO ODB-0223]     Created 13 technology layers
[INFO ODB-0224]     Created 25 technology vias
[INFO ODB-0225]     Created 457 library cells
[INFO ODB-0226] Finished LEF file:  /openlane/designs/fabric_3x4/runs/fabric_3x4/tmp/merged_unpadded.lef
[INFO ODB-0127] Reading DEF file: /openlane/designs/fabric_3x4/runs/fabric_3x4/results/placement/eFPGA.placement.def
[INFO ODB-0128] Design: eFPGA
[INFO ODB-0130]     Created 541 pins.
[INFO ODB-0131]     Created 8643 components and 29620 component-terminals.
[INFO ODB-0133]     Created 4954 nets and 9092 connections.
[INFO ODB-0134] Finished DEF file: /openlane/designs/fabric_3x4/runs/fabric_3x4/results/placement/eFPGA.placement.def
[INFO]: Setting signal min routing layer to: li1 and clock min routing layer to li1. 
[INFO]: Setting signal max routing layer to: met4 and clock max routing layer to met4. 
[WARNING GRT-0030] Specified layer met5 for adjustment is greater than max routing layer met4 and will be ignored.
[INFO GRT-0020] Min routing layer: li1
[INFO GRT-0021] Max routing layer: met4
[INFO GRT-0022] Global adjustment: 0%
[INFO GRT-0023] Grid origin: (0, 0)
...
[INFO GRT-0096] Final congestion report:
Layer         Resource        Demand        Usage (%)    Max H / Max V / Total Overflow
---------------------------------------------------------------------------------------
li1               4624           162            3.50%             0 /  0 /  0
met1            312460         36078           11.55%             0 /  0 /  0
met2            236640         37269           15.75%             0 /  0 /  0
met3            158929          2630            1.65%             0 /  0 /  0
met4            113617             0            0.00%             0 /  0 /  0
---------------------------------------------------------------------------------------
Total           826270         76139            9.21%             0 /  0 /  0

[INFO GRT-0018] Total wirelength: 762401 um
[INFO GRT-0006] Repairing antennas, iteration 1.
[WARNING GRT-0043] No OR_DEFAULT vias defined.
[INFO GRT-0224] Chose via L1M1_PR as default.
...
Notice 0: Split top of 1 T shapes.
Notice 0: 
disconnected net 962  Tile_X1Y0_S1BEG\[0\]
Notice 0: Split top of 2 T shapes.
Notice 0: Split top of 2 T shapes.
Notice 0: Split top of 2 T shapes.
Notice 0: 
disconnected net 965  Tile_X1Y0_S1BEG\[3\]
Notice 0: Split top of 2 T shapes.
[ERROR]: during executing: "openroad -exit /openlane/scripts/openroad/groute.tcl |& tee >&@stdout /openlane/designs/fabric_3x4/runs/fabric_3x4/logs/routing/13-fastroute.log"
[ERROR]: Exit code: 1
[ERROR]: Last 10 lines:
child killed: segmentation violation
```

nguyendao-uom, 2022-02-17:
Hi @maliberty, 
It seems that the recent OpenLane version still doesn't resolve the routing problem for FPGA fabric designs. I'm trying to stitch a small fabric but couldn't pass the groute step. There is no helpful warnings/errors that can be used for debugging. Please check the testcase and let me know what's wrong with the design/flow? Thanks.

vijayank88, 2022-02-17:
@nguyendao-uom 
`OpenLane Git Version: 2021.11.23_01.42.34` its an older version. Please try with latest version `[2022.02.16_01.55.24](https://github.com/The-OpenROAD-Project/OpenLane/releases/tag/2022.02.16_01.55.24)`and still facing the issue, do attach updated tar file.

nguyendao-uom, 2022-02-17:
Hi @vijayank88, I used all default tags mpw-5a from caravel_user_project, also there is a suggestion on Slack that we should not change openlane version otherwise the flow can be broken(?)

nguyendao-uom, 2022-02-17:
Sorry, please ignore the above comment, that's applied for the pdks only. I'll update the openlane and re-run the test.

vijayank88, 2022-02-17:
@nguyendao-uom thanks for the update.
@maliberty  @donn 
Even with latest OR App error repeats.
```
Notice 0: Split top of 1 T shapes.
Notice 0: Split top of 1 T shapes.
Notice 0: Split top of 1 T shapes.
Notice 0: 
disconnected net 965  Tile_X1Y0_S1BEG\[3\]
Notice 0: Split top of 1 T shapes.
Segmentation fault (core dumped)

```

nguyendao-uom, 2022-02-22:
@donn @maliberty have you had a look on this issue? Any suggestions?
The issue still remains with the latest openlane version (2022.02.21_02.33.47)
[openroad_issue_reproducible.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/8117728/openroad_issue_reproducible.tar.gz)

maliberty, 2022-02-23:
@eder-matheus this crash is in the dreaded order_wires code called on the groute generated segments inside antenna fixing.  The input is odd in that I see 7 segments for a simple two pin net (Tile_X1Y0_S2BEG\[0\]):

[image]

The segments are overlapping eachother in odd ways.  If we can fix this above order_wires that would be good.  Would you take a look and see if you understand how the groute segs were generated here?

The segs are all on layer = 3 with a init_x = 251850:

{init_y = 969450, final_x = 251850, final_y = 976350}
{init_y = 976350, final_x = 251850, final_y = 962550}
{init_y = 969450, final_x = 251850, final_y = 983250}
{init_y = 960950, final_x = 251930, final_y = 960950}
{init_y = 960950, final_x = 251850, final_y = 962550}
{init_y = 981650, final_x = 251930, final_y = 981650}
{init_y = 981650, final_x = 251850, final_y = 983250}

eder-matheus, 2022-02-23:
> 

It happens because this routing connects two macro pins. Grt creates the fake pins for them and add the guides that connect the real pin position to the routing. I'll try to enhance the function that merge the segments to avoid these odd routes.

nguyendao-uom, 2022-02-23:
Thanks @eder-matheus, can you also check if the tool supports channel less placement of blocks?

maliberty, 2022-02-23:
@eder-matheus I don't believe OL uses macro pin extension - are you saying grt always creates fake pins?

@nguyendao-uom there is nothing to route in channel-less placement so I'm not sure what needs checking.

nguyendao-uom, 2022-02-24:
Hi @eder-matheus, seems the issue have not resolved with the updated openlane 2022.02.23_02.50.41?
Can you check again? Thanks
[openroad_issue_reproducible.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/8131879/openroad_issue_reproducible.tar.gz)

eder-matheus, 2022-02-24:
> Hi @eder-matheus, seems the issue have not resolved with the updated openlane 2022.02.23_02.50.41? Can you check again? Thanks [openroad_issue_reproducible.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/8131879/openroad_issue_reproducible.tar.gz)

The latest openlane is probably not using my last commits on the OpenROAD repository. I've tested your testcase locally and I don't see the segfault.

donn, 2022-02-24:
That may have something to do with the fact OpenLane has not been updated to the latest version of OpenROAD yet. This is a manual process that someone has to do (@openlane-bot makes it easier, but doesn't automatically apply it.)

If you do any critical fixes in OpenROAD, please just @ me or a member of the OpenLane team who is not on vacation atm.

eder-matheus, 2022-02-24:
> That may have something to do with the fact OpenLane has not been updated to the latest version of OpenROAD yet. This is a manual process that someone has to do (@openlane-bot makes it easier, but [...]

Got it, I'll do it next time I have a critical update regarding OpenLane flow. Thanks.

donn, 2022-02-25:
Fixed in 161699a68bf82102f66f9d588d3a83e626cc5d05. This issue will remain open as it is still relevant for the mpw-5 version of OpenLane and it documents a workaround.