The-OpenROAD-Project/OpenLane #930: Choosing a Synthesis Strategy

mousaq92, 2022-02-17:
Hello,

Below is a screenshot of a synthesis exploration that I have done on spm. According to the OpenLane documentation, there is a SYNTH_STRATEGY variable that can be used in config.tcl to choose a synthesis strategy. This SYNTH_STRATEGY variable can only be set using "DELAY 0" or "AREA 1" for example.

My question is: how can I choose any of the strategies in the image below? For example how can I choose strategy 1 or 3 or 6. I find that for other designs there are more than 7 strategies.

[image]

Thank you,

Mousa

maliberty, 2022-02-17:
I ran into the same question today.  The strategies are poorly labelled but  there seem to be 4 delay and 3 area strategies in that order.  I think the labelling should be

DELAY 1
DELAY 2
DELAY 3
DELAY 4
AREA 1
AREA 2
AREA 3

I haven't tested it yet though.

maliberty, 2022-02-17:
@donn This looks like both a doc issue and an enhancement to the labelling.

Also the docs says "summary report under reports that compares the 4 outputs." which there are now 7 outputs.

dralabeing, 2022-02-23:
@mousaq92 there isn't clear documentation nor is the scope for choosing the best strategy fully tested yet. The definition is here https://github.com/The-OpenROAD-Project/OpenLane/blob/master/scripts/yosys/synth.tcl which allows you to access multiple synthesis scripts/recipes for abc based on area, delay settings. The number in the synthesis strategy (as in 5 in S5) is an index into the list (area or delay) to use. The optimal area vs delay recipe for the type of design goal is not always obvious or consistent as seen from initial testing. Please feel free to explore and report your findings.

donn, 2022-09-07:
I think I fixed this a while back, it's just mislabeled as "documentation." Synthesis exploration now prints the actual strategy name.

closed 2022-09-07 by donn