The-OpenROAD-Project/OpenLane #935: Issue setting up sky130 PDK (make pdk)

allanjimegonza, 2022-02-18:
Hi! I was wondering if someone else is having trouble setting up sky130 in the installation process (running "make pdk"). The error I'm having is related with the the magic version. Here is the output of the console:

Adding Tcl script options from file custom/scripts/gds_import_setup.tcl
Migrating GDS files to layout.
Searching for supported devices in PDK script /home/allan/Desktop/ALS/OpenLane/pdks/open_pdks/sky130/sky130A/libs.tech/magic/sky130A.tcl.
PDK library is sky130
Getting GDS file list from /home/allan/Desktop/ALS/OpenLane/pdks/open_pdks/sky130/sky130A/libs.ref/sky130_fd_pr/gds.
Creating magic generation script to generate magic database files.
No source for abstract views:  Abstract views not made.
Running magic to create magic database files.

Magic 8.3 revision 209 - Compiled on Wed Dec 29 12:54:06 UTC 2021.
Starting magic under Tcl interpreter
Using the terminal as the console.
Using NULL graphics device.
Processing system .magicrc file
Sourcing design .magicrc for technology sky130A ...
2 Magic internal units = 1 Lambda
Input style sky130(vendor): scaleFactor=2, multiplier=2
Loading sky130A Device Generator Menu ...
Scaled tech values by 2 / 1 to match internal grid scaling
Error message output from magic script:
/home/allan/Desktop/ALS/OpenLane/pdks/open_pdks/sky130/sky130A/libs.tech/magic/sky130A.tech: line 19: section version:
        Error:  Magic version 8.3.260 is required by this techfile, but this version of magic is 8.3.209.
The following sections of /home/allan/Desktop/ALS/OpenLane/pdks/open_pdks/sky130/sky130A/libs.tech/magic/sky130A.tech contained errors:
    version
Traceback (most recent call last):
  File "../common/foundry_install.py", line 1696, in <module>
    cwd = destlibdir,
  File "../common/foundry_install.py", line 274, in subprocess_run
    raise SystemError("".join(emsg))
SystemError: Command magic failed with exit code: -11
  magic -dnull -noconsole
Input script was:

 !/usr/bin/env wish

  Script to generate .mag files from .gds    

crashbackups stop
drc off
locking off
gds readonly true
gds drccheck false
gds flatten true
gds rescale false
tech unlock *
 Set the GDS input style to sky130(vendor).  This treats labels on the
 TXT purpose (5) as pins, which is unfortunately done in a lot of the
 vendor GDS files.
cif istyle sky130(vendor)
gds read /home/allan/Desktop/ALS/OpenLane/pdks/open_pdks/sky130/sky130A/libs.ref/sky130_fd_pr/gds/sky130_fd_pr__rf_pnp_05v5_W0p68L0p68.gds
gds read /home/allan/Desktop/ALS/OpenLane/pdks/open_pdks/sky130/sky130A/libs.ref/sky130_fd_pr/gds/sky130_fd_pr__rf_pnp_05v5_W3p40L3p40.gds
gds read /home/allan/Desktop/ALS/OpenLane/pdks/open_pdks/sky130/sky130A/libs.ref/sky130_fd_pr/gds/sky130_fd_pr__rf_npn_11v0_W1p00L1p00.gds
set devlist {sky130_fd_pr__rf_npn_05v5_W1p00L1p00 sky130_fd_pr__rf_npn_05v5_W1p00L2p00 sky130_fd_pr__rf_pnp_05v5_W0p68L0p68 sky130_fd_pr__rf_pnp_05v5_W3p40L3p40 sky130_fd_pr__rf_test_coil1 sky130_fd_pr__rf_test_coil2 sky130_fd_pr__rf_test_coil3 sky130_fd_pr__cap_vpp_11p5x11p7_m1m2m3m4_shieldl1m5 sky130_fd_pr__cap_vpp_11p5x11p7_m1m2_noshield sky130_fd_pr__cap_vpp_08p6x07p8_m1m2_shieldl1 sky130_fd_pr__cap_vpp_04p4x04p6_m1m2_shieldl1}
set topcell [lindex [cellname list top] 0]
foreach cellname $devlist {
    load $cellname
    property gencell $cellname
    property parameter m=1
    property library sky130
}
load $topcell
puts stdout "Writing all magic database files"
writeall force
puts stdout "Done."
quit -noprompt

make[2]: *** [primitive-A] Error 1
make[2]: Leaving directory /home/allan/Desktop/ALS/OpenLane/pdks/open_pdks/sky130
make[1]: *** [primitive-build-A] Error 2
make[1]: Leaving directory /home/allan/Desktop/ALS/OpenLane/pdks/open_pdks/sky130
make: *** [all-A] Error 2
make: *** [dependencies/pdk.mk:75: build-pdk] Error 2

tdene, 2022-02-18:
I believe that
```Error message output from magic script:
/home/allan/Desktop/ALS/OpenLane/pdks/open_pdks/sky130/sky130A/libs.tech/magic/sky130A.tech: line 19: section version:
Error: Magic version 8.3.260 is required by this techfile, but this version of magic is 8.3.209.
The following sections of /home/allan/Desktop/ALS/OpenLane/pdks/open_pdks/sky130/sky130A/libs.tech/magic/sky130A.tech contained errors:```
is the key part of your error log, and that this issue is the same as #934

allanjimegonza, 2022-02-19:
Oh, I definitely missed that issue, thaks a lot for the help!

donn, 2022-02-28:
I'm gonna go ahead and assume this is solved now- Thanks @tdene for the PR.

closed 2022-02-28 by donn