The-OpenROAD-Project/OpenLane #936: ECO flow doesn't place fill cells

antonblanchard, 2022-02-18:
The ECO flow doesn't place fill cells. Looking at the GDS:

[image]

The flow disables fill insertion while in the ECO loop:

```
    if { ($::env(DIODE_INSERTION_STRATEGY) != 3) && ($::env(ECO_ENABLE) ==0 || $::env(ECO_FINISH) ==1) } {
        ins_fill_cells
    }
```

We disable fill cell insertion because the ECO loop is going to insert buffers. The issue is we never call routing a final time with `ECO_FINISH=1` so that fill insertion can be done.

The downside to doing it this way is that we'll always call routing an extra time, just for fill. I presume we have to place fill before detailed routing, otherwise we might end up with DRC issues in DRT. I wonder if we could insert fill as per normal, and if we need to add buffers then remove all fill, insert buffers and insert fill.

msaligane, 2022-02-18:
I like the idea of inserting fills before DRT, remove them during ECO + insert buffers then insert fills.

msaligane, 2022-02-19:
That's actually what I typed on the slack channel just now. @antonblanchard have you tested just adding the fills manually after DRT. You can test DRC/LVS. I don't think you ll have any issues. 

The cleaner way to do this, is when we place the inserted buffer, we do a localized removal of the fills. and then run fills again.

antonblanchard, 2022-02-19:
@maliberty I was under the impression we could not add fill cells after DRT. Openlane adds them before detailed routing, and even has a comment about wanting to add them before global routing where possible. I assumed that was because routing might place tracks on layers that fill cells use.

@msaligane localised removal of fills sounds good, but we'd need to ensure enough fill is removed on a high density design to give space for detailed placement to fit the buffers.

maliberty, 2022-02-19:
In sky130 the stdcell pins are on li1 which is not used as a routing layer due to its high resistance.  So the fill cells have no effect on routing.

commit e40929f8 (2022-02-20): ECO Flow Fix (#940)

Fixes #936 by re-running the insfill step again when ECO is done.
closed 2022-02-20 by donn