The-OpenROAD-Project/OpenLane #945: Flow fails during groute - "Run global_route before repair_antennas"

snmarathe, 2022-02-21:
### Description
Flow fails during groute on one of my designs with the error "Run global_route before repair_antennas". 
### Environment
```
Kernel: Linux v4.19.128-microsoft-standard
Distribution: ubuntu 18.04
Python: v3.6.9 (OK)
Container Engine: docker v20.10.12 (OK)
OpenLane Git Version: 2022.02.15_02.01.39
pip:click: INSTALLED
pip:pyyaml: INSTALLED
pip:venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

74b9e10 2022-02-14T22:46:40+02:00 Changes PDK recipe dependency to be order-only (#922) - Teodor-Dumitru Ene -  (HEAD -> master, tag: 2022.02.15_02.01.39, origin/master, origin/HEAD)
19f0875 2022-02-14T18:52:44+02:00 CVC Script Cleanup (#916) - Donn -  ()
f42a0fc 2022-02-14T14:41:56+02:00 Make Klayout an OpenLane Tool Proper, Update to Klayout 0.27.7 (#919) - Donn -  ()
```

### Reproduction Material
Commands - 
```
make mount
OpenLane Container (74b9e10):/openlane$ ./flow.tcl -design processor
```
Design - [processor_design.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/8107556/processor_design.tar.gz)

Issue reproducible - [processor_issue.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/8107575/processor_issue.tar.gz)

### Logs
```
[INFO]: Running Global Routing...
OpenROAD 3eafa6da2942ec22d27e2208c99c7a449af34fbf
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[INFO ODB-0222] Reading LEF file: /openlane/designs/processor/runs/RUN_2022.02.21_08.14.43/tmp/merged_unpadded.lef
[INFO ODB-0223]     Created 13 technology layers
[INFO ODB-0224]     Created 25 technology vias
[INFO ODB-0225]     Created 441 library cells
[INFO ODB-0226] Finished LEF file:  /openlane/designs/processor/runs/RUN_2022.02.21_08.14.43/tmp/merged_unpadded.lef
[INFO ODB-0127] Reading DEF file: /openlane/designs/processor/runs/RUN_2022.02.21_08.14.43/tmp/cts/12-resizer_timing.def
[INFO ODB-0128] Design: processor
[INFO ODB-0130]     Created 4 pins.
[INFO ODB-0131]     Created 31 components and 110 component-terminals.
[INFO ODB-0132]     Created 2 special nets and 110 connections.
[INFO ODB-0133]     Created 2 nets and 0 connections.
[INFO ODB-0134] Finished DEF file: /openlane/designs/processor/runs/RUN_2022.02.21_08.14.43/tmp/cts/12-resizer_timing.def
###############################################################################
# Created by write_sdc
# Mon Feb 21 08:15:07 2022
###############################################################################
current_design processor
###############################################################################
# Timing Constraints
###############################################################################
create_clock -name clk -period 10.0000 [get_ports {clk}]
set_clock_transition 0.1500 [get_clocks {clk}]
set_clock_uncertainty 0.2500 clk
set_propagated_clock [get_clocks {clk}]
set_input_delay 2.0000 -clock [get_clocks {clk}] -add_delay [get_ports {rst}]
###############################################################################
# Environment
###############################################################################
set_driving_cell -lib_cell sky130_fd_sc_hd__inv_2 -pin {Y} -input_transition_rise 0.0000 -input_transition_fall 0.0000 [get_ports {clk}]
set_driving_cell -lib_cell sky130_fd_sc_hd__inv_2 -pin {Y} -input_transition_rise 0.0000 -input_transition_fall 0.0000 [get_ports {rst}]
set_timing_derate -early 0.9500
set_timing_derate -late 1.0500
###############################################################################
# Design Rules
###############################################################################
set_max_fanout 5.0000 [current_design]
[INFO]: Setting signal min routing layer to: met1 and clock min routing layer to met1.
[INFO]: Setting signal max routing layer to: met5 and clock max routing layer to met5.
-congestion_iterations 50 -verbose
[INFO GRT-0020] Min routing layer: met1
[INFO GRT-0021] Max routing layer: met5
[INFO GRT-0022] Global adjustment: 30%
[INFO GRT-0023] Grid origin: (0, 0)
[WARNING GRT-0043] No OR_DEFAULT vias defined.
[INFO GRT-0088] Layer li1     Track-Pitch = 0.4600  line-2-Via Pitch: 0.3400
[INFO GRT-0088] Layer met1    Track-Pitch = 0.3400  line-2-Via Pitch: 0.3400
[INFO GRT-0088] Layer met2    Track-Pitch = 0.4600  line-2-Via Pitch: 0.3500
[INFO GRT-0088] Layer met3    Track-Pitch = 0.6800  line-2-Via Pitch: 0.6150
[INFO GRT-0088] Layer met4    Track-Pitch = 0.9200  line-2-Via Pitch: 1.0400
[INFO GRT-0088] Layer met5    Track-Pitch = 3.4000  line-2-Via Pitch: 3.1100
[INFO GRT-0019] Found 0 clock nets.
[INFO GRT-0001] Minimum degree: 2147483647
[INFO GRT-0002] Maximum degree: -2147483648
[INFO GRT-0003] Macros: 0
[INFO GRT-0004] Blockages: 0

[INFO GRT-0053] Routing resources analysis:
          Routing      Original      Derated      Resource
Layer     Direction    Resources     Resources    Reduction (%)
---------------------------------------------------------------
li1        Vertical            0             0          0.00%
met1       Horizontal        560           307          45.18%
met2       Vertical          420           342          18.57%
met3       Horizontal        280           180          35.71%
met4       Vertical          168           131          22.02%
met5       Horizontal         56            24          57.14%
---------------------------------------------------------------

[INFO GRT-0197] Via related to pin nodes: 0
[INFO GRT-0198] Via related Steiner nodes: 0
[INFO GRT-0199] Via filling finished.
[INFO GRT-0111] Final number of vias: 0
[INFO GRT-0112] Final usage 3D: 0

[INFO GRT-0096] Final congestion report:
Layer         Resource        Demand        Usage (%)    Max H / Max V / Total Overflow
---------------------------------------------------------------------------------------
li1                  0             0            0.00%             0 /  0 /  0
met1               307             0            0.00%             0 /  0 /  0
met2               342             0            0.00%             0 /  0 /  0
met3               180             0            0.00%             0 /  0 /  0
met4               131             0            0.00%             0 /  0 /  0
met5                24             0            0.00%             0 /  0 /  0
---------------------------------------------------------------------------------------
Total              984             0            0.00%             0 /  0 /  0

[INFO GRT-0018] Total wirelength: 0 um
[ERROR GRT-0045] Run global_route before repair_antennas.
Error: groute.tcl, 74 GRT-0045
[ERROR]: during executing openroad script /openlane/scripts/openroad/groute.tcl
[ERROR]: Exit code: 1
[ERROR]: Last 10 lines:
child process exited abnormally
```

Please let me know what is wrong with the design/configuration.
Thanks.

vijayank88, 2022-02-21:
@snmarathe 
No standard cells to route here.

snmarathe, 2022-02-21:
Error resolved. Thanks a lot, @vijayank88

closed 2022-02-21 by snmarathe