The-OpenROAD-Project/OpenLane #948: Detailed placement failed

vijayank88, 2022-02-22:
### Description
GDS file generated successfully with OL tag: [2022.02.16_01.55.24](https://github.com/The-OpenROAD-Project/OpenLane/releases/tag/2022.02.16_01.55.24) but for latest OL tag: [2022.02.21_02.33.47](https://github.com/The-OpenROAD-Project/OpenLane/releases/tag/2022.02.21_02.33.47) detailed placement got failed.

### Environment
```
Kernel: Linux v3.10.0-1160.49.1.el7.x86_64
Distribution: centos 7
Python: v3.6.8 (OK)
Container Engine: docker v20.10.12 (OK)
OpenLane Git Version: 2022.02.21_02.33.47
pip:click: INSTALLED
pip:pyyaml: INSTALLED
pip:venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

e40929f 2022-02-20T10:29:53+02:00 ECO Flow Fix (#940) - donghl17 -  (HEAD -> master, tag: 2022.02.21_02.33.47, origin/master, origin/HEAD)
a7d97cc 2022-02-19T10:00:15+02:00 Update README.md (#939) - wale soyinka -  (tag: 2022.02.20_02.17.26)
58f206f 2022-02-18T16:02:03+02:00 Update Magic, Open_PDKS (#917) - Donn -  (tag: 2022.02.19_02.27.52)

```

### Reproduction Material
* Upload a tarball containing the relevant design. 
[DPL_issue_jacaranda8.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/8116365/DPL_issue_jacaranda8.zip)

* List the commands used to run the design.
* unzip the file and goto openroad_issue_reproducible
* openroad> source run.tcl

### Expected behavior
Has previous OL tag ran successfully, looking for current tag not produce this issue.

### Logs
```
[INFO DPL-0034] Detailed placement failed on:
[INFO DPL-0035]  ANTENNA_clkbuf_leaf_357_wb_clk_i_A
[INFO DPL-0035]  ANTENNA_clkbuf_leaf_357_wb_clk_i_A
[INFO DPL-0035]  ANTENNA_clkbuf_leaf_357_wb_clk_i_A
[ERROR DPL-0036] Detailed placement failed.
Error: opendp.tcl, 32 DPL-0036
```

vijayank88, 2022-02-22:
@maliberty FYI

maliberty, 2022-02-23:
The density is too high.  You should either add the diode to the padding exclude list or use a lower density during global placement.  I would do the former.

dralabeing, 2022-02-23:
@vijayank88 looks like this is resolved now for both the options.

closed 2022-02-23 by dralabeing