The-OpenROAD-Project/OpenLane #954: Flow Failed with Magic DRC Violations

vijayank88, 2022-02-23:
### Description
Flow got completed successfully with OL tag: [2022.02.16_01.55.24](https://github.com/The-OpenROAD-Project/OpenLane/releases/tag/2022.02.16_01.55.24). Same design got failed with Magic DRC violations with OL tag: [2022.02.21_02.33.47](https://github.com/The-OpenROAD-Project/OpenLane/releases/tag/2022.02.21_02.33.47)

### Environment
```
python3 ./env.py issue-survey

Kernel: Linux v3.10.0-1160.49.1.el7.x86_64
Distribution: centos 7
Python: v3.6.8 (OK)
Container Engine: docker v20.10.12 (OK)
OpenLane Git Version: 2022.02.21_02.33.47
pip:click: INSTALLED
pip:pyyaml: INSTALLED
pip:venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

e40929f 2022-02-20T10:29:53+02:00 ECO Flow Fix (#940) - donghl17 -  (HEAD -> master, tag: 2022.02.21_02.33.47, origin/master, origin/HEAD)
a7d97cc 2022-02-19T10:00:15+02:00 Update README.md (#939) - wale soyinka -  (tag: 2022.02.20_02.17.26)
58f206f 2022-02-18T16:02:03+02:00 Update Magic, Open_PDKS (#917) - Donn -  (tag: 2022.02.19_02.27.52)
```

### Reproduction Material
* Upload a tarball containing the relevant design. Inside OL_tag `finishing` directory has final GDS files.
* OL tag: [2022.02.16_01.55.24] gds files:
* OL tag: [2022.02.21_02.33.47] gds files:
* Due to file upload limit zip file attached here: https://slack-files.com/T01699QAZBQ-F034BBR5T8U-2bc88bbcfc
* List the commands used to run the design.

### Expected behavior
Magic DRC check should pass without violations.

### Logs
```
[INFO]: Running Magic DRC...
[INFO]: Converting Magic DRC Violations to Magic Readable Format...
[INFO]: Converting Magic DRC Violations to Klayout XML Database...
[INFO]: Converting DRC Violations to RDB Format...
[INFO]: Converted DRC Violations to RDB Format
[ERROR]: There are violations in the design after Magic DRC.
[ERROR]: Total Number of violations is 6
[INFO]: Calculating Runtime From the Start...
[INFO]: Generating Final Summary Report...
[INFO]: Design Name: user_proj_example
Run Directory: /openlane/designs/FCNet_prj/runs/RUN_2022.02.22_06.52.45
----------------------------------------

Magic DRC Summary:
Source: /openlane/designs/FCNet_prj/runs/RUN_2022.02.22_06.52.45/reports/finishing/drc.rpt
Violation Message "Metal3 minimum area < 0.24um^2 (met3.6) "found 6 Times.
Total Magic DRC violations is 6
----------------------------------------

LVS Summary:
Source: /openlane/designs/FCNet_prj/runs/RUN_2022.02.22_06.52.45/logs/finishing/32-user_proj_example.lvs.lef.log
LVS reports no net, device, pin, or property mismatches.
Total errors = 0
----------------------------------------

Antenna Summary:
No antenna report found.
[INFO]: check full report here: /openlane/designs/FCNet_prj/runs/RUN_2022.02.22_06.52.45/reports/final_summary_report.csv
[INFO]: Saving runtime environment...
[ERROR]: Flow failed.
```

maliberty, 2022-02-23:
What sort of violations are they?

vijayank88, 2022-02-24:
@mithro Your view on this?

vijayank88, 2022-02-24:
@maliberty can we link this issue to https://github.com/RTimothyEdwards/magic/issues ?

maliberty, 2022-02-24:
Do we know it is a magic issue?  Maybe there really are minarea problems.

vijayank88, 2022-02-24:
@maliberty Last two OL tags producing this issue.
Here is user repo: https://github.com/mxiangyue/caravel_user_project/blob/main/openlane/user_proj_example/config.tcl

maliberty, 2022-02-24:
@vijayank88 can you investigate and see if these are correct or not based on the LEF rules?

vijayank88, 2022-02-25:
@maliberty 
I just diff both LEF files and seen
[image]

vijayank88, 2022-03-02:
@donn  still the DRC issue exists.

vijayank88, 2022-03-03:
@maliberty 
Added Magic DRC violation details here:

Here is `drc.rpt`
```
user_proj_example
----------------------------------------
Metal3 minimum area < 0.24um^2 (met3.6)
----------------------------------------
 692.840um 966.105um 693.370um 966.150um
 692.825um 966.100um 693.370um 966.105um
 692.825um 966.080um 693.060um 966.100um
 692.825um 965.800um 692.850um 966.080um
 692.825um 965.780um 693.060um 965.800um
 692.825um 965.775um 693.155um 965.780um
----------------------------------------
[INFO]: COUNT: 6
[INFO]: Should be divided by 3 or 4
```
`drc.tcl` generated by magic tool
```
box 692.840um 966.105um 693.370um 966.150um; feedback add "Metal3 minimum area < 0.24um^2 (met3.6)" medium
box 692.825um 966.100um 693.370um 966.105um; feedback add "Metal3 minimum area < 0.24um^2 (met3.6)" medium
box 692.825um 966.080um 693.060um 966.100um; feedback add "Metal3 minimum area < 0.24um^2 (met3.6)" medium
box 692.825um 965.800um 692.850um 966.080um; feedback add "Metal3 minimum area < 0.24um^2 (met3.6)" medium
box 692.825um 965.780um 693.060um 965.800um; feedback add "Metal3 minimum area < 0.24um^2 (met3.6)" medium
box 692.825um 965.775um 693.155um 965.780um; feedback add "Metal3 minimum area < 0.24um^2 (met3.6)" medium
```
OL tag 2022.02.16_01.55.24 which passed without any magic DRC violation. 
Magic commit: `0bb6ac1fa98b5371c73156b6e876925397fb7cbc` 
open_pdks commit: `476f7428f7f686de51a5164c702629a9b9f2da46`
Corresponding bbox location screenshot:
[image]

OL tag [2022.02.21_02.33.47](https://github.com/The-OpenROAD-Project/OpenLane/releases/tag/2022.02.21_02.33.47) which throws Magic DRC violations:
Magic Commit: ` 7d601628e4e05fd17fcb80c3552dacb64e9f6e7b`
Open_pdks commit: `7519dfb04400f224f140749cda44ee7de6f5e095 `
Magic drc error bbox screenshot:
[image]

vijayank88, 2022-03-04:
Here is Magic DRC highlight:
[image]

vijayank88, 2022-03-04:
[image]
In the attached screenshot, metal3 width is 0.025um. So its less than 0.25um. Why magic reports this as violation?

AdhithyanHaridas, 2022-03-23:
@vijayank88  can you please share the drc report and the log file for the run where it fails.

vijayank88, 2022-03-26:
@AdhithyanHaridas  https://slack-files.com/T01699QAZBQ-F034BBR5T8U-2bc88bbcfc

vijayank88, 2022-03-26:
@dralabeing 
With OpenLane tag: [2022.03.15_02.27.26](https://github.com/The-OpenROAD-Project/OpenLane/releases/tag/2022.03.15_02.27.26) I am not getting any Magic DRC violations.

mousaq92, 2022-03-31:
make user_proj_example stuck at synthesis using OL Tag: 2022.03.31_02.35.19 

Used the design in this github repo
https://github.com/mxiangyue/caravel_user_project/blob/main/openlane/user_proj_example/config.tcl

closed 2022-05-03 by vijayank88