The-OpenROAD-Project/OpenLane #955: Routing congestion too high error during global routing

vijayank88, 2022-02-23:
### Description
Its a successful MPW-4 precheck passed design. Now failing with routing congestion too high error during global routing.

### Environment
```
python3 ./env.py issue-survey

Kernel: Linux v3.10.0-1160.49.1.el7.x86_64
Distribution: centos 7
Python: v3.6.8 (OK)
Container Engine: docker v20.10.12 (OK)
OpenLane Git Version: 2022.02.23_02.50.41
pip:click: INSTALLED
pip:pyyaml: INSTALLED
pip:venv: INSTALLED
---
PDK Version Verification Status: OK
---
 
```

### Reproduction Material
* Upload a tarball containing the relevant design. 
[flexbex_Routing_issue.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/8123388/flexbex_Routing_issue.zip)

* List the commands used to run the design.
- unzip the file.
- cd openroad_issue_reproducible
- openroad> source run.tcl

### Expected behavior
Should pass routing stage

### Logs
```
[INFO GRT-0053] Routing resources analysis:
          Routing      Original      Derated      Resource
Layer     Direction    Resources     Resources    Reduction (%)
---------------------------------------------------------------
li1        Vertical            0             0          0.00%
met1       Horizontal     338000        155574          53.97%
met2       Vertical       253500        185631          26.77%
met3       Horizontal     169000        118164          30.08%
met4       Vertical       101400         66330          34.59%
met5       Horizontal      33800         16770          50.38%
---------------------------------------------------------------

[INFO GRT-0101] Running extra iterations to remove overflow.
[INFO GRT-0103] Extra Run for hard benchmark.
[INFO GRT-0197] Via related to pin nodes: 149104
[INFO GRT-0198] Via related Steiner nodes: 3239
[INFO GRT-0199] Via filling finished.
[INFO GRT-0111] Final number of vias: 214085
[INFO GRT-0112] Final usage 3D: 1023424
[ERROR GRT-0118] Routing congestion too high.
Error: groute.tcl, 69 GRT-0118
[ERROR]: during executing openroad script /openlane/scripts/openroad/groute.tcl
[ERROR]: Exit code: 1
[ERROR]: Last 10 lines:
child process exited abnormally

```

vijayank88, 2022-02-23:
Configuration file used for OpenLane flow: https://github.com/phthinh/soric_project/blob/master/openlane/flexbex_core/config.tcl
If I use default `set ::env(SYNTH_STRATEGY)` flow not throwing congestion issue.
If I use user defined `set ::env(SYNTH_STRATEGY) "DELAY 2"` its failing with congestion issue.

eder-matheus, 2022-02-23:
@vijayank88 I think it is a placement issue. Grt didn't have any update regarding how it handles the congestion, so the degradation should be in the early stages of the flow. Also, I've tried to run the global router of a commercial tool on this design, and it also reports high congestion.

vijayank88, 2022-02-25:
@eder-matheus 
If I use `set ::env(SYNTH_STRATEGY) "DELAY 1"` placement congestion still looks high, but passed routing stage and flow got complete successfully.
Any idea how this synthesis strategy working?

mousaq92, 2022-02-25:
Hello, I was facing a similar issue with a design that I am working on. I'm not sure if my response will resolve your issue.

First, make sure that you set all the variables ie FP_CORE_UTIL, PL_TARGET_DENSITY in config.tcl. For some reason if you create a file called config2.tcl, and try to use ./flow.tcl -design my_design config2.tcl, config2.tcl will not be used.

Second, try to change PL_TARGET_DENSITY.

For example try to set the following in your config.tcl file:

set ::env(FP_CORE_UTIL) "40"
set ::env(PL_TARGET_DENSITY) "0.42" or "0.45"

By default PL_TARGET_DENSITY is set to 0.55 and I believe that it will not change if you change the core utilization alone. I hope this helps.

armleo, 2022-02-26:
Try reducing the core util/target density. This worked for me in the past. This should resolve it, but it's still not clear why is the congestion so high.

dralabeing, 2022-03-01:
@vijayank88 - this is a placement issue Please report after trying a different placement density. Also it looks like AREA=0 synthesis strategy resolves the congestion problem.

vijayank88, 2022-03-02:
> @eder-matheus If I use `set ::env(SYNTH_STRATEGY) "DELAY 1"` placement congestion still looks high, but passed routing stage and flow got complete successfully. Any idea how this synthesis strategy working?

@dralabeing 
I confirmed already this issue get resolved with DELAY=1/AREA=0. But MPW user passed this successfully with DELAY=2. 
So just wanted to understand how this `SYNTHESIS_STRATEGY` works. That's why I kept open this issue.

mousaq92, 2022-03-02:
Hello,

Unfortunately, I do not have OP’s RTL, but I would like to participate in this issue. I will do my analysis on SPM. One very nice feature in OpenLane is that it allows its users to do synthesis exploration. Below is a screenshot of a synthesis exploration that I conducted on spm.

[image]
 
After synth_explore, I made a run on spm with SYNTH_STRATEGY “AREA 0.” The synthesis reports show that “AREA 0” corresponds to synthesis strategy 4 in the figure above. Synthesis strategy 4 according to the synthesis exploration results is the synthesis strategy with the largest area.

I conducted another run on spm with SYNTH_STRATEGY “AREA 1.” The expectation is that “AREA 1” corresponds to the second largest area in the synthesis exploration results. The synthesis reports however show that strategy 5 was selected for “AREA 1.” This is not the second largest area.

"AREA 2" Corresponds to synthesis strategy 6. This is equivalent to strategy 5
"AREA 3" Fails at synthesis due to misformatted “AREA 3.”

Hence, I suggest that OP runs:
./flow.tcl -design my_design -tag synth_explore -synth_explore

Then change the synthesis strategy in config.tcl, run the design, and check ~/OpenLane/designs/my_design/runs/my_run/reports/synthesis

The reports there will will give OP insight on which synthesis strategy was chosen and why some strategies passed while others failed.

I hope this helps. Thank you everyone for your continued support.

dralabeing, 2022-03-03:
@vijayank88 It looks like there is no longer a congestion issues-hence marking this closed.
@mousaq92 There is no other way to pick a good synthesis strategy other than to run the slew of explorations and choose one as you have done. If you find an error during the explorations as you describe- please file a separate GH issue.

closed 2022-03-03 by dralabeing

luis201420, 2022-03-09:
@vijayank88 Could you upload the tarball containing the stage before the global placement?