The-OpenROAD-Project/OpenLane #958: Synthesis hangs at clock skew

jimbo1990, 2022-02-24:
### Description
Synthesis hangs at report_clock_skew and never continues passed that point. No errors are given it simply doesnt print and then doesnt proceed forward. 

### Environment
```
Kernel: Linux v5.10.60.1-microsoft-standard-WSL2
Distribution: ubuntu 20.04
Python: v3.8.10 (OK)
Container Engine: docker v20.10.12 (OK)
OpenLane Git Version: 2022.02.20_02.17.26
pip:click: INSTALLED
pip:pyyaml: INSTALLED
pip:venv: INSTALLED
---
PDK Version Verification Status: OK
---
```

### Reproduction Material
[image]

### Expected behavior
Should print clock skew report and continue synthesis

maliberty, 2022-02-24:
There is no attached test case so this is unactionable.  Including the tool output is not a method of reproduction.   Please see https://github.com/The-OpenROAD-Project/OpenLane/blob/master/docs/source/using_or_issue.md

jimbo1990, 2022-02-24:
> There is no attached test case so this is unactionable. Including the tool output is not a method of reproduction. Please see https://github.com/The-OpenROAD-Project/OpenLane/blob/master/docs/source/using_or_issue.md

Thanks. The only problem is the tool never fails so I don't think that or_issue.py ever executes.

maliberty, 2022-02-24:
You can run it yourself

vijayank88, 2022-02-25:
@jimbo1990 
If synthesis issue you need provide your `config.tcl` and source .v/.sv files  to check it our end.

jimbo1990, 2022-02-25:
> @jimbo1990 If synthesis issue you need provide your `config.tcl` and source .v/.sv files to check it our end.

Thanks, I will leave the files here if someone would be kind enough to try and run it to see if it passes the clock_skew_report on their side.

[Top_cell.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/8139477/Top_cell.zip)
,

vijayank88, 2022-02-25:
@jimbo1990  
Yes getting same issue at my end.
we had same issue with previous shuttle and project called `sta_hangs`. I'll try to get that project and its solutions and let you know.

jimbo1990, 2022-02-25:
> @jimbo1990 Yes getting same issue at my end. we had same issue with previous shuttle and project called `sta_hangs`. I'll try to get that project and its solutions and let you know.

Thanks for your help. This has been driving me mad.

vijayank88, 2022-02-25:
@mattvenn 
I think for one of the design, you are raised such STA hangs issue.
I don't remember the issue tag. Do you have any solution/workaround for this issue?
Thanks...

jimbo1990, 2022-02-25:
Is it possible to turn off the clock reporting? It's not really a solution but possibly to see if synthesis continues

mousaq92, 2022-02-25:
I am having the same issue with a different design. Stuck at report_clock_skew

mousaq92, 2022-02-25:
Hello,

Here is how I temporarily fixed this issue.

Comment the report_clock_skew line in /tools/OpenLane/openroad/sta.tcl and rerun.

This is definitely not the right way to do this, but the flow is moving for me. I hope this helps.

maliberty, 2022-02-26:
This is still not a well package test case.  Please provide a single self-contained invocation of the tool that fails rather than a pile of files.

jimbo1990, 2022-02-26:
I fixed temporarily solved this problem as suggested in a previous post by commenting out the report_clock_skew from the sta.tcl and also from the multiple_cornes.tcl. obviously not ideal however at least the flow finished. 

Probably IMPORTANT is that in the SDA.tcl file there is a comment which states that the function will hang if a clock less design is detected. What I don't get is why it doesn't detect the clock at that point or why it is detecting the design as clock less at that point in the synthesis while before it was calculating the slack and other parameters fine.

mithro, 2022-02-27:
Hi @jimbo1990 and @vijayank88,

There are instructions in the issue template to create the items required by the developers to fix this problem. I have included these instructions below for your convenience too;

> ### Environment
> ```
> Please run the following command in the OpenLane folder:
> 
> python3 ./env.py issue-survey
> 
> And copy and paste the ENTIRE output between the triple-backticks. Please do not gzip and upload the output.
> 
> If there's no env.py, you are using an out of date version of OpenLane and should probably update. 
> ```
> 
> ### Reproduction Material
> * Upload a tarball containing the relevant design.
> * List the commands used to run the design.
> 
> If you see a message like `Reproducible packaged: Please tarball and upload <PATH> if you're going to submit an issue` in your logs, please also tarball and include that path. This will greatly speed up the fixing process.
> 

If one of you can provide a tarball containing the relevant design, I'm sure someone like @maliberty or @donn will be able to fix the `report_clock_skew` to not hang on your design. This will be a great help to future users of these tools!

Looking forward to seeing you upload the needed information and seeing the problem getting fixed.

Tim '@mithro' Ansell

vijayank88, 2022-02-28:
> Thanks, I will leave the files here if someone would be kind enough to try and run it to see if it passes the clock_skew_report on their side. [...]

@maliberty 
or_issue.py used to reproduce issue tar file only for OpenRoad flow stage fails only. For synthesis stage if flow fails we normally upload source RTL files with config.tcl.
Steps to reproduce the issue:
1. Unzip Top_cell.zip
2. place Top_cell directory inside OpenLane/designs/ 
3. Goto OpenLane Root directory and excute following steps:
4. make mount 
5. OpenLane Container (161699a): ./flow.tcl -design Top_cell
The above step will hang during report_clock_skew
@donn  please correct me if I'm wrong
7.

vijayank88, 2022-02-28:
@maliberty  @donn 
Here's standalone testcase for sta hangs: 
[run3_sta_packaged.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/8151319/run3_sta_packaged.zip)
Steps to reproduce the issue:
-Goto OpenLane directory
-unzip the attached file inside OpenLane/_build
-Do `make mount` from `OpenLane` directory
-cd _build/run3_sta_packaged
-sta
%source opensta.tcl

This will hang during report_clock_skew

donn, 2022-02-28:
Issue confirmed... @maliberty Be aware that the issue is packaged a bit differently and so the usual run scripts are not functional here.

maliberty, 2022-02-28:
I packaged it myself and have reported it.  @donn please address this hold in packaging test cases.

maliberty, 2022-02-28:
@Manarabdelaty when you did this in sta.tcl you should have filed a bug report:
# report clock skew if the clock port is defined
# OR hangs if this command is run on clockless designs
[https://github.com/The-OpenROAD-Project/OpenLane/pull/674]

mousaq92, 2022-02-28:
How did you package the case? Thanks

maliberty, 2022-02-28:
> How did you package the case? Thanks

By hand, just looked at the sta.tcl and pulled out the relevant parts.  I'll upload my test case for comparison.

maliberty, 2022-02-28:
[bug.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/8155671/bug.zip)

jjcherry56, 2022-02-28:
@Manarabdelaty this comment is wrong. report_clock_skew works just find if there are no clocks defined.

> # OR hangs if this command is run on clockless designs

The issue has nothing to do with whether or not clocks are defined. It is a bug dealing with combinational feedback loops. OpenSTA has already been fixed but has not been integrated into openroad yet because of all the required verification steps.

donn, 2022-03-01:
@jjcherry56 @maliberty @Manarabdelaty is no longer with us aside from informal consultations. If you have any comments or observations, you tell me.

jjcherry56, 2022-03-01:
OpenROAD e3315ba41 fixes it on our end

commit b2c49ef5 (2022-03-02): [BOT] Update openroad_app (#961)

Resolves #958
closed 2022-03-02 by donn