The-OpenROAD-Project/OpenLane #960: placement.tcl removes resizer buffers

jjcherry56, 2022-02-25:
In run_placement scripts/tcl_commands/placement.tcl
    run_resizer_design

    if { [info exists ::env(DONT_BUFFER_PORTS) ]} {
        remove_buffers
    }

This undoes any buffers inserted for long wire, max slew, max cap design rule repair done by repair_design
in run_resizer_design.
The buffers should be removed BEFORE repair to removes buffers inserted by synthesis.

antonblanchard, 2022-02-25:
I've often wondered what `remove_buffers` is trying to achieve. In particular it seems to completely remove buffer tree insertion by abc, and OpenROAD has nothing similar to replace it with. Do we have evidence of abc over buffering designs?

maliberty, 2022-02-26:
@antonblanchard abc is buffering based on wire load models and may not do a good job because of it.  Trying to have the resizer fix their buffer trees hasn't worked well in the past.  The best solution would be to have resizer have a stronger buffer tree capability rather than rely on the abc buffering.  In any case it seems a bad idea to remove the resizer results.

donn, 2022-08-24:
So this was more of a nomenclature problem that has been addressed: This removes buffers placed on specific ports, such as analog ports and the like. This is unavoidable until a public API to set nets and instances as "don't touch" during resizing is surfaces, which the OR team expressed no immediate interest in.

The function has been renamed `remove_buffers_from_nets` to mitigate any further confusion.

closed 2022-08-24 by donn

maliberty, 2022-08-24:
support for dont touch has been going in this week and should be available soon

donn, 2022-08-24:
That's very good to know. Please send the function documentation when it's ready.