The-OpenROAD-Project/OpenLane #963: make pull-openlane fails on centos7

msaideroglu, 2022-02-26:
When I'm trying to run make pull-openlane command on my centos7 machine, it gives this error:

[synopsys@localhost OpenLane]$ make pull-openlane
Cannot determine git branch. Please specify OPENLANE_IMAGE_NAME manually.
Full output: error: unknown option `show-current'
usage: git branch [options] [-r | -a] [--merged | --no-merged]
   or: git branch [options] [-l] [-f] <branchname> [<start-point>]
   or: git branch [options] [-r] (-d | -D) <branchname>...
   or: git branch [options] (-m | -M) [<oldbranch>] <newbranch>

Generic options
    -v, --verbose         show hash and subject, give twice for upstream branch
    -q, --quiet           suppress informational messages
    -t, --track           set up tracking mode (see git-pull(1))
    --set-upstream        change upstream info
    -u, --set-upstream-to <upstream>
                          change the upstream info
    --unset-upstream      Unset the upstream info
    --color[=<when>]      use colored output
    -r, --remotes         act on remote-tracking branches
    --contains <commit>   print only branches that contain the commit
    --abbrev[=<n>]        use <n> digits to display SHA-1s

Specific git-branch actions:
    -a, --all             list both remote-tracking and local branches
    -d, --delete          delete fully merged branch
    -D                    delete branch (even if not merged)
    -m, --move            move/rename a branch and its reflog
    -M                    move/rename a branch, even if target exists
    --list                list branch names
    -l, --create-reflog   create the branch's reflog
    --edit-description    edit the description for the branch
    -f, --force           force creation (when already exists)
    --no-merged <commit>  print only not merged branches
    --merged <commit>     print only merged branches
    --column[=<style>]    list branches in columns
Pulling most recent OpenLane image relative to your commit...
docker pull 
"docker pull" requires exactly 1 argument.
See 'docker pull --help'.

Usage:  docker pull [OPTIONS] NAME[:TAG|@DIGEST]

Pull an image or a repository from a registry
make: *** [pull-openlane] Error 1

chenliyumich, 2022-02-27:
I also got this issue. Is git version 1.8.3.1 new enough?

msaideroglu, 2022-02-27:
I upgraded git to 2.35 from 1.8, and it worked. Thanks.

closed 2022-02-27 by msaideroglu

bean0408, 2022-03-15:
Sorry for chiming in but I'm facing the same problem with Ubuntu 18.04.6 LTS. Already updated to the latest git but to no avail. Couldn't find the git 2.35 version for my distro. Is there a workaround for this?

armleo, 2022-03-15:
It states in the error the workaround to this: Please specify OPENLANE_IMAGE_NAME manually.

Take a look at Makefile to determine what format it is.

Or build git from source.

Or update your os.

bean0408, 2022-03-15:
Hi @armleo thanks for the reply and your suggestion, really appreciate it since I'm new at Linux. I built git from source and now it's working!

luccareinehr, 2022-05-29:
Tried with git version 2.36.1 in Ubuntu 22.04 and the same error occurred. Downgrading to git 2.35 fixed it!