The-OpenROAD-Project/OpenLane #975: Running global placement through detailed routing given only lef/def

njkrichardson, 2022-03-02:
Hi there, 

Many of the global/detailed placement tools (e.g., RePlAce, OpenDP) and global/detailed routing tools (e.g., cu-gr, TritonRoute) are designed with submission to ISPD competitions in mind. Typically these competitions are not concerned with chip synthesis; and provide only an lef/def. 

Given just an lef/def for a design, how can I run global placement, detailed placement, global routing, and detailed routing via the interactive flow? 

Note: not "merged lef"

maliberty, 2022-03-02:
For the tools that are part of openroad you can just use it directly if you want to benchmark.  For example we regularly run the ispd routing benchmarks with https://github.com/The-OpenROAD-Project/OpenROAD/blob/master/src/drt/test/run-ispd.py

njkrichardson, 2022-03-02:
Cheers for the response. Two follow up questions if you don't mind. 

1. I've built the OpenLane project but not OpenRoad, is it possible to run the script you link using OpenLane, or do I additionally need to build OpenRoad? 
2. I appreciate the benchmark script, although for further context I have my own lef/def pairs for custom chip designs that I'd like to place and route. Is there a way to accomplish this for **generic** lef/def design pairs rather than the ispd designs specifically?

maliberty, 2022-03-02:
OL includes OR so you can use it from there.  I'm just saying you don't need to use the OL flow scripts.

The script shows the TCL script it generates to run the benchmark at https://github.com/The-OpenROAD-Project/OpenROAD/blob/944855835623e651e7b9c7c50efcce1fb04b4fee/src/drt/test/run-ispd.py#L83

you can do something similar with your lef/def.

njkrichardson, 2022-03-02:
Thanks again very much. I just attempted to run the script using `OpenROAD/src/drt/test` as the benchmark directory, but it seems ispd18_test1 is missing. Is there an additional step required to download and/or configure the benchmarks?

njkrichardson, 2022-03-02:
Also, can you confirm whether this script just routes, or places and routes? Thank you!

maliberty, 2022-03-02:
That script just routes.   It wouldn't be hard to add placement commands to it.

You can get the test cases from https://ispd.cc/ispd2022/index.php?page=contests  (2018 & 2019 are routing).

closed 2022-03-09 by donn