The-OpenROAD-Project/OpenLane #979: Openlane not considering initial MACRO placement ?

ratul619, 2022-03-04:
### Description

I am providing an input DEF where MACROS are arranged in array fashion and are designated as FIXED. like below 
as an input in config.tcl with below var : 

set ::env(FP_DEF_TEMPLATE)  "./designs/XYZ/top.def"

 - XYZ.abc\[0\].sram\[0\].U_sram_64_128_sky130A sram_64_128_sky130A + FIXED ( 0 0  ) N ; 
 - XYZ.abc\[0\].sram\[1\].U_sram_64_128_sky130A sram_64_128_sky130A + FIXED ( 0 100000  ) N ; 
 - XYZ.abc\[0\].sram\[2\].U_sram_64_128_sky130A sram_64_128_sky130A + FIXED ( 0 200000  ) N ; 
 - XYZ.abc\[0\].sram\[3\].U_sram_64_128_sky130A sram_64_128_sky130A + FIXED ( 0 300000  ) N ; 
 - XYZ.abc\[1\].sram\[0\].U_sram_64_128_sky130A sram_64_128_sky130A + FIXED ( 200000 0  ) N ; 
 - XYZ.abc\[1\].sram\[1\].U_sram_64_128_sky130A sram_64_128_sky130A + FIXED ( 200000 100000  ) N ; 
 - XYZ.abc\[1\].sram\[2\].U_sram_64_128_sky130A sram_64_128_sky130A + FIXED ( 200000 200000  ) N ; 
 - XYZ.abc\[1\].sram\[3\].U_sram_64_128_sky130A sram_64_128_sky130A + FIXED ( 200000 300000  ) N ; 

But seems although read in the flow  , this is NOT honored , as i see in below file 

tmp/floorplan/3-initial_fp.def 

INFO ODB-0134] Finished DEF file: /openlane/designs/XYZ/runs/RUN_2022.03.03_18.37.17/tmp/floorplan/3-initial_fp.def
[WARNING PPL-0015] Macro XYZ.abc\[0\].sram\[0\].U_sram_64_128_sky130A is not placed.
[WARNING PPL-0015] Macro XYZ.abc\[0\].sram\[1\].U_sram_64_128_sky130A is not placed.
[WARNING PPL-0015] Macro XYZ.abc\[0\].sram\[2\].U_sram_64_128_sky130A is not placed.
[WARNING PPL-0015] Macro XYZ.abc\[0\].sram\[3\].U_sram_64_128_sky130A is not placed.
[WARNING PPL-0015] Macro XYZ.abc\[1\].sram\[0\].U_sram_64_128_sky130A is not placed.
[WARNING PPL-0015] Macro XYZ.abc\[1\].sram\[1\].U_sram_64_128_sky130A is not placed.
[WARNING PPL-0015] Macro XYZ.abc\[1\].sram\[2\].U_sram_64_128_sky130A is not placed.

Does openlane honor these initial fixed attrbute OR there is a bug ?? 

### Environment
```

Kernel: Linux v5.13.0-30-generic
Distribution: centos 7
Python: v3.6.8 (OK)
Critical Alert: No Docker or Docker-compatible container engine was found: {e}.
OpenLane Git Version: 2f75eb2
pip:click: INSTALLED
pip:pyyaml: INSTALLED
pip:venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

2f75eb2 2022-02-09T07:41:34+02:00 [BOT] Update netgen (#817) - Openlane Bot -  (HEAD -> master, tag: 2022.02.10_02.15.04, origin/master, origin/HEAD)
1860ec4 2022-02-07T22:27:50+02:00 add sky130hs/hd setup for gcd (#905) - Matt Liberty -  (tag: 2022.02.08_01.48.52)
5f7d5ca 2022-02-05T04:20:30+02:00 More ASAP7 Tweaks (#904) - Donn -  (tag: 2022.02.06_01.54.17)

```

### Logs
```
INFO GPL-0021] MacroInstsArea: 7192367048000
[InitialPlace]  Iter: 1 CG Error: 0.03083327 HPWL: 3847596530
[InitialPlace]  Iter: 2 CG Error: 0.00001568 HPWL: 3666953558
[InitialPlace]  Iter: 3 CG Error: 0.00001278 HPWL: 3664912776
[InitialPlace]  Iter: 4 CG Error: 0.00001767 HPWL: 3665020684
[InitialPlace]  Iter: 5 CG Error: 0.00003950 HPWL: 3665246740
[InitialPlace]  Iter: 6 CG Error: 0.00002820 HPWL: 3665375439
[InitialPlace]  Iter: 7 CG Error: 0.00004185 HPWL: 3665387688
[InitialPlace]  Iter: 8 CG Error: 0.00004118 HPWL: 3665404711
[InitialPlace]  Iter: 9 CG Error: 0.00004291 HPWL: 3665396910
[InitialPlace]  Iter: 10 CG Error: 0.00057883 HPWL: 3665422761
[InitialPlace]  Iter: 11 CG Error: 0.00003200 HPWL: 3665774806
[InitialPlace]  Iter: 12 CG Error: 0.00003018 HPWL: 3664738521
[InitialPlace]  Iter: 13 CG Error: 0.00005711 HPWL: 3664652012
[InitialPlace]  Iter: 14 CG Error: 0.00003870 HPWL: 3664635349
[InitialPlace]  Iter: 15 CG Error: 0.00002841 HPWL: 3664601830
[InitialPlace]  Iter: 16 CG Error: 0.00004972 HPWL: 3664610971
[InitialPlace]  Iter: 17 CG Error: 0.00002752 HPWL: 3664599426
[InitialPlace]  Iter: 18 CG Error: 0.00007053 HPWL: 3664612217
[InitialPlace]  Iter: 19 CG Error: 0.00002628 HPWL: 3664590098
[InitialPlace]  Iter: 20 CG Error: 0.00018324 HPWL: 3664596757

[ERROR]: during executing openroad script /openlane/scripts/openroad/replace.tcl
[ERROR]: Exit code: 1
[ERROR]: Last 10 lines:
child killed: kill signal

```

kunalg123, 2022-03-04:
We had done 2 Macro based PNR using OpenLane. You might want to take a look
https://github.com/manili/VSDBabySoC
https://github.com/vsdip/VSDMemSoC

vijayank88, 2022-03-04:
@ratul619 
Missing issue reproducible tar file. Check your run directory or Please package a standalone test case with https://github.com/The-OpenROAD-Project/OpenLane/blob/master/docs/source/using_or_issue.md

ratul619, 2022-03-04:
[openroad_issue_reproducible.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/8187380/openroad_issue_reproducible.tar.gz)

vijayank88, 2022-03-05:
@ratul619 
missing library files from your attachment.
```
Error: pdn.tcl, 19 cannot read file ./designs/vector_processor_v1/sram_64_128_sky130A_SS_1p8V_25C.lib.
while evaluating source run.tcl
```

ratul619, 2022-03-05:
[sram_64_128_sky130A_SS_1p8V_25C.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/8190825/sram_64_128_sky130A_SS_1p8V_25C.zip)
[sram_64_128_sky130A.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/8190826/sram_64_128_sky130A.zip)

ratul619, 2022-03-05:
uploaded in zip files

ratul619, 2022-03-05:
Quick help will be highly appreciated !

vijayank88, 2022-03-07:
@ratul619 
If it's a MPW-5 design, share your repo details.
I'm facing issue while loading DEF file in OpenROAD GUI.

vijayank88, 2022-03-07:
@ratul619 
Child killed signal is related to system resources not enough to run your design.
My local machine with 16GB got hanged for your design, but cloud server with 128GB RAM able to complete PDN stage. Here's logs:
```
[WARNING PSM-0030] VSRC location at (12605.520um, 3370.880um) and size 10.000um, is not located on a power stripe. Moving to closest stripe at (12603.600um, 3397.250um).
[WARNING PSM-0030] VSRC location at (13445.520um, 3370.880um) and size 10.000um, is not located on a power stripe. Moving to closest stripe at (13446.000um, 3397.250um).
[WARNING PSM-0030] VSRC location at (14285.520um, 3370.880um) and size 10.000um, is not located on a power stripe. Moving to closest stripe at (14283.000um, 3397.250um).
[WARNING PSM-0030] message limit reached, this message will no longer print
[INFO PSM-0031] Number of PDN nodes on net VPWR = 64834223.

```

ratul619, 2022-03-07:
Hi , Havent made any repo yet  ; my local machine is 24GB , does not hang though and i am using docker..

What is your suggestion ? 

What is your recommendation ?

vijayank88, 2022-03-07:
@ratul619 
Try to update openlane latest version and try again

ratul619, 2022-03-07:
I just reduced the size of the die and seems PDN stage went ahead. but stuck in resizer step. 
I am attaching the tar file in sometime

ratul619, 2022-03-07:
also can u share steps for loading and viewing the DEF in openroad GUI. 

I suspect some issue in the DEF macro placements

vijayank88, 2022-03-07:
Inside OL Container:
%openroad -gui
or
%openroad
openroad > gui::show

vijayank88, 2022-03-07:
> @ratul619 Try to update openlane latest version and try again. Seems you're using 2022.02.10_02.15.04. Latest tag is [2022.03.03_02.36.44](https://github.com/The-OpenROAD-Project/OpenLane/releases/tag/2022.03.03_02.36.44)

Have you updated the OpenLane Tag?

dralabeing, 2022-03-07:
@vijayank88 Why is this an Openlane release issue? Can you check the DEF file and see if there are any issues with placement
@ratul619 You need to ensure you have an you X11 server up and configure the DISPLAY option.  Refer to this pr for setup details: https://github.com/The-OpenROAD-Project/OpenLane/pull/740. Unfortunately, there isn't sufficient documentation on this.

vijayank88, 2022-03-08:
> also can u share steps for loading and viewing the DEF in openroad GUI.
> 
> I suspect some issue in the DEF macro placements

@ratul619 
Just viewed your raw DEF file. Seems DIE AREA is smaller than your fixed MACRO placements.
`DIEAREA ( 0 0 ) ( 50000000 50000000 ) ;`
```
    - U_controller.lane\[3\].sram\[0\].U_sram_64_128_sky130A sram_64_128_sky130A + FIXED ( 600000000 0 ) N ;
    - U_controller.lane\[3\].sram\[1\].U_sram_64_128_sky130A sram_64_128_sky130A + FIXED ( 600000000 100000000 ) N ;
    - U_controller.lane\[3\].sram\[2\].U_sram_64_128_sky130A sram_64_128_sky130A + FIXED ( 600000000 200000000 ) N ;
    - U_controller.lane\[3\].sram\[3\].U_sram_64_128_sky130A sram_64_128_sky130A + FIXED ( 600000000 300000000 ) N ;
```
Please make sure your macros should place inside core area.
Do correct your macro placement or DIE area and re-try again.

vijayank88, 2022-03-08:
@dralabeing 
Latest version of OpenLane configured to load OpenRoad GUI inside docker itself.

In my GCP not able to load DEF file. 

Its got struck during power domain creation, as macros not placed properly.

vijayank88, 2022-03-08:
Something wrong with your design configuration.
[image]

ratul619, 2022-03-09:
HI , 

Now i fixed the DEF and coodrinates for the macros.

Also updated to latest openlane version 

But still failing , not sure why . 

DIEAREA ( 0 0 ) ( 375000 375000 ) ;
COMPONENTS 1653 ;
    - U_controller.lane\[0\].sram\[0\].U_sram_64_128_sky130A sram_64_128_sky130A + FIXED ( 1000 3000 ) N ;
    - U_controller.lane\[0\].sram\[1\].U_sram_64_128_sky130A sram_64_128_sky130A + FIXED ( 1000 5000 ) N ;
    - U_controller.lane\[0\].sram\[2\].U_sram_64_128_sky130A sram_64_128_sky130A + FIXED ( 1000 7000 ) N ;
    - U_controller.lane\[0\].sram\[3\].U_sram_64_128_sky130A sram_64_128_sky130A + FIXED ( 1000 10000 ) N ;
    - U_controller.lane\[1\].sram\[0\].U_sram_64_128_sky130A sram_64_128_sky130A + FIXED ( 3000 3000 ) N ;
    - U_controller.lane\[1\].sram\[1\].U_sram_64_128_sky130A sram_64_128_sky130A + FIXED ( 3000 5000 ) N ;

Log  :

INFO ODB-0128] Design: top
[INFO ODB-0130]     Created 35 pins.
[INFO ODB-0131]     Created 1653 components and 12840 component-terminals.
[INFO ODB-0133]     Created 2696 nets and 5204 connections.
[INFO ODB-0134] Finished DEF file: /openlane/designs/vector_processor_v1/runs/RUN_2022.03.09_13.24.11/results/floorplan/top.def
[INFO PDN-0016] Power Delivery Network Generator: Generating PDN
  config: /openlane/scripts/openroad/pdn_cfg.tcl
Error: pdn_cfg.tcl, 45 empty command name ""
[ERROR]: during executing openroad script /openlane/scripts/openroad/pdn.tcl
[ERROR]: Exit code: 1
[ERROR]: Last 10 lines:
child process exited abnormally

[INFO]: Saving runtime environment...
[INFO]: Creating reproducible...

dralabeing, 2022-03-09:
@vijayank88 Can you help isolate the macro placement problem separately without the GUI to see where and why pdn  is hanging? That should help clarify the root cause of this problem. Thanks

vijayank88, 2022-03-10:
@ratul619 
Your floorplan seems have issue:
[image]

Still your macro placed outside die area.
If I try to run through GUI getting following error:
```
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[INFO ODB-0222] Reading LEF file: ./tmp/merged_unpadded.lef
[INFO ODB-0223]     Created 13 technology layers
[INFO ODB-0224]     Created 25 technology vias
[INFO ODB-0225]     Created 442 library cells
[INFO ODB-0226] Finished LEF file:  ./tmp/merged_unpadded.lef
[INFO ODB-0127] Reading DEF file: ./in.def
[INFO ODB-0128] Design: top
[INFO ODB-0130]     Created 35 pins.
[INFO ODB-0131]     Created 1653 components and 12840 component-terminals.
[INFO ODB-0133]     Created 2696 nets and 5204 connections.
[INFO ODB-0134] Finished DEF file: ./in.def
[INFO PDN-0016] Power Delivery Network Generator: Generating PDN
  config: openlane/scripts/openroad/pdn_cfg.tcl
Error: pdn_cfg.tcl, 45 empty command name ""
[ERROR PSM-0058] No rows defined in design. Use initialize_floorplan to add rows.
[ERROR GUI-0070] Error: pdn.tcl, 44 PSM-0058
```

Are you running `flow.tcl` with automated mode or interactive stage?
Share your input DB (RTL/macros/config.tcl) as a zip file to check what's the issue.

ratul619, 2022-03-10:
Are you running flow.tcl with automated mode or interactive stage?
-- Running in automated mode. 

Since i am not able to open the GUI, hence difficult to check . 

Sharing the config.tcl

ratul619, 2022-03-10:
[RTL.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/8221116/RTL.tar.gz)
[config.tcl.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/8221118/config.tcl.tar.gz)

[macros.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/8221117/macros.tar.gz)

vijayank88, 2022-03-10:
> Are you running flow.tcl with automated mode or interactive stage? -- Running in automated mode.
> 
> Since i am not able to open the GUI, hence difficult to check .
> 
> Sharing the config.tcl

@ratul619 
For standalone to openROAD GUI install this: https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts
Instruction for installation: https://openroad.readthedocs.io/en/latest/user/BuildLocally.html
Invoke GUI : OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad -gui

vijayank88, 2022-03-10:
> [RTL.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/8221116/RTL.tar.gz) [config.tcl.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/8221118/config.tcl.tar.gz)
> 
> [macros.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/8221117/macros.tar.gz)

@ratul619  macros.cfg missing

ratul619, 2022-03-10:
U_controller.lane\[0\].sram\[0\].U_sram_64_128_sky130A   1 3   N  
U_controller.lane\[0\].sram\[1\].U_sram_64_128_sky130A   1 5   N  
U_controller.lane\[0\].sram\[2\].U_sram_64_128_sky130A   1 7   N  
U_controller.lane\[0\].sram\[3\].U_sram_64_128_sky130A   1 10   N  
U_controller.lane\[1\].sram\[0\].U_sram_64_128_sky130A   3 3   N  
U_controller.lane\[1\].sram\[1\].U_sram_64_128_sky130A   3 5   N  
U_controller.lane\[1\].sram\[2\].U_sram_64_128_sky130A   3 7   N  
U_controller.lane\[1\].sram\[3\].U_sram_64_128_sky130A   3 10   N  
U_controller.lane\[2\].sram\[0\].U_sram_64_128_sky130A   5 3   N  
U_controller.lane\[2\].sram\[1\].U_sram_64_128_sky130A   5 5   N  
U_controller.lane\[2\].sram\[2\].U_sram_64_128_sky130A   5 7   N  
U_controller.lane\[2\].sram\[3\].U_sram_64_128_sky130A   5 10   N  
U_controller.lane\[3\].sram\[0\].U_sram_64_128_sky130A   7 3   N  
U_controller.lane\[3\].sram\[1\].U_sram_64_128_sky130A   7 5   N  
U_controller.lane\[3\].sram\[2\].U_sram_64_128_sky130A   7 7   N  
U_controller.lane\[3\].sram\[3\].U_sram_64_128_sky130A   7 10   N

vijayank88, 2022-03-10:
@ratul619 
Have you explored how to do macro placement with OpenLane flow? example here https://github.com/The-OpenROAD-Project/OpenLane/tree/master/designs/manual_macro_placement_test
The tool will automatically will place macros if you define `set ::env(EXTRA_LIBS)` and `set ::env(EXTRA_LEFS)`.
Other way manually place based on macro size with fixed locations.

Also while updating configuration variable in `config.tcl` understand the requirement and include it.

So much confusion with the `config.tcl` you shared.

Here is my simple `config.tcl` which passed macro placement stage based on your design.
```
# User config
set ::env(DESIGN_NAME) top

# Change if needed
set ::env(VERILOG_FILES) [glob $::env(DESIGN_DIR)/src/*.v  $::env(DESIGN_DIR)/src/*.sv]

# Fill this
set ::env(CLOCK_PERIOD) "10.0"
set ::env(CLOCK_PORT) "clk"

set ::env(FP_CORE_UTIL) 35
set ::env(FP_PDN_VOFFSET) 0
set ::env(FP_PDN_VPITCH) 30

set filename $::env(DESIGN_DIR)/$::env(PDK)_$::env(STD_CELL_LIBRARY)_config.tcl
if { [file exists $filename] == 1} {
	source $filename
}

set ::env(BASE_SDC_FILE) $::env(DESIGN_DIR)/src/top.sdc

set ::env(EXTRA_LIBS) $::env(DESIGN_DIR)/macros/sram_64_128_sky130A_SS_1p8V_25C.lib

set ::env(EXTRA_LEFS) $::env(DESIGN_DIR)/macros/sram_64_128_sky130A.lef
#set ::env(MACRO_PLACEMENT_CFG) $::env(DESIGN_DIR)/macro.cfg
#set ::env(FP_PDN_CHECK_NODES) 0
set ::env(VDD_PIN) {VPWR vdd}
set ::env(GND_PIN) {VGND gnd}
```

ratul619, 2022-03-10:
Ok. Now even im confused. 

Are you saying if i include below two lines , tool WILL ignore the macro.cfg file which has placement information for macros ?

set ::env(EXTRA_LIBS) $::env(DESIGN_DIR)/macros/sram_64_128_sky130A_SS_1p8V_25C.lib
set ::env(EXTRA_LEFS) $::env(DESIGN_DIR)/macros/sram_64_128_sky130A.lef 

If i need to resolve the LEF and LIB for the macro , what variables should i use then? 

The example provided has both macro config file as well as extra LEF var , but NO .lib flie 

set ::env(MACRO_PLACEMENT_CFG) $::env(DESIGN_DIR)/macro_placement.cfg
set ::env(EXTRA_LEFS) [glob $::env(DESIGN_DIR)/macros/lef/*.lef]

vijayank88, 2022-03-10:
> Ok. Now even im confused.
> 
> Are you saying if i include below two lines , tool WILL ignore the macro.cfg file which has placement information for macros ?
macro.cfg is user fixed location for macro placement. If you define `set ::env(MACRO_PLACEMENT_CFG) [glob $::env(DESIGN_DIR)/macro.cfg]` tool will give priority for user input.
> 
> set ::env(EXTRA_LIBS) $::env(DESIGN_DIR)/macros/sram_64_128_sky130A_SS_1p8V_25C.lib set ::env(EXTRA_LEFS) $::env(DESIGN_DIR)/macros/sram_64_128_sky130A.lef
> 
> If i need to resolve the LEF and LIB for the macro , what variables should i use then?
The above set variables are right for user LEF/LIB. You can modify the path as per your need and proceed.
> 
> The example provided has both macro config file as well as extra LEF var , but NO .lib flie
> 
> set ::env(MACRO_PLACEMENT_CFG) $::env(DESIGN_DIR)/macro_placement.cfg set ::env(EXTRA_LEFS) [glob $::env(DESIGN_DIR)/macros/lef/*.lef]

ratul619, 2022-03-10:
Now stopping at placement stage 

NesterovSolve] Iter: 2120 overflow: 249.471 HPWL: 4132759579
[NesterovSolve] Iter: 2130 overflow: 249.47 HPWL: 4133815412
[NesterovSolve] Iter: 2140 overflow: 249.47 HPWL: 4134563060
[NesterovSolve] Iter: 2150 overflow: 249.471 HPWL: 4135514266
[NesterovSolve] Iter: 2160 overflow: 249.47 HPWL: 4136322386
[NesterovSolve] Iter: 2170 overflow: 249.471 HPWL: 4137213283
[NesterovSolve] Iter: 2180 overflow: 249.471 HPWL: 4137902909
[NesterovSolve] Iter: 2190 overflow: 249.471 HPWL: 4138341551
[NesterovSolve] Iter: 2200 overflow: 249.471 HPWL: 4139122254
[NesterovSolve] Iter: 2210 overflow: 249.471 HPWL: 4139999563
[NesterovSolve] Iter: 2220 overflow: 249.47 HPWL: 4140804660
[NesterovSolve] Iter: 2230 overflow: 249.471 HPWL: 4141775490
[NesterovSolve] Iter: 2240 overflow: 249.47 HPWL: 4142585470
[NesterovSolve] Iter: 2250 overflow: 249.47 HPWL: 4143073909
[NesterovSolve] Iter: 2260 overflow: 249.471 HPWL: 4143762109
[ERROR GPL-0305] RePlAce diverged at newStepLength.
Error: replace.tcl, 90 GPL-0305
[ERROR]: during executing openroad script /openlane/scripts/openroad/replace.tcl
[ERROR]: Exit code: 1
[ERROR]: Last 10 lines:
child process exited abnormally

Die area and sram locations seems to be OK , all within boundary : 

DIEAREA ( 0 0 ) ( 3000000 3000000 ) ;
.
.
    - U_controller.lane\[0\].sram\[0\].U_sram_64_128_sky130A sram_64_128_sky130A + FIXED ( 1000 3000 ) N ;
    - U_controller.lane\[0\].sram\[1\].U_sram_64_128_sky130A sram_64_128_sky130A + FIXED ( 1000 5000 ) N ;
    - U_controller.lane\[0\].sram\[2\].U_sram_64_128_sky130A sram_64_128_sky130A + FIXED ( 1000 7000 ) N ;
    - U_controller.lane\[0\].sram\[3\].U_sram_64_128_sky130A sram_64_128_sky130A + FIXED ( 1000 10000 ) N ;
    - U_controller.lane\[1\].sram\[0\].U_sram_64_128_sky130A sram_64_128_sky130A + FIXED ( 3000 3000 ) N ;
    - U_controller.lane\[1\].sram\[1\].U_sram_64_128_sky130A sram_64_128_sky130A + FIXED ( 3000 5000 ) N ;
    - U_controller.lane\[1\].sram\[2\].U_sram_64_128_sky130A sram_64_128_sky130A + FIXED ( 3000 7000 ) N ;
    - U_controller.lane\[1\].sram\[3\].U_sram_64_128_sky130A sram_64_128_sky130A + FIXED ( 3000 10000 ) N ;
    - U_controller.lane\[2\].sram\[0\].U_sram_64_128_sky130A sram_64_128_sky130A + FIXED ( 5000 3000 ) N ;
    - U_controller.lane\[2\].sram\[1\].U_sram_64_128_sky130A sram_64_128_sky130A + FIXED ( 5000 5000 ) N ;
    - U_controller.lane\[2\].sram\[2\].U_sram_64_128_sky130A sram_64_128_sky130A + FIXED ( 5000 7000 ) N ;
    - U_controller.lane\[2\].sram\[3\].U_sram_64_128_sky130A sram_64_128_sky130A + FIXED ( 5000 10000 ) N ;
    - U_controller.lane\[3\].sram\[0\].U_sram_64_128_sky130A sram_64_128_sky130A + FIXED ( 7000 3000 ) N ;
    - U_controller.lane\[3\].sram\[1\].U_sram_64_128_sky130A sram_64_128_sky130A + FIXED ( 7000 5000 ) N ;
    - U_controller.lane\[3\].sram\[2\].U_sram_64_128_sky130A sram_64_128_sky130A + FIXED ( 7000 7000 ) N ;
    - U_controller.lane\[3\].sram\[3\].U_sram_64_128_sky130A sram_64_128_sky130A + FIXED ( 7000 10000 ) N ;

ratul619, 2022-03-10:
[openroad_issue_reproducible.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/8222641/openroad_issue_reproducible.tar.gz)

armleo, 2022-03-10:
Wait, how big is the sram? No way they are 1um x 1um? The srams in the def you provided are definitely overlapping.

vijayank88, 2022-03-10:
@ratul619  Are you able to install OpenROAD for GUI usage?
Still macro placement going wrong. Please load your DEF file KLayout/OpenROAD GUI and check it how your macros got placed.

vijayank88, 2022-03-10:
> in LEF size :
> 
> MACRO sram_64_128_sky130A CLASS BLOCK ; ORIGIN 0.000 0.000 ; SIZE 1011.05 BY 444.61 ; SYMMETRY X Y R90 ; PIN din0[0] DIRECTION INPUT ; PORT LAYER met4 ;
> 
> So the size is around ~1200um x 450um (correct ?? )
> 
> So seemingly kept a spacing of ~2000um in between them in DEF.

how you generated this SRAM module?

armleo, 2022-03-10:
You sure the lef is correct? Its size (assuming the same dbu per micron) is 1.2um by 0.4um which seems wrong

ratul619, 2022-03-10:
The original LEF version below (generated from compiler) 

VERSION 5.4 ;
NAMESCASESENSITIVE ON ;
BUSBITCHARS "[]" ;
DIVIDERCHAR "/" ;
UNITS
  DATABASE MICRONS 2000 ;
END UNITS
MACRO sram_64_128_sky130A
   CLASS BLOCK ;
   SIZE 1011.05 BY 444.61 ;
   SYMMETRY X Y R90 ;
   PIN din0[0]
      DIRECTION INPUT ;
      PORT
         LAYER met4 ;
         RECT  159.46 0.0 160.2 1.93 ;
      END
   END din0[0]

I just changed to below (referring to some released area - where i saw LEFs did not have the DATABASE microns command) 

ERSION 5.4 ;
NAMESCASESENSITIVE ON ;
BUSBITCHARS "[]" ;
DIVIDERCHAR "/" ;

MACRO sram_64_128_sky130A
   CLASS BLOCK ;
   ORIGIN 0.000 0.000 ;
   SIZE 1011.05 BY 444.61 ;
   SYMMETRY X Y R90 ;
   PIN din0[0]
      DIRECTION INPUT ;
      PORT
         LAYER met4 ;
         RECT  159.46 0.0 160.2 1.93 ;
      END
   END din0[0]

armleo, 2022-03-10:
lef is definitely wrong. The placement seems to be wrong.

Fix this before proceeding with openlane

ratul619, 2022-03-10:
Yes. Combinations of ALL inputs files dont seem to be in SYNC.  

@vijayank88  - let me install and check. 

I am closing this for now. 

Thanks all for the help.

closed 2022-03-10 by ratul619