The-OpenROAD-Project/OpenLane #980: Detailed Placement Error

Sya0, 2022-03-04:
```
Kernel: Linux v5.11.0-49-generic
Distribution: ubuntu 21.04
Python: v3.9.5 (OK)
Container Engine: docker v20.10.7 (OK)
OpenLane Git Version: -dev
pip:click: INSTALLED
pip:pyyaml: INSTALLED
pip:venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

e4bfdd7 2022-02-22T16:59:14-03:00 Remove pip install, no longer needed (#953) - Vitor Bandeira -  (grafted, HEAD, tag: 2022.02.23_02.50.41)
```

I have tried to harden my design but I have got below problem. My tcl scripts look ok, however, it still gives me error. I saw the same issue [here](https://github.com/The-OpenROAD-Project/OpenLane/issues/810). They patched openroad for that issue and mine is newer version of openlane.
[openroad_issue_reproducible.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/8183649/openroad_issue_reproducible.tar.gz)

vijayank88, 2022-03-04:
@Sya0 
Update your `config.tcl` with below variable and run again.. it will resolve the issue.
```
set ::env(CELL_PAD) 2
```

Sya0, 2022-03-04:
which config ?

vijayank88, 2022-03-04:
@Sya0 
Its a MPW project. Check https://github.com/efabless/caravel_user_project/tree/main/openlane/user_project_*/config.tcl
Update the `config.tcl`

Sya0, 2022-03-04:
In `user_project_wrapper/config.tcl` there is no line for `CELL_PAD`. I added to config.tcl `set ::env(CELL_PAD) 2` but same error occured.

vijayank88, 2022-03-04:
@Sya0 
If you do hardening first you have to run `make user_proj_example` then you've to run `make user_project_wrapper`.

During which command the attached openroad_issue_reproducible.zip generated?

closed 2022-03-04 by Sya0