The-OpenROAD-Project/OpenLane #981: Flow crashes at the detailed routing stage.

kauser-rl, 2022-03-04:
### Description
I'm running my design through the latest version of Openlane and it crashes at the detailed routing phase.

### Environment
```
Kernel: Linux v3.10.0-1160.36.2.el7.x86_64
Distribution: centos 7
Python: v3.6.8 (OK)
Critical Alert: No Docker or Docker-compatible container engine was found: {e}.
OpenLane Git Version: 9602463
pip:click: INSTALLED
pip:pyyaml: INSTALLED
pip:venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

9602463 2022-03-03T00:07:19+02:00 Add Commit-Based Docker Tags (#967) - Donn -  (HEAD -> master, tag: 2022.03.03_02.36.44, origin/master, origin/HEAD)
b2c49ef 2022-03-02T21:39:51+02:00 [BOT] Update openroad_app (#961) - Openlane Bot -  ()
4076f3e 2022-03-02T01:24:37+02:00 Synthesis Exploration Tweaks (#973) - Donn -  (tag: 2022.03.02_02.59.05)
```

### Reproduction Material
You can download the tarball from the link [here](https://transfer.sh/tsMj3p/openroad_issue_reproducible.tgz).

### Expected behavior
The design hasn't changed. I was re-running a regression after updating the version of Openlane. First it was just getting stuck after multiple passes on detailed routing. I tried to reduce the FP density and then hit this issue.

### Logs
```
[INFO DRT-0194] Start detail routing.
[INFO DRT-0195] Start 0th optimization iteration.
    Completing 10% with 0 violations.
    elapsed time = 00:00:47, memory = 5998.45 (MB).
    Completing 20% with 0 violations.
    elapsed time = 00:04:14, memory = 10657.78 (MB).
[ERROR]: during executing openroad script /openlane/scripts/openroad/droute.tcl
[ERROR]: Exit code: 1
[ERROR]: Last 10 lines:
child killed: kill signal

[INFO]: Saving runtime environment...
[INFO]: Creating reproducible...
```

vijayank88, 2022-03-04:
@kauser-rl 
May i know system RAM size currently used for this design run?

kauser-rl, 2022-03-04:
RAM: 16GB.
I tried the link again, it seems to work for me.
[image]

vijayank88, 2022-03-04:
Secure Connection Failed

An error occurred during a connection to transfer.sh.

    The page you are trying to view cannot be shown because the authenticity of the received data could not be verified.
    Please contact the website owners to inform them of this problem.

kauser-rl, 2022-03-04:
[openroad_issue_reproducible.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/8184760/openroad_issue_reproducible.zip)
Here you go.

vijayank88, 2022-03-04:
@kauser-rl 
Even I've faced same issue with 2 routing cores with 16GB. How many core does your CPU have?
My cloud server running good with your design routing. 
```
[INFO DRT-0195] Start 57th optimization iteration.
    Completing 10% with 0 violations.
    elapsed time = 00:00:00, memory = 14456.70 (MB).
    Completing 20% with 0 violations.
    elapsed time = 00:00:00, memory = 14456.70 (MB).
    Completing 30% with 0 violations.
    elapsed time = 00:00:00, memory = 14456.70 (MB).
    Completing 40% with 0 violations.
    elapsed time = 00:00:00, memory = 14456.70 (MB).
    Completing 50% with 0 violations.
    elapsed time = 00:00:00, memory = 14456.70 (MB).
    Completing 60% with 0 violations.
    elapsed time = 00:00:00, memory = 14456.70 (MB).
    Completing 70% with 0 violations.
    elapsed time = 00:00:00, memory = 14456.70 (MB).
    Completing 80% with 0 violations.
    elapsed time = 00:00:00, memory = 14456.70 (MB).
    Completing 90% with 0 violations.
    elapsed time = 00:00:00, memory = 14456.70 (MB).
    Completing 100% with 0 violations.
    elapsed time = 00:00:00, memory = 14456.70 (MB).
[INFO DRT-0199]   Number of violations = 0.
[INFO DRT-0267] cpu time = 00:00:04, elapsed time = 00:00:00, memory = 14456.70 (MB), peak = 15295.34 (MB)
Total wire length = 4057449 um.
Total wire length on LAYER li1 = 0 um.
Total wire length on LAYER met1 = 1856311 um.
Total wire length on LAYER met2 = 1927691 um.
Total wire length on LAYER met3 = 218300 um.
Total wire length on LAYER met4 = 55110 um.
Total wire length on LAYER met5 = 35 um.
Total number of vias = 350684.
Up-via summary (total 350684):.

-------------------------
 FR_MASTERSLICE         0
            li1    126368
           met1    192477
           met2     27381
           met3      4456
           met4         2
-------------------------
                   350684

[INFO DRT-0198] Complete detail routing.
Total wire length = 4057449 um.
Total wire length on LAYER li1 = 0 um.
Total wire length on LAYER met1 = 1856311 um.
Total wire length on LAYER met2 = 1927691 um.
Total wire length on LAYER met3 = 218300 um.
Total wire length on LAYER met4 = 55110 um.
Total wire length on LAYER met5 = 35 um.
Total number of vias = 350684.
Up-via summary (total 350684):.

-------------------------
 FR_MASTERSLICE         0
            li1    126368
           met1    192477
           met2     27381
           met3      4456
           met4         2
-------------------------
                   350684

[INFO DRT-0267] cpu time = 04:10:12, elapsed time = 00:15:45, memory = 14456.70 (MB), peak = 15295.34 (MB)

[INFO DRT-0180] Post processing.
Saving to ./out.def
```
See the Peak Memory close to 15GB. Stop backend RAM usage if any other process running. Re-try again, its your system resource issue, not the tool issue.

kauser-rl, 2022-03-04:
Thanks @vijayank88. In case you are interested to know the config-
```
Architecture:          x86_64
CPU op-mode(s):        32-bit, 64-bit
Byte Order:            Little Endian
CPU(s):                4
On-line CPU(s) list:   0-3
Thread(s) per core:    2
Core(s) per socket:    2
Socket(s):             1
NUMA node(s):          1
Vendor ID:             GenuineIntel
CPU family:            6
```
I'm happy for you to say it is not a tool issue. Sadly this is the machine I have for now so perhaps I'll have to explore alternatives. My question would be, the design hasn't changed in the last few months and I simply updated the Openlane version. The tool was comfortably able to route the design in the same machine before. Food for thought.

vijayank88, 2022-03-04:
@kauser-rl 
As I said try to stop backend process(close all application other the OpenLane flow)
8 CORES you have. Try setting below variable in config.tcl and try once
```
set ::env(ROUTING_CORES) 6
```

vijayank88, 2022-03-04:
@donn Please share your view on RAM usage.

donn, 2022-03-04:
@vijayank88 @kauser-rl If you have concerns about RAM usage, the best approach is a [local install](https://github.com/The-OpenROAD-Project/OpenLane/blob/master/docs/source/local_installs.md), bypassing the Docker container entirely.

Otherwise, not much that can be done on our side. The tools do require some RAM.

vijayank88, 2022-03-04:
@kauser-rl 
Hope above solution will help to fix your system resource. So closing this issue. If any other issue feel free to re-open this.

closed 2022-03-04 by vijayank88

kauser-rl, 2022-03-04:
I tried with 6 cores and I'm still hitting the same issue. Looks like the options I have is to either not use the Docker image (which is what I'm going to try next) or find the cash for a beefier machine 🤑.

kauser-rl, 2022-03-04:
Thanks for the help.