The-OpenROAD-Project/OpenLane #992: Getting error in openroad/opendp.tcl

EmJunaid, 2022-03-09:
[openroad_issue_reproducible (1).zip](https://github.com/The-OpenROAD-Project/OpenLane/files/8217579/openroad_issue_reproducible.1.zip)
What I have done so far is that I have not make any github repo yet I have just clone caravel_user_project complete repo and install all requirements and were able to run counter example as well. Now I just want to run my own design so what I did is that i just place my verilog file in same folder where counter.v is placed and made these changes to config.tcl of openlane:
[image]
I'm not able to understand what do I need to write in CLOCK_NET variable, my main module calls 7 more modules so do I need to define all of them here like it was defined for counter?
so kindly guide me how to correctly edit config.tcl file required for my design and why I'm getting error.

vijayank88, 2022-03-10:
@EmJunaid 
CLOCK_NET is the clock input port used in Clock Tree synthesis. If you make it empty CTS stage will be skipped.
Refer this: https://github.com/The-OpenROAD-Project/OpenLane/search?l=Markdown&q=CLOCK_NET

vijayank88, 2022-03-10:
@EmJunaid 
While file a github issue please follow issue template and provide all inputs for what's the exact issue with the log files.
As per slack channel message: https://skywater-pdk.slack.com/archives/C016UL7AQ73/p1646827636470839 
The error log is:

```
[INFO]: Incremented step index to 15.
[INFO]: Running Detailed Placement...
OpenROAD 8d53e9b018dec98fa63e907ddeb6c5406f035361
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[INFO ODB-0222] Reading LEF file: /home/em/mpw/caravel_user_project/openlane/user_proj_example/runs/user_proj_example/tmp/merged_unpadded.lef
[INFO ODB-0223]     Created 13 technology layers
[INFO ODB-0224]     Created 25 technology vias
[INFO ODB-0225]     Created 441 library cells
[INFO ODB-0226] Finished LEF file:  /home/em/mpw/caravel_user_project/openlane/user_proj_example/runs/user_proj_example/tmp/merged_unpadded.lef
[INFO ODB-0127] Reading DEF file: /home/em/mpw/caravel_user_project/openlane/user_proj_example/runs/user_proj_example/tmp/routing/15-diodes.def
[INFO ODB-0128] Design: Core
[INFO ODB-0094] 		Created 100000 Insts
[INFO ODB-0130]     Created 176 pins.
[INFO ODB-0131]     Created 182492 components and 501426 component-terminals.
[INFO ODB-0132]     Created 2 special nets and 440168 connections.
[INFO ODB-0133]     Created 11292 nets and 61258 connections.
[INFO ODB-0134] Finished DEF file: /home/em/mpw/caravel_user_project/openlane/user_proj_example/runs/user_proj_example/tmp/routing/15-diodes.def
[INFO DPL-0034] Detailed placement failed on:
[INFO DPL-0035]  ANTENNA__16421__A1
[INFO DPL-0035]  ANTENNA__16421__B2
[INFO DPL-0035]  ANTENNA__19468__D
[INFO DPL-0035]  ANTENNA__19468__D
[INFO DPL-0035]  ANTENNA__19468__D
[INFO DPL-0035]  ANTENNA__19468__D
[INFO DPL-0035]  ANTENNA_clkbuf_leaf_24_clock_A
[INFO DPL-0035]  ANTENNA_clkbuf_leaf_24_clock_A
[INFO DPL-0035]  ANTENNA_rebuffer20_A
[INFO DPL-0035]  ANTENNA_rebuffer49_A
[INFO DPL-0035]  ANTENNA_rebuffer64_A
[INFO DPL-0035]  ANTENNA_rebuffer64_A
[INFO DPL-0035]  ANTENNA_rebuffer88_A
[INFO DPL-0035]  ANTENNA_rebuffer88_A
[INFO DPL-0035]  ANTENNA_rebuffer88_A
[ERROR DPL-0036] Detailed placement failed.
Error: opendp.tcl, 32 DPL-0036
[ERROR]: during executing openroad script /openlane/scripts/openroad/opendp.tcl
[ERROR]: Exit code: 1
[ERROR]: Last 10 lines:
child process exited abnormally[INFO]: Creating reproducible...
[INFO]: Saving runtime environment...or_issue.py OpenROAD Issue PackagerEFABLESS CORPORATION AND ALL AUTHORS OF THE OPENLANE PROJECT SHALL NOT BE HELD
LIABLE FOR ANY LEAKS THAT MAY OCCUR TO ANY PROPRIETARY DATA AS A RESULT OF USING
THIS SCRIPT. THIS SCRIPT IS PROVIDED ON AN "AS IS" BASIS, WITHOUT WARRANTIES OR
CONDITIONS OF ANY KIND.BY USING THIS SCRIPT, YOU ACKNOWLEDGE THAT YOU FULLY UNDERSTAND THIS DISCLAIMER
AND ALL IT ENTAILS.Parsing config file(s)…
Setting up /home/em/mpw/caravel_user_project/openlane/user_proj_example/runs/user_proj_example/openroad_issue_reproducible…
[FIN] Done.
[INFO]: Reproducible packaged: Please tarball and upload user_proj_example/runs/user_proj_example/openroad_issue_reproducible if you're going to submit an issue.
[INFO]: Calculating Runtime From the Start...
[INFO]: Generating Final Summary Report...
[INFO]: Design Name: Core
Run Directory: /home/em/mpw/caravel_user_project/openlane/user_proj_example/runs/user_proj_example
Source not found.
```
To fix above issue set following variable in `user_proj_example/config.tcl`
```
set ::env(CELL_PAD) 2
```

EmJunaid, 2022-03-10:
@vijayank88 
Again getting the same exact error even after defining the above variable

vijayank88, 2022-03-10:
Here's log. No errors with setting CELL_PAD=2
```
OpenLane Container (e4bfdd7):/openlane/designs/gh_OL_992$ openroad
OpenROAD 8d53e9b018dec98fa63e907ddeb6c5406f035361 
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
openroad> source run.tcl 
[INFO ODB-0222] Reading LEF file: ./tmp/merged_unpadded.lef
[INFO ODB-0223]     Created 13 technology layers
[INFO ODB-0224]     Created 25 technology vias
[INFO ODB-0225]     Created 441 library cells
[INFO ODB-0226] Finished LEF file:  ./tmp/merged_unpadded.lef
[INFO ODB-0127] Reading DEF file: ./in.def
[INFO ODB-0128] Design: Core
[INFO ODB-0094] 		Created 100000 Insts
[INFO ODB-0130]     Created 176 pins.
[INFO ODB-0131]     Created 182492 components and 501426 component-terminals.
[INFO ODB-0132]     Created 2 special nets and 440168 connections.
[INFO ODB-0133]     Created 11292 nets and 61258 connections.
[INFO ODB-0134] Finished DEF file: ./in.def
Placement Analysis
---------------------------------
total displacement    4134804.4 u
average displacement       22.7 u
max displacement          542.6 u
original HPWL         3089095.0 u
legalized HPWL        5006229.1 u
delta HPWL                   62 %

[INFO DPL-0020] Mirrored 20657 instances
[INFO DPL-0021] HPWL before          5006229.1 u
[INFO DPL-0022] HPWL after           4996099.1 u
[INFO DPL-0023] HPWL delta               -0.2 %
openroad> 
```

vijayank88, 2022-03-10:
@EmJunaid 
what is CORE_UTIL and PL_DENSITY?
There are plenty free space in your Core.
[image]

closed 2022-03-10 by donn

Abdulwadoodd, 2022-03-18:
@vijayank88 How do you see this core in gui? Is this MAGIC tool? Can you briefly explain how to get that view of the core in gui?

vijayank88, 2022-03-18:
@Abdulwadoodd 
Try the below command.
`openroad -gui`
Followed by you need to do 
`read_lef` and `read_def`