The-OpenROAD-Project/OpenLane #996: OpenLane flow causes segfaults during groute

abdullahyildiz, 2022-03-12:
### Description
I get the following error during groute:
`[ERROR]: during executing openroad script /openlane/scripts/openroad/groute.tcl`

### Environment
OpenLane environment has been installed as part of
[Caravel User Project - mpw-5c release](https://github.com/efabless/caravel_user_project/tree/mpw-5c)
```
$ python3 ./env.py issue-survey

dependencies.get_tag.NoGitException: Cannot determine git branch. Please specify OPENLANE_IMAGE_NAME manually.
Full output: error: unknown option `show-current'

```

Commands used to run the design:
`make oc_can MGMT_AREA_ROOT=$UPRJ_ROOT/mgmt_core_wrapper`

### Reproduction Material
[openroad_issue_reproducible.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/8238224/openroad_issue_reproducible.tar.gz)

### Expected behavior
Process should not cause segfaults.

### Logs
```
[INFO GRT-0103] Extra Run for hard benchmark.
Stack trace:
 0# 0x0000000000FD1B49 in openroad
 1# 0x00007F469DF93400 in /lib64/libc.so.6
 2# grt::FastRouteCore::mazeRouteMSMD(int, int, float, int, int, bool, int, float, int, int, int) in openroad
 3# grt::FastRouteCore::run() in openroad
 4# grt::GlobalRouter::findRouting(std::vector<grt::Net*, std::allocator<grt::Net*> >&, int, int) in openroad
 5# grt::GlobalRouter::globalRoute() in openroad
 6# 0x00000000017ADF1F in openroad
 7# 0x00007F46A0F26EB2 in /lib64/libtcl8.5.so
 8# 0x00007F46A0F6B36C in /lib64/libtcl8.5.so
 9# TclObjInterpProcCore in /lib64/libtcl8.5.so
10# 0x00007F46A0F26EB2 in /lib64/libtcl8.5.so
11# 0x00007F46A0F6B36C in /lib64/libtcl8.5.so
12# 0x00007F46A0F73647 in /lib64/libtcl8.5.so
13# TclEvalObjEx in /lib64/libtcl8.5.so
14# 0x00007F46A0FAE27F in /lib64/libtcl8.5.so
15# 0x00007F46A0F26EB2 in /lib64/libtcl8.5.so
16# 0x00007F46A0F6B36C in /lib64/libtcl8.5.so
17# 0x00007F46A0F73647 in /lib64/libtcl8.5.so
18# TclEvalObjEx in /lib64/libtcl8.5.so
19# 0x00007F46A0F2E1D0 in /lib64/libtcl8.5.so
20# 0x00007F46A0F26EB2 in /lib64/libtcl8.5.so
21# 0x00007F46A0F6B36C in /lib64/libtcl8.5.so
22# 0x00007F46A0F73647 in /lib64/libtcl8.5.so
23# TclEvalObjEx in /lib64/libtcl8.5.so
24# 0x00007F46A0F31F00 in /lib64/libtcl8.5.so
25# 0x00007F46A0F26EB2 in /lib64/libtcl8.5.so
26# 0x00007F46A0F6B36C in /lib64/libtcl8.5.so
27# TclObjInterpProcCore in /lib64/libtcl8.5.so
28# 0x00007F46A0F26EB2 in /lib64/libtcl8.5.so
29# 0x00007F46A0F6B36C in /lib64/libtcl8.5.so
30# TclObjInterpProcCore in /lib64/libtcl8.5.so
31# 0x00007F46A0F26EB2 in /lib64/libtcl8.5.so
32# 0x00007F46A0F27F1E in /lib64/libtcl8.5.so
33# Tcl_EvalEx in /lib64/libtcl8.5.so
34# Tcl_Eval in /lib64/libtcl8.5.so
35# sta::sourceTclFile(char const*, bool, bool, Tcl_Interp*) in openroad
36# ord::tclAppInit(Tcl_Interp*) in openroad
37# Tcl_Main in /lib64/libtcl8.5.so
38# main in openroad
39# __libc_start_main in /lib64/libc.so.6
40# 0x0000000000FCE417 in openroad
[ERROR]: during executing openroad script /openlane/scripts/openroad/groute.tcl
```

jurevreca12, 2022-03-19:
I am getting the same error. Is there any progress/workaround on this?

maliberty, 2022-03-19:
This design has a very high placement density at the center:
[image]

I suggest you try lowering the placement density as a workaround for now.  Was this actually run through OR's global placer?  It seems unusual to have a diamond pattern and it suggests it wasn't.

abdullahyildiz, 2022-03-20:
@maliberty,

When we tried to lower PL_TARGET_DENSITY, the flow failed by suggesting to increase the placement density.

maliberty, 2022-03-20:
PL_BASIC_PLACEMENT is generally a bad idea imho

Prit1097, 2022-03-24:
@abdullahyildiz Can you please upload the whole config setup for your design so that I can solve the error at my end ?

abdullahyildiz, 2022-03-25:
@Prit1097,

Do you mean [Reproduction Material](https://github.com/The-OpenROAD-Project/OpenLane/files/8238224/openroad_issue_reproducible.tar.gz)?

Prit1097, 2022-03-25:
@abdullahyildiz yes.

Prit1097, 2022-03-25:
@abdullahyildiz the reproduction material you already have uploaded and that is not enough. Can you please upload your entire design if it's possible?

Prit1097, 2022-03-25:
@abdullahyildiz we are suspecting that the placement quality is bad. So if you upload your entire design we will be able to make a run for placement from our side.

abdullahyildiz, 2022-03-25:
@Prit1097,

Please run the following commands:

```
git clone https://github.com/yongatek/caravel_yonga-occan-controller.git <PROJ_ROOT>
cd <PROJ_ROOT>
git checkout bb6cd36
make oc_can
```

mousaq92, 2022-03-25:
Hello,

@Prit1097 and I are working on this issue. We cloned the design and updated config.tcl in /openlane/oc_can by removing PL_BASIC_PLACEMENT. We also changed PL_TARGET_DENSITY to 0.25.

The design passed global routing. It is currently in detailed routing. I will update you with the results when detailed routing completes.

Thanks,

Mousa

mousaq92, 2022-03-25:
Hello @abdullahyildiz ,

This is to confirm that the flow passes using the configuration changes mentioned above.

Please note that the flow may also pass with a target utilization in the range [0.25, 0.6).

Here is a screenshot showing that the flow succeeded at 0.25 target utilization.

[image]

Thanks,

Mousa

abdullahyildiz, 2022-03-26:
@mousaq92, @Prit1097,

Thanks for your support.

maliberty, 2022-03-28:
The crash still needs to be resolved so I'm reopening for that.