The-OpenROAD-Project/OpenLane #1000: Number of pins exceed number of valid positions

0616ygh, 2022-03-14:
### Description
In order to test the joint workflow of Genus and OpenLane, I synthesized a simple design based on Sky130A with Genus and imported it into OpenLane. When running pin-placement, the program gave the error report as shown in the figure.
[image]

### Environment
```
Kernel: Linux v4.18.0-240.1.1.el8_3.x86_64
Distribution: centos 8
Python: v3.10.2 (OK)
Container Engine: docker v20.10.7 (OK)
OpenLane Git Version: 5f20beb7928c7329ea1a199b9f494f48f2e6c080
pip:click: INSTALLED
pip:pyyaml: INSTALLED
pip:venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

5f20beb 2022-03-11T01:56:51+02:00 Fix issue with threads option in `run_designs.py` (#994) - Donn -  (HEAD -> master, tag: 2022.03.11_02.27.13, origin/master, origin/HEAD)
a82bf22 2022-03-11T01:47:51+02:00 scripts/checkers: fix report_file_relative scope (#989) - Johan Euphrosine -  ()
ec35dd7 2022-03-11T01:47:12+02:00 scripts/tcl_commands/scrot_klayout: add missing -log arguments (#990) - Johan Euphrosine -  (). 
```
### Reproduction Material
* Upload a tarball containing the relevant design.
[openlane_reproducible_6.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/8245147/openlane_reproducible_6.zip)
* List the commands used to run the design.
`./flow.tcl -interactive`
`prep -design BM64`
`run_floorplan`

### Expected behavior
Know the cause of the problem and fix it.

### Logs
```
OpenROAD 944855835623e651e7b9c7c50efcce1fb04b4fee 
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[INFO ODB-0222] Reading LEF file: /openlane/designs/BM64/runs/RUN_2022.03.13_10.54.33/tmp/merged_unpadded.lef
[INFO ODB-0223]     Created 13 technology layers
[INFO ODB-0224]     Created 25 technology vias
[INFO ODB-0225]     Created 441 library cells
[INFO ODB-0226] Finished LEF file:  /openlane/designs/BM64/runs/RUN_2022.03.13_10.54.33/tmp/merged_unpadded.lef
[WARNING IFP-0028] Core area lower left (5.520, 10.880) snapped to (5.520, 10.880).
[INFO IFP-0001] Added 202 rows of 1197 sites.
[INFO] Extracting DIE_AREA and CORE_AREA from the floorplan
[INFO] Floorplanned on a die area of 0.0 0.0 561.7 572.42 (microns). Saving to /openlane/designs/BM64/runs/RUN_2022.03.13_10.54.33/reports/floorplan/7-initial_fp_die_area.rpt.
[INFO] Floorplanned on a core area of 5.52 10.88 556.14 560.32 (microns). Saving to /openlane/designs/BM64/runs/RUN_2022.03.13_10.54.33/reports/floorplan/7-initial_fp_core_area.rpt.
[INFO ODB-0222] Reading LEF file: /openlane/designs/BM64/runs/RUN_2022.03.13_10.54.33/tmp/merged.lef
[INFO ODB-0223]     Created 13 technology layers
[INFO ODB-0224]     Created 25 technology vias
[INFO ODB-0225]     Created 441 library cells
[INFO ODB-0226] Finished LEF file:  /openlane/designs/BM64/runs/RUN_2022.03.13_10.54.33/tmp/merged.lef
[INFO ODB-0127] Reading DEF file: /openlane/designs/BM64/runs/RUN_2022.03.13_10.54.33/tmp/floorplan/7-initial_fp.def
[INFO ODB-0128] Design: BM64
[INFO ODB-0130]     Created 1028 pins.
[INFO ODB-0131]     Created 4426 components and 35055 component-terminals.
[INFO ODB-0133]     Created 5460 nets and 17351 connections.
[INFO ODB-0134] Finished DEF file: /openlane/designs/BM64/runs/RUN_2022.03.13_10.54.33/tmp/floorplan/7-initial_fp.def
Found 0 macro blocks.
Using 1u default distance from corners.
[INFO PPL-0007] Random pin placement.
[ERROR PPL-0072] Number of pins (1028) exceed number of valid positions (688).
Error: ioplacer.tcl, 60 PPL-0072
```

maliberty, 2022-03-14:
This is a user error - you have more pins than places along the edges of your design to place them.  You need a bigger area or to use more layers.  If you feel the message is erroneous please state why.

donn, 2022-03-16:
Matt is correct. You simply need a larger footprint. There physically isn't enough space for the number of pins you have.

closed 2022-03-16 by donn

0616ygh, 2022-03-16:
Ok, thank you for your reply. Let me reconfigure it carefully. Thank you very much.

donn, 2022-04-28:
@shustinski `set ::env(DIE_AREA) {0 0 1000 1000}` sets an absolute size. You can increase or decrease it as you see fit.

shustinski, 2022-04-28:
I have already tried this, but it doesn't increase the number of valid pins positions

shustinski, 2022-04-28:
@donn Thank you, it really helps

b224hisl, 2023-02-17:
I have set` "FP_SIZING": "relative"`, but I also got this problem, doesn't that option means the area size is automatic increased? @donn

donn, 2023-02-18:
@b224hisl It does not. Relative only means the size is based on the amount of cells you have and your utilization. Pins are currently not factored into this equation.